raspberrypi/bookworm-feedback #1: Expanding filesystem currently broken

ghollingworth, 2023-08-22:
Expanding the ext4 filesystem to fill the SD card on first boot is currently broken.

paulober, 2023-08-22:
Can confirm. Had the same issue but `raspi-config` was able to extend it manually.

lurch, 2023-09-22:
@XECDesign This has been fixed now, right?

XECDesign, 2023-09-22:
yes

martignoni, 2023-09-27:
Confirming, resizing works out of the box with image 2023-09-22.

andrum993, 2023-09-28:
Adding my +1 - resize works on Pi 4B with both USB boot and root (SATA SSD) and internal SD card boot and root.

closed 2023-10-06 by ghollingworth

munderwoods, 2024-03-20:
I'm actually having this issue on the version I downloaded from here:
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2023-12-11/2023-12-11-raspios-bookworm-armhf-lite.img.xz

Am I doing something dumb?

lurch, 2024-03-21:
@munderwoods I've just tested this myself, writing the image you linked above onto a 16GB microSD card, and booting it up on a Raspberry Pi Zero 2 W.
On the first bootup, it generates the SSH keys and resizes the root filesystem. It then automatically reboots and on the second bootup it asks me for keyboard and username settings. When I then login and run `lsblk` I can see that `mmcblk0p2` (the root filesystem) has been resized to 14.3GB.

Can you please give more details on how this isn't working for you / what behaviour you're seeing instead?

munderwoods, 2024-04-04:
> @munderwoods I've just tested this myself, writing the image you linked above onto a 16GB microSD card, and booting it up on a Raspberry Pi Zero 2 W. On the first bootup, it generates the SSH keys [...]

Super sorry, I totally wasted your time. It was entirely my mistake!

lurch, 2024-04-04:
Easily done, when the difference is `>` vs. `>>` :slightly_smiling_face:   Glad to hear that things are working now.