raspberrypi/bookworm-feedback #3: Issue setting WiFi country requires kernel patch

ghollingworth, 2023-08-22:
https://forums.raspberrypi.com/viewtopic.php?t=354313&sid=6c7a670a1aea59e78eb573c8ed09a259#p2124616

pelwell, 2023-08-22:
Likewise - no reason not to update stable to the latest 6.1, it's been quiet for a while.

lurch, 2023-08-23:
IIRC this kernel also includes the patch that improves the compatibility between the Touch Display and the Camera Module 3 ?

XECDesign, 2023-09-22:
This was fixed a while back.

martignoni, 2023-09-27:
Confirming. Fixed as image 2023-09-22:
```
$ sudo update-alternatives --display regulatory.db
regulatory.db - auto mode
  link best version is /lib/firmware/regulatory.db-debian
  link currently points to /lib/firmware/regulatory.db-debian
  link regulatory.db is /lib/firmware/regulatory.db
  slave regulatory.db.p7s is /lib/firmware/regulatory.db.p7s
/lib/firmware/regulatory.db-debian - priority 100
  slave regulatory.db.p7s: /lib/firmware/regulatory.db.p7s-debian
/lib/firmware/regulatory.db-upstream - priority 50
  slave regulatory.db.p7s: /lib/firmware/regulatory.db.p7s-upstream
```

closed 2023-10-06 by ghollingworth