raspberrypi/bookworm-feedback #4: Python environment externally managed

ghollingworth, 2023-08-22:
Debian have inserted a check to stop people from breaking the computer by using pip install...

This means you will need to create a virtual environment if you want to install python packages, or apt install the python3-xxxxxx package (if one exists for the import you are looking for)

What should we do for Bookworm?

1) Undo the Debian'ism hoping that the distribution doesn't actually get broken when you pip install or sudo pip install.  In the meantime update all Python tutorials and guides to add instructions to create and use virtual environments.  Then make this change in the next Debian release hoping that in the next two years, we can safely assume people are doing this properly.

2) Automatically create a virtual environment and source the activate script from a menu command.  This won't actually fix the problem unless people know to use the link instead of just trying from the command line.

3) Write an application which is a python 'project manager' from which you can create and destroy projects.  Refer to the project manager in the pip error message.

nmstoker, 2023-09-28:
The check has proven quite useful in my experience (avoids accidents).

The versions I've seen it on all have a helpful message explaining the options (similar to what's below).

Therefore might it be reasonable for one more option:

**0. Accept the Debianism, keep the user instructions (if present, add it if not) but add a link to a Raspberry Pi documentation page that gives a fuller explanation for those who may need extra guidance.**

The wording below would probably need some minor tweaks to reference Pi OS Bookworm in place of Debian references.

````
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
````

BlackNet, 2023-10-06:
In the 3D printing universe most all of the applications creates a virtual environment and installs into that. Which is the proper way to handle this change.  

Having base core libraries installed via apt is workable and helpful to.  As for creating a virtual environment or a management tool that may not be needed at all.

lurch, 2023-10-06:
See https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/os/using-python.adoc

ghollingworth, 2023-10-06:
Still not sure if it wouldn't be better to change the document to recommend virtualenv instead, it looks much easier to use...

nmstoker, 2023-10-06:
@ghollingworth - that's certainly an option but it's effectively the same as using venv as covered below. The essential steps are pretty simple in both and since venv is built in it gets my vote

https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/os/using-python.adoc#using-pip-with-virtual-environments

nmstoker, 2023-10-06:
That whole page seemed spot on to me, and gives the background for why the change is sensible (even if the word "imposed" starts things off as if it's going to be divisive but it isn't)

closed 2023-10-09 by ghollingworth

timonsku, 2023-10-19:
@ghollingworth 
Would it be possible to maybe implement what is proposed in the PEP here?
https://peps.python.org/pep-0668/#create-separate-distro-and-local-directories

The change is quite disruptive and a "simple" split of distro and global adminstrator packages seem like a very good solution.
I'm not sure why this behaviour is currently only happening when compiling and installing your own upstream install of Python. I feel like this could be better solved with keeping python-full for user use and a separate python-full-distro.

Aside from that potential solution it would be good to have pipx shipped by default as the note currently suggests using it but needs to be installed manually first.
Also I'm not sure if its supposed to print this message when not installing globally? It pops up for both `sudo pip` and `pip`.
If I execute it unprivileged I would normally expect the install to happen into local user path and not warrant a warning like this.
Unsure if this differs at all from upstream Debian or not but its a bit puzzling behaviour.

timonsku, 2023-10-20:
Another good thing to add would be pdm as an alternative to pip for things that aren't applications:
https://pdm.fming.dev/latest/

It takes a lot of the hassle out of venvs and if a good split between distro packages and user packages can't be made, transitioning to pdm for tutorials etc. might be the best next thing.