raspberrypi/bookworm-feedback #5: Bundled WayVNC version is old, and lacks websocket support

maxnet, 2023-08-22:
The Wayvnc version shipped is old, and lacks websocket support, requiring users to install TigerVNC to connect to the Pi.
Would be nicer if user could just point their browser to http://raspberrypi.local/ and be presented a modern web vnc client like novnc without having to install anything on their normal computer....

Steps needed to remedy this problem would be:

- Provide newer packages for wayvnc and its dependencies (from same author) libaml0, libneatvnc0
- Change raspi-config to provide easy option for user to enable webvnc instead of traditional vnc.

Configuration change raspi-config would need to do would be:

- Install a webserver, novnc web client, and change the configuration so that user authentication is done on webserver instead of at wayvnc level. Change wayvnc to only listen to localhost or an unix socket, and tell webserver to proxy remote address to that.

I have had success with compiling and installing a newer version of wayvnc from source, and doing:

```
sudo apt install novnc nginx libnginx-mod-http-auth-pam
```

Changing configuration as follows:

/etc/xdg/autostart/wayvnc.desktop

```
[Desktop Entry]
Type=Application
Name=wayvnc
Comment=Start wayvnc
NoDisplay=true
Exec=/usr/bin/wayvnc --render-cursor --websocket 127.0.0.1
OnlyShowIn=wayfire
```

~/.config/wayvnc/config

```
address=127.0.0.1
enable_auth=false
enable_pam=true
private_key_file=/home/max/.config/wayvnc/key.pem
certificate_file=/home/max/.config/wayvnc/cert.pem
```

/etc/nginx/sites-available/default;

```
server {
        listen 80 default_server;
        listen [::]:80 default_server;
        root /usr/share/novnc;
        index vnc_lite.html index.html;
        server_name _;

        location / {
                try_files $uri $uri/ =404;
        }

        location /websockify {
            proxy_pass http://127.0.0.1:5900;
            proxy_http_version 1.1;
            proxy_set_header Upgrade $http_upgrade;
            proxy_set_header Connection "Upgrade";
            proxy_set_header Host $host;
        }

        auth_pam              "Secure Zone";
        auth_pam_service_name "vncserver";
}
```

Add www-data to shadow (so that nginx can read /etc/shadow for PAM autentication), and restart nginx

```
sudo usermod -aG shadow www-data
sudo systemctl restart nginx
```

(That's for the most basic configuration in which webserver just listens to port 80. Naturally you could also let it do SSL instead).

ghollingworth, 2023-08-23:
Nice,  Would this also support the tigerVNC like interface as well as the novnc one?

Definitely worth investigating, although possibly not in the first release.

maxnet, 2023-08-23:
>Nice, Would this also support the tigerVNC like interface as well as the novnc one?

Don't think Wayvnc offers an option to listen for both normal VNC connections, and websocket connections at the same time, if that is what you are asking.

And I personally prefer to handle authentication at a different level when using websockets, in which case it does requires different settings anyway.
NoVNC does offer prompting the user for a normal VNC password as well, like a normal VNC client would, and using that to connect to the VNC server.
But it does not currently support the exact authentication scheme Wayvnc expects.

And doing authentication at webserver level (with VNC server software only listening locally, and webserver proxying external connections) has the advantage that it provides an additional level of security.
User needs to enter the username+password before nginx will serve the NoVNC files, and play proxy server for the websocket connection.
(While security bugs could theoretically occur in both VNC server software and webserver software, I trust webserver software better as gatekeeper, because webserver software is designed to be used on the big bad Internet, and had a lot more "real-life security testing").

[image]

[image]

lurch, 2023-10-09:
Just a bit more info for context: `apt show` says that wayvnc 0.6.2-1~bpo12+1 is installed, and https://github.com/any1/wayvnc/releases says that waynvc 0.7.0 is the version that added websocket support.

ghollingworth, 2023-10-09:
We're expecting to test and push 0.7.0 after the official launch of Bookworm.

Gordon

Jhutjens92, 2023-10-31:
0.7.0 is released but not yet available according to apt-cache
[image]

sweetrobot, 2023-11-10:
update?

ghollingworth, 2023-11-10:
Yes...

sudo apt update

closed 2023-11-10 by ghollingworth