raspberrypi/bookworm-feedback #6: "apt changelog" broken for some Raspberry Pi-provided packages

andrum993, 2023-08-22:
The `apt changelog` command fails to display the package changelog for some Raspberry Pi-provided packages in Bookworm.

Example:

`andrew@raspberrypi:~ $ apt changelog raspi-config
E: Failed to fetch changelog:/raspi-config.changelog  Changelog unavailable for raspi-config=20230821
`

The problem affects at least the following packages:

- raspi-config
- vlc
- raspberrypi-ui-mods

I would expect to see the same issue with other Raspberry Pi-provided packages, but not all of them. The Raspberry Pi-provided `kodi` package does not suffer from this problem, and neither does `wf-panel-pi`.

I have previously thought that the problem was caused by apt ignoring the changelog if the file was named `changelog.gz`, and only displaying it if it was named `changelog.Debian.gz`. It seems that this is not the case, however, since for the `wf-panel-pi` package the changelog file is named `changelog.gz`, and it displays correctly. So something else is going on.

Linked forum thread: https://forums.raspberrypi.com/viewtopic.php?t=355062.

XECDesign, 2023-09-22:
I don't think this is a Raspberry Pi OS issue. We haven't modified apt in any way and our packaging is relatively standard as far as changelogs go.

I was under the impression that these sorts of tools download the changelog from one of pre-defined locations. We don't store our changelogs anywhere outside of the package, so I would not have expected this to work in the first place.

If it has some fallback where it tried to extract the changelog from the .deb and fails, then it's a Debian issue.

andrum993, 2023-09-22:
I don't know where the problem lies - I was just reporting the issue.

As to how `apt` works, I have read some of the relevant source code (at https://salsa.debian.org/apt-team/apt/-/blob/main/apt-pkg/acquire-item.cc), and it seems to have a fallback to allow it to read the changelog file from `/usr/share/doc`, which is where I was manually getting the changelogs from when `apt changelog` failed. I am assuming that all installed packages have their changelog deposited there when the package is installed.

If the bug is in Debian, then assuming Raspberry Pi want it fixed (which may or may not be the case), then Raspberry Pi will need to speak to the Debian folks about it, since I don't have the requisite knowledge and experience to deal with this, and I may not have the requisite skills either.

cillian64, 2023-10-09:
Had a little dig in to this, I don't think the problem is the changelog.gz files: On my system `apt changelog rpi-firefox-mods` works but `apt changelog rpi-chromium-mods` doesn't, and both of those have a /usr/share/doc/<PKGNAME>/changelog.gz file in the correct format.

Spent a while debugging apt and it seems that in the bad case it's deciding it can't find the specified package.  But from strace it is opening `/usr/share/doc/rpi-chromium-mods/changelog.gz`, but doesn't seem to be doing anything with it.

I did find that if you have a version of the package installed which is no longer available in the apt repo (eg because you haven't upgraded and we've pruned an older version of the package) then `apt changelog` seems to fail, but I also have a bad example where that isn't the case.