raspberrypi/bookworm-feedback #10: NFS Mount different from Bullseye?

GeekyTim, 2023-08-23:
I have a tool that I use to configure a bunch of small specific tools on all my Pi's, including some NFS mounts to my NAS. Currently, the /etc/fstab entry is:

prometheus:PiShared /home/geekytim/PiShared nfs nfsvers=4,rw,soft 0 0

But this does not work within bookworm.

GeekyTim, 2023-08-23:
This might be something specific to my NAS; the firmware was recently updated. I just went back to some of my other Pi servers, and they are having problems, but reporting different issues.

GeekyTim, 2023-08-23:
Allowing ALL local IP addresses allows me to mount a drive from the NAS on my Pi5/Bookworm. Using the Pi 5's hostname does not work - although it does work on Bullseye images on earlier Pis (2/3/4). Hard to tell if this is a bookworm issue or a NAS issue at the moment, so I need to look into this further before we consider this a problem with this latest Pi OS. 

I must admit I have disabled SMB on the NAS, which many would take as their default sharing protocol, but since 95% of my home servers are Pi's and the other two run Linux, I want to stay away from the dark side.

gitbls, 2023-08-31:
I'm not seeing any issues mounting shares on Bullseye Pi or x64 servers from my Bookworm client.

closed 2023-10-09 by ghollingworth