raspberrypi/bookworm-feedback #11: Overscan setting in first-boot wizard has no effect

lurch, 2023-08-24:


lurch, 2023-08-24:
Same is true of the "Raspberry Pi Configuration" program.

popcornmix, 2023-08-24:
I think its likely that the overscan setting will be disabled on first release.

The correct way for handling overscan is by reducing both source and destination window sizes
(rather than bullseye way of just reducing the destination size which requires more resources and produces a lower quality result).

But the bookworm version of wlroots isn't new enough, so that needs bumping. And rebasing our existing patches onto that is not trivial.

For now its best to fix the issue in the display's settings menu.

lurch, 2023-09-11:
The overscan setting has now been removed if the first-boot wizard is running under Wayland (i.e. if you're using a Pi4 or a Pi400). Should I close this issue, or should I leave it open (and rename it) until wlroots / WayFire properly supports overscan?

closed 2023-10-09 by lurch