raspberrypi/bookworm-feedback #14: No on-screen keyboard available in Wayland

lurch, 2023-08-25:
Which means that (unlike in Bullseye) you can't use the Touch Display for typing text-input without a USB keyboard attached.

ghollingworth, 2023-10-09:
Have you tried squeekboard?

steveiliop56, 2023-10-12:
Onboard is the best option.

lurch, 2023-10-12:
> Onboard is the best option.

That doesn't work with the Wayland desktop - as soon as you click on one of the keys the whole application just quits.

steveiliop56, 2023-10-12:
I think it works... Not sure tho will try again...

ghollingworth, 2023-10-12:
No onboard won't work with Wayland (so won't work on Pi 4 or Pi 5), it displays fine (because it is using the XWayland backend) but is trying to use an X backend to send the keys...

It seems the answer is:

sudo apt install wvkbd

closed 2023-10-12 by ghollingworth

steveiliop56, 2023-10-21:
Is there something better? This keyboard is a bit trash and the switches doesn't work...

steveiliop56, 2024-05-21:
Nope but I have found a solution I am happy with. I created a small menu item to open the keyboard with a button. Just open the menu editor and create a new item (I created it in help). Then for the command add this ```sh -c "pkill wvkbd-mobintl || wvkbd-mobintl -L 200"``` (the 200 is the size in pixels). After you do that you will have a button in the menu you selected and when you click it the keyboard appears if you click it again it disappears.

thisisgregm, 2024-05-21:
Thank you @steveiliop56 - this is what I found as well (on the official forums). Maybe it was your post there, not sure. I use full screen browser window a lot, so the button is then not visible :(

steveiliop56, 2024-05-21:
Yeah it was my comment probably. I think a user in the forums managed to do it with a swipe gesture

AgentRev, 2024-05-28:
I'm somewhat flabbergasted that RPiOS doesn't ship with a pre-installed onscreen keyboard, requiring a physical keyboard is quite impractical

ghollingworth, 2024-06-06:
We've been looking into squeekboard on Wayland:

sudo apt install squeekboard
sudo apt install dconf-editor

and then go into org/gnome/desktop/a11y/applications and turn “screen-keyboard-enabled” on

It seems to work in Firefox, but Chromium doesn't seem to work at the moment...  Usual problem, GTK doesn't draw the text entry box so squeekboard doesn't know you are trying to enter keystrokes.  Will need some looking into...

steveiliop56, 2024-08-12:
@BlackRockCity you may be interested in my previous comment:

> Nope but I have found a solution I am happy with. I created a small menu item to open the keyboard with a button. Just open the menu editor and create a new item (I created it in help). Then for the [...]

steveiliop56, 2024-08-12:
Oh didn't know that, I will give it a try!

nguyenlamlll, 2024-09-02:
Hi @BlackRockCity, how do you get `squeekboard` to work with Firefox in kiosk mode?
I've tried running `squeekboard` on one terminal and `firefox—-kiosk "my.web" on another terminal. But then the rendering got really buggy/frozen, like in the picture below.
[image]

steveiliop56, 2024-09-02:
That's a lot of cursors.

thisisgregm, 2024-10-10:
I've just figured it out that I can install OS Lite and add XFCE (which I already us on other Linux machines, so I'm familiar with it anyway), add OnBoard keyboard and other stuff I need. Luckily XFCE still uses Xorg, not Wayland.
I don't know why I didn't think about it before.
This is the way for everyone who desperately needs OnBoard keyboard..

steveiliop56, 2025-05-14:
By the way raspberry pi os now has a built-in on screen keyboard for touchscreens so, the best idea would be to use that.

lurch, 2025-05-14:
> By the way raspberry pi os now has a built-in on screen keyboard for touchscreens so, the best idea would be to use that.

Indeed; and we've made a bunch of tweaks to it too, so that it better integrates into Raspberry Pi OS https://github.com/raspberrypi-ui/squeekboard/commits/master