raspberrypi/bookworm-feedback #15: Double-click doesn't work with the Touch Display

lurch, 2023-08-25:
E.g. if you launch the file-manager you can't double-click on a folder to open it.

lurch, 2023-10-10:
Still a problem under Wayland on a Pi4 with the latest version of Bookworm.

steveiliop56, 2023-10-12:
That's a problem for a long time now. The problem is that your finger doesn't click the exact same problem in the double tap so the screen registers 2 clicks on different places...

lurch, 2023-12-19:
> Do you know if this is enabled https://wayland.freedesktop.org/libinput/doc/latest/tapping.html?

That'd be a question for @spl237 to answer, but he's on holiday until next year. But I just tried the various tap / drag gestures described on that page, and none of them appear to be working in the default Wayfire desktop on Bookworm.

steveiliop56, 2023-12-19:
I think you have to enable something in the when building it...

Jebonicus, 2024-06-20:
This is still an issue on my fresh Pi5/Bookworm/Wayland installation.