raspberrypi/bookworm-feedback #16: Windows open "too big" on the Touch Display

lurch, 2023-08-25:
E.g. if you boot a Pi4 with Bookworm and **just** a Raspberry Pi Touch Display connected (no HDMI monitor), and then tap on the file-manager icon in the menu-bar, the window opens much bigger than the actual screen, which means that the menubar and half the toolbar aren't even visible.

njhollinghurst, 2023-10-06:
What we should have is a minimum supported screen size for the full GUI, and test against it!  I suggest it should be one of:
- 800x480 (official 7" display)
- 640x480 (traditional VGA monitor)

although some TV Out modes are even shorter than this (432 lines for Pi5 TV Out in the overscan-safe NTSC mode)

steveiliop56, 2023-10-11:
I have the same problem too...

maroonmoon13, 2023-10-12:
The same applies for Software center.

steveiliop56, 2023-10-12:
So what do we do? Is there a fix? That's an extremely important problem...

maroonmoon13, 2023-10-12:
The touch screen experience is quite poor in general. Not just big windows, I cannot even double click. So no way to open any files etc.

maroonmoon13, 2023-10-12:
looks like someone just opened a ticket: https://github.com/raspberrypi/bookworm-feedback/issues/15

steveiliop56, 2023-10-12:
That's an other problem... Guys we need a fix for this... I personally can't upgrade only for this reason.

ghollingworth, 2023-10-12:
@steveiliop56 @maroonmoon13 
None of these comments are useful to the developers.  If you have useful input, for example, reproduction scenarios (where it's not obvious), that's fine but otherwise you're just creating unnecessary email traffic for people who are trying to work on this software.  That is why I'm about to delete lots of comments, don't get annoyed it's just to try and keep the discussion focused on the issues.

spl237, 2023-10-12:
It would be useful to know how the same apps look on the same screens using Openbox and X rather than Wayland - use the Advanced Options menu in raspi-config to switch.

In theory, GTK and the window manager will negotiate the size of any window with the WM informing GTK of the maximum size it can have, which would usually be the screen size. If this mechanism is working correctly with Wayfire, then there will be no difference between the size of windows under X and under Wayfire. But it is possible that Wayfire is not doing this negotiation correctly resulting in windows being bigger under Wayfire than they are under X.

If windows are too big on both X and on Wayfire (which I suspect will be the case, but I could be wrong) then there is nothing that can be done about this other than individual applications being redesigned to put fewer widgets in their windows...

lurch, 2023-10-12:
I'm not back to the office until Monday, but IIRC when I did the same thing under X on a Pi 4 on the touchscreen (in Bookworm), they **did** make themselves the same size as the touchscreen.

steveiliop56, 2023-10-13:
The problem seems to happen to the pcamanfm and nothing else. I tried recommend software good, Raspberry pi configuration menu good, both browsers good the only slightly off display was the terminal app with the border being just a bit bigger that the panel... What other apps should I test? Also does Wayland have the alt+space function that allowed you to move the window outside of the screen (for example the hidden appearance menu I had to press alt+space and move it around to tap thr ok on the screen)?

spl237, 2023-10-13:
You can move a window in Wayfire by holding down the Windows key on your keyboard and dragging anywhere on the window.

I can confirm that with my monitor here set to 800x600, the only application that doesn't seem to size to the window is pcmanfm. My guess is that the problem is caused by pcmanfm including the default window size as a config parameter, and this overrides what the window manager tells it to do. That ought to be fixable, if I can find the relevant part of the code!

steveiliop56, 2023-10-15:
Ok I tested from my pios 11 with wayland enabled from the advanced menu of raspi-config (I know its not exactly the same) but the pcmanfm window opens just fine...

spl237, 2023-10-16:
I have confirmed that if a window explicitly requests a default size, X constrains the actual size with the screen dimensions and Wayfire doesn't.

I have a (slightly hacky but workable) fix for this in the file manager - I basically check the screen dimensions and if they are smaller than the default window size, I shrink the window accordingly. This is complicated by the taskbar and window title, but I have something that works pretty well and which will certainly be better than the current behaviour.

At some point I will need to go through the GTK backend code and try to see if it is possible to apply the same screen resolution constraint in the Wayland backend, which is probably the right thing to do, but this will work for now...

spl237, 2023-10-16:
I'll push the fix out later today, btw.

lurch, 2023-10-16:
Yup, the file-manager is now behaving much better on the touchscreen. But the "Screen Layout Editor" program is still opening taller than the touchscreen-size. Is it possible to "hack" that too, as I imagine that's a program you're likely to be running if you have both a touchscreen and HDMI monitor attached!

spl237, 2023-10-17:
> Yup, the file-manager is now behaving much better on the touchscreen. But the "Screen Layout Editor" program (EDIT: on a Pi 5, if that makes a difference) is still opening taller than the touchscreen-size. Is it possible to "hack" that too, as I imagine that's a program you're likely to be running if you have both a touchscreen and HDMI monitor attached!

I don't see that here. With my monitor set to 800 x 600, the screen layout editor takes up no more than around 10% of the screen - see attached screenshot. It seems to automatically scale itself for me already.
[image]

spl237, 2023-10-17:
Are you running with two monitors of different resolutions attached? It may be that it is using the size of the larger monitor to determine its size, and then displaying on the smaller screen.

lurch, 2023-10-17:
> Are you running with two monitors of different resolutions attached?

Yes, you're right. With _just_ the DSI screen connected, I get this:
[image]

But if I have both the DSI screen and an HDMI monitor attached, then I get this on the touchscreen:
[image]
I can then use the Windows-key+mouse-drag trick so that the resize-corners are visible; and the Screen Layout Editor window can be resized _larger_ than that, but it's **impossible** to make the window-height small enough so that it actually fits on the touchscreen display :confused:

6by9, 2023-10-17:
View menu and select 1:16?

spl237, 2023-10-17:
But you can still use all the functionality - the three buttons on the bottom bar are all in the menus as well - Layout has Apply and Undo, and File has Quit.

Plus the obvious solution is surely just to drag it onto the larger monitor, which will easily be big enough for it?

The problem is that the app is sizing itself to cope with the entire set of monitors, which includes the larger non-touchscreen - so it is going to be big, unavoidably. I don't think in this situation that resizing for a large window arrangement so that it fits on the smallest screen is the right thing to do - or even likely to be possible in some circumstances.

The only potential fix that occurs to me is to try to persuade the window manager to always launch it on the largest screen, if that is possible - I'll have a look.

lurch, 2023-10-17:
> View menu and select 1:16?

Ooh, will try that tomorrow.

> Plus the obvious solution is surely just to drag it onto the larger monitor, which will easily be big enough for it?

Sure, I was just reporting what I saw in case there was an "easy fix".

> The problem is that the app is sizing itself to cope with the entire set of monitors, which includes the larger non-touchscreen - so it is going to be big, unavoidably.

Hmm, but the (minimum) application window height is **much** taller than the combined height of both the `DSI-2` and `HDMI-A-1` screens. Do you happen to know how it chooses the minimum window-size?

> The only potential fix that occurs to me is to try to persuade the window manager to always launch it on the largest screen, if that is possible - I'll have a look.

That behaviour might be even _more_ confusing than the current behaviour? Especially as most "dual screen" setups will be using two HDMI monitors, rather than an HDMI monitor and a DSI touchscreen :wink:

steveiliop56, 2023-12-01:
Guys the pcmanfm is still way to big. Tested on brand new install.

spl237, 2023-12-01:
Make sure your image is fully up to date - the image available for download does not have the fix for the file manager windows in it - you will need to fully update using the wizard, the update icon on the taskbar, or manually from apt.

MateuszKusmierz, 2024-10-14:
Still not fixed?
[image]

Raspberry Pi 4B 4GB
Raspberry Pi 7" Touch Screen

Raspberry Pi OS with desktop and recommended software (+ all updates as per 14.10.2024)
Release date: July 4th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)
SHA ca4954b2173ad3a9b18bbe4a913b2efc8937ebfa6902bf6b423e3ff6702c9f2a

steveiliop56, 2024-10-14:
It is kind of fixed but sometimes it lags out and the windows appear like this. Most of the times it's good for me.