raspberrypi/bookworm-feedback #17: Wayland doesn't work on composite video output

lurch, 2023-08-25:
Console (KMS) mode works on composite video, but Wayland just gives a blank screen.

popcornmix, 2023-08-25:
It's a general problem currently that wayland rejects interlaced modes (so if a hdmi style display only supported 1080i50, it would also fail).
I believe there is a workaround by specifying a custom mode? @njhollinghurst may know how.

njhollinghurst, 2023-08-25:
Yes, @6by9 suggested this. It's only a stopgap as you have to get all the timings right; and `arandr` will merrily delete the custom mode line given half a chance.

njhollinghurst, 2023-10-03:
Pi5 does support some half-height progressive modes, so there will be a picture of sorts; unfortunately it has a silly aspect ratio and is too small for our GUI to function properly.

Perhaps the custom mode line(s) can be put into some global script to protect them from deletion by the configuration tool?
Unfortunately the exact mode specifications are different for Pi1-4 and Pi5, and may also vary with TV standard.

njhollinghurst, 2023-10-05:
https://github.com/raspberrypi-ui/wlroots/pull/1 should address this.

lurch, 2025-01-09:
I've just checked and @njhollinghurst says that this should be working now, so closing this issue.

closed 2025-01-09 by lurch