raspberrypi/bookworm-feedback #20: When using the X11 backend, screen-settings sometimes get "forgotten"

lurch, 2023-08-25:
Running the armhf version of Bookworm on a Pi2 (so defaulting to the X11 backend instead of Wayland). If I change the screen resolution and / or screen rotation using the "Screen Configuration" program and then click Apply, the setting is correctly applied immediately. But if I then reboot, that change I just made is "sometimes" forgotten about, and a previous setting will be applied. So there's obviously some combination-of-actions that determines why it does or doesn't forget the setting after a reboot, but I've not managed to figure out what that is yet.

spl237, 2023-08-28:
If you boot to desktop, the display manager runs and makes the relevant display settings. If you boot to console and then start the desktop with startx, the display manager doesn't run, so the display settings do not load. 

So it is probably that in one case you are booting to desktop, and in the other, you are booting to console and then starting the desktop.

You can check if the settings have saved correctly by looking in /usr/share/dispsetup.sh - this is the script called from the display manager to configure the display.

It is probably possible to add a call to this script somewhere in the startx sequence, but I don't think this is new behaviour - it should have been the same under bullseye.

lurch, 2023-08-28:
Nope, I was _always_ booting to desktop. Thanks for the pointer to the `dispsetup.sh` script, I'll have another poke at it tomorrow.

lurch, 2023-08-29:
Ahh, I think I've figured out what's going on here... when you set the screen resolution or rotation using the "Screen Configuration" program, the results are applied immediately. But they're only actually written out to `/usr/share/dispsetup.sh` when you close the "Screen Configuration" program. But if you click on the "Shutdown..." menu to reboot the Pi whilst the "Screen Configuration" program is still open, then `/usr/share/dispsetup.sh` **doesn't** get updated, and this is why I thought that the newly-applied settings were being "forgotten".

spl237, 2023-08-29:
That would make sense.

You need sudo to write to dispsetup.sh, so in case the user has disabled passwordless sudo, I thought it would be less annoying if they only ever got asked for the password once rather than potentially every time they made a change. So I only update that file once on every run, at the end.

lurch, 2023-08-29:
Okay, so this is a "known problem" if the user reboots before closing the "Screen Configuration" program?

spl237, 2023-08-29:
Yes, and not one which is fixable in any trivial fashion....

aallan, 2023-09-06:
>  Okay, so this is a "known problem" if the user reboots before closing the "Screen Configuration" program?

So we need a `NOTE` block in the screen configuration documentation warning the user not to reboot before closing the config program in the docs. Can you add that @lurch. At which point we can probably close this one.

lurch, 2023-09-07:
Done in https://github.com/raspberrypi/documentation/commit/b1b5cb5aeb342e590099f225e8c434ca575e98e2

closed 2023-09-07 by lurch