raspberrypi/bookworm-feedback #21: overlayroot package does work

CarlFK, 2023-08-26:
Steps: (see below for details)
1. sudo apt install overlayroot
2. enable overlayroot using any of the documented steps. 
like append overlayroot=tmpfs to /boot/cmdline.txt
3. reboot

observed: / mounted on /dev/root (/dev/mmcblk0p2)
expected: / mounted on overlayfs 

I think I know why this happens, and have posted a patch to address the root problem.
(to be fair, you have a choice of what is the problem; how raspios manages kernel versions, or how 'linux-version list' interacts with the way raspios manages kernel versions.)
An alternative (rework kernel packaging..) has been promised, it remains to be seen if the results addresses this bug.

The 'linux-version list'  problem an patch is here:
https://github.com/RPi-Distro/repo/issues/320
https://github.com/raspberrypi/bookworm-feedback/issues/8

details of installing / enabling:
```
pi@raspberrypi:~ $ sudo apt install overlayroot 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  cryptsetup cryptsetup-bin
Suggested packages:
  cryptsetup-initramfs haveged
The following NEW packages will be installed:
  cryptsetup cryptsetup-bin overlayroot
0 upgraded, 3 newly installed, 0 to remove and 2 not upgraded.
Need to get 694 kB of archives.
After this operation, 3,040 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://deb.debian.org/debian bookworm/main arm64 cryptsetup-bin arm64 2:2.6.1-4~deb12u1 [466 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 cryptsetup arm64 2:2.6.1-4~deb12u1 [212 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 overlayroot all 0.18.debian13 [15.4 kB]
Fetched 694 kB in 0s (2,644 kB/s)      
Preconfiguring packages ...
Selecting previously unselected package cryptsetup-bin.
(Reading database ... 51246 files and directories currently installed.)
Preparing to unpack .../cryptsetup-bin_2%3a2.6.1-4~deb12u1_arm64.deb ...
Unpacking cryptsetup-bin (2:2.6.1-4~deb12u1) ...
Selecting previously unselected package cryptsetup.
Preparing to unpack .../cryptsetup_2%3a2.6.1-4~deb12u1_arm64.deb ...
Unpacking cryptsetup (2:2.6.1-4~deb12u1) ...
Selecting previously unselected package overlayroot.
Preparing to unpack .../overlayroot_0.18.debian13_all.deb ...
Unpacking overlayroot (0.18.debian13) ...
Setting up cryptsetup-bin (2:2.6.1-4~deb12u1) ...
Setting up cryptsetup (2:2.6.1-4~deb12u1) ...
Setting up overlayroot (0.18.debian13) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for initramfs-tools (0.142) ...

pi@raspberrypi:~ $ 
```
```
pi@raspberrypi:~ $ head /etc/overlayroot.conf
# This is the overlayroot config file
# By default, overlayroot is not enabled.
# To enable overlayroot:
#   1) edit the 'overlayroot' definition below
#   2) reboot
#
# Supported values:
#  * overlayroot=tmpfs or overlayroot=tmpfs:PARAMETERS
#    write all changes to a temporary (ram only) backing device
#    A tmpfs mount will be created, and usable filesystem can
```
```
pi@raspberrypi:~ $ findmnt /
TARGET SOURCE         FSTYPE OPTIONS
/      /dev/mmcblk0p2 ext4   rw,noatime
```
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

XECDesign, 2023-09-22:
Should be resolved

lurch, 2023-10-09:
https://forums.raspberrypi.com/viewtopic.php?t=356399 says "Good news: This now works"

closed 2023-10-09 by lurch