raspberrypi/bookworm-feedback #22: Unable to sign-in if out of disk space

maxnet, 2023-08-27:
If out of disk space lightdm will not auto-login, and will prompt for password instead. After entering it, it does not work either, and it returns to login screen immediately without any message whatsover.

Going to a different virtual terminal to get shell, and deleting stuff there (or expanding file system) and rebooting does work.
But it is not clear to the average user that this is the problem.

Error reporting could be improved, and it would be nice if stuff was added to lightdm to pro-actively give a notification to the user if disk space on / is low when still signed in.
Instead of user finding out on the next boot he can no longer get into his system in a normal way.

lurch, 2023-08-27:
Yes, I also found this myself on Thursday (but forgot to log an issue for it, whoops).

spl237, 2023-08-28:
These issues are only arising due to the current lack of auto-expand for the file system; once that is fixed, this situation should be rare.

lurch, 2023-08-28:
...unless you do what I did and try to install **everything** in the "Recommended Software" application, which doesn't fit on a 16GB SD card even after you've expanded the filesystem :rofl:

spl237, 2023-08-28:
In theory, Recommended Software should warn about insufficient space, but unfortunately the calculation of how much space is going to be used is almost impossible to get accurately right, as PackageKit won't tell you the size of any dependencies. I did once try to show install sizes in the app, but gave up as the numbers bore very little relation to reality...

I really don't think there is much we can do about this once the file system is full; even if we were able to warn the user that the reason they couldn't log in was the disk being full, the instructions for how to recover from that state would be too complicated for most people to follow.

maxnet, 2023-08-28:
>I really don't think there is much we can do about this once the file system is full; even if we were able to warn the user that the reason they couldn't log in was the disk being full, the instructions for how to recover from that state would be too complicated for most people to follow.

Well, then they would at least know what is wrong, and could seek help from someone more knowledgeable in fixing it...
Or re-image if not succeeding in that.

Now people think they may have forgotten their password instead, as it returns to login screen.
And may waste an hour searching for a solution to that...

spl237, 2023-08-28:
Thinking about it, it might make more sense to check whether the disk is full when an install operation has completed in Recommended Software and to automatically undo the install which caused it, and to potentially do the same in Add / Remove Software.

Obviously that wouldn't help if someone runs out of disk space after an apt install though.

Now I think about it, it wouldn't be difficult to run a background process every 30 seconds or so as a panel plugin to monitor free space on the disk and to generate a warning notification when it was getting low - that might be a better plan all round. Given how difficult it is to recover from that situation if you aren't experienced in using the command line, I think prevention is probably better than cure.

lurch, 2023-08-28:
`sudo apt clean` can often free a large chunk of disk space, and is probably less disruptive than "automatically undo the install which caused it" ?

Just out of curiosity Simon, what part of the graphical-login-process is it that fails when the disk is 100% full?

spl237, 2023-08-28:
> sudo apt clean can often free a large chunk of disk space, and is probably less disruptive than "automatically undo the install which caused it" ?

The problem with that is that it isn't guaranteed to fix the problem, as it still might not free enough space for the package to install properly.

> Just out of curiosity Simon, what part of the graphical-login-process is it that fails when the disk is 100% full?

No idea - I'd have to pull the lightdm and greeter code apart to try and work that out. Something is presumably trying to write something to a file - possibly just a log file - which then hangs for long enough that lightdm decides the attempt to start the desktop has failed and aborts it.

lurch, 2023-08-28:
> The problem with that is that it isn't guaranteed to fix the problem

Sure - this is a problem that has no "guaranteed fix", only a bunch of heuristics which may or may not help :wink:

maxnet, 2023-08-28:
>No idea - I'd have to pull the lightdm and greeter code apart to try and work that out

Could just be the ~/.Xauthority file and /var/lib/lightdm/data/username.

If it turns out to be that, do note that you can also configure lightdm to put authority file in /var/run/lightdm which is  tmpfs (ram).

We are already doing so on Piserver setups: https://github.com/raspberrypi/piserver/blob/master/scripts/convert_chrooted_cmds#L101

Can also put /var/lib/lightdm in tmpfs (ram), either by adding an entry in /etc/fstab for that folder, like:
https://github.com/raspberrypi/piserver/blob/master/scripts/convert_chrooted_cmds#L40
Or create a systemd's /etc/tmpfiles.d/something.conf file that does the same ( https://www.freedesktop.org/software/systemd/man/tmpfiles.d.html )

spl237, 2023-08-28:
> > No idea - I'd have to pull the lightdm and greeter code apart to try and work that out
> 
> Could just be the ~/.Xauthority file.

Unlikely on Wayland - that file is only created once the desktop has started up, because I create it for Xwayland use in an autostart file.

lurch, 2023-10-10:
Sounds like there's not much that can be done here, so closing.

closed 2023-10-10 by lurch