raspberrypi/bookworm-feedback #23: No audio on Pi 400 with IIyama HDMI monitor

maxnet, 2023-08-27:
Do not get any audio output on Bookworm with a Pi 400 connected to a IIyama HDMI monitor.
Same hardware does work fine out-of-the-box with a (clean) Bullseye installation.

When running "speaker-test" (or other applications playing audio, like Chromium visiting Youtube, or Orca) the program playing audio seemingly runs fine, but no sound can be heard, and when querying the journal of pipewire it shows "snd_pcm_mmap_commit error: Broken pipe" lines appearing while the application playing audio is running.

```
$ journalctl --user -u pipewire -f

Aug 27 14:35:50 bookworm systemd[684]: Started pipewire.service - PipeWire Multimedia Service.
Aug 27 14:35:50 bookworm pipewire[1220]: mod.rt: Can't find org.freedesktop.portal.Desktop. Is xdg-desktop-portal running?
Aug 27 14:35:51 bookworm pipewire[1220]: spa.v4l2: '/dev/video14' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:35:51 bookworm pipewire[1220]: spa.v4l2: '/dev/video15' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:35:51 bookworm pipewire[1220]: spa.v4l2: '/dev/video21' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:35:51 bookworm pipewire[1220]: spa.v4l2: '/dev/video22' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:35:51 bookworm pipewire[1220]: spa.v4l2: '/dev/video14' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:35:51 bookworm pipewire[1220]: spa.v4l2: '/dev/video15' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:35:51 bookworm pipewire[1220]: spa.v4l2: '/dev/video21' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:35:51 bookworm pipewire[1220]: spa.v4l2: '/dev/video22' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:36:19 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:20 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:21 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:24 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:25 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:25 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:25 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:26 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:26 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:27 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:27 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:27 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:27 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:27 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:45 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:46 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:46 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:47 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:36:47 bookworm pipewire[1220]: spa.alsa: hdmi:0: snd_pcm_mmap_commit error: Broken pipe
^C
```

If I connect the HDMI display to the other HDMI port, same problem with hdmi:1...

```
$ journalctl --user -u pipewire -f      
Aug 27 14:37:52 bookworm systemd[678]: Started pipewire.service - PipeWire Multimedia Service.
Aug 27 14:37:52 bookworm pipewire[1228]: mod.rt: Can't find org.freedesktop.portal.Desktop. Is xdg-desktop-portal running?
Aug 27 14:37:53 bookworm pipewire[1228]: spa.v4l2: '/dev/video14' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:37:53 bookworm pipewire[1228]: spa.v4l2: '/dev/video15' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:37:53 bookworm pipewire[1228]: spa.v4l2: '/dev/video21' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:37:53 bookworm pipewire[1228]: spa.v4l2: '/dev/video22' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:37:53 bookworm pipewire[1228]: spa.v4l2: '/dev/video14' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:37:53 bookworm pipewire[1228]: spa.v4l2: '/dev/video15' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:37:53 bookworm pipewire[1228]: spa.v4l2: '/dev/video21' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:37:53 bookworm pipewire[1228]: spa.v4l2: '/dev/video22' VIDIOC_ENUM_FRAMEINTERVALS: Inappropriate ioctl for device
Aug 27 14:38:10 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:38:10 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:38:10 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:38:10 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:38:10 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:38:10 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:30 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:30 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:30 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:31 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:32 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:32 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:32 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:32 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:32 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:32 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:32 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:36 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:39 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:40 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:40 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
Aug 27 14:39:41 bookworm pipewire[1228]: spa.alsa: hdmi:1: snd_pcm_mmap_commit error: Broken pipe
```

spl237, 2023-08-27:
Try switching back to Pulseaudio (raspi-config->Advanced Options->Audio Config) and see if that helps.

popcornmix, 2023-08-29:
In case the issue is related to audio formats reported in edid, can you post output of:
`base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid`

(try card0 if card1 doesn't work).

maxnet, 2023-08-30:
```
$ base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
AP///////wAmzSBWbAYAACcWAQOAMBt4KoTVolpSoiYNUFS/7wDRwIGAlQCzAIFAcU+VDwEBAjqA
GHE4LUBYLEUA3QwRAAAeAAAA/QA3TB5TEQAKICAgICAgAAAA/wAxMTE3MVYyOTAxNjQ0AAAA/ABQ
TDIyODBICiAgICAgAQYCAx/xTAECAwQFEBESExQeHyMJBwGDAQAAZQMMABAAjArQiiDgLRAQPpYA
3QwRAAAYAR0AclHQHiBuKFUA3QwRAAAejArQkCBAMSAMQFUA3QwRAAAYAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAATg==
$ edid-decode  /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
edid-decode (hex):

00 ff ff ff ff ff ff 00 26 cd 20 56 6c 06 00 00
27 16 01 03 80 30 1b 78 2a 84 d5 a2 5a 52 a2 26
0d 50 54 bf ef 00 d1 c0 81 80 95 00 b3 00 81 40
71 4f 95 0f 01 01 02 3a 80 18 71 38 2d 40 58 2c
45 00 dd 0c 11 00 00 1e 00 00 00 fd 00 37 4c 1e
53 11 00 0a 20 20 20 20 20 20 00 00 00 ff 00 31
31 31 37 31 56 32 39 30 31 36 34 34 00 00 00 fc
00 50 4c 32 32 38 30 48 0a 20 20 20 20 20 01 06

02 03 1f f1 4c 01 02 03 04 05 10 11 12 13 14 1e
1f 23 09 07 01 83 01 00 00 65 03 0c 00 10 00 8c
0a d0 8a 20 e0 2d 10 10 3e 96 00 dd 0c 11 00 00
18 01 1d 00 72 51 d0 1e 20 6e 28 55 00 dd 0c 11
00 00 1e 8c 0a d0 90 20 40 31 20 0c 40 55 00 dd
0c 11 00 00 18 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 4e

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: IVM
    Model: 22048
    Serial Number: 1644
    Made in: week 39 of 2012
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 48 cm x 27 cm
    Gamma: 2.20
    DPMS levels: Off
    RGB color display
    First detailed timing is the preferred timing
  Color Characteristics:
    Red  : 0.6347, 0.3515
    Green: 0.3212, 0.6328
    Blue : 0.1513, 0.0517
    White: 0.3134, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    Apple   :   640x480    66.666667 Hz   4:3     35.000 kHz     30.240000 MHz
    DMT 0x05:   640x480    72.808802 Hz   4:3     37.861 kHz     31.500000 MHz
    DMT 0x06:   640x480    75.000000 Hz   4:3     37.500 kHz     31.500000 MHz
    DMT 0x08:   800x600    56.250000 Hz   4:3     35.156 kHz     36.000000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x0a:   800x600    72.187572 Hz   4:3     48.077 kHz     50.000000 MHz
    DMT 0x0b:   800x600    75.000000 Hz   4:3     46.875 kHz     49.500000 MHz
    Apple   :   832x624    74.551266 Hz   4:3     49.726 kHz     57.284000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
    DMT 0x11:  1024x768    70.069359 Hz   4:3     56.476 kHz     75.000000 MHz
    DMT 0x12:  1024x768    75.028582 Hz   4:3     60.023 kHz     78.750000 MHz
    DMT 0x24:  1280x1024   75.024675 Hz   5:4     79.976 kHz    135.000000 MHz
  Standard Timings:
    DMT 0x52:  1920x1080   60.000000 Hz  16:9     67.500 kHz    148.500000 MHz
    DMT 0x23:  1280x1024   60.019740 Hz   5:4     63.981 kHz    108.000000 MHz
    DMT 0x2f:  1440x900    59.887445 Hz  16:10    55.935 kHz    106.500000 MHz
    DMT 0x3a:  1680x1050   59.954250 Hz  16:10    65.290 kHz    146.250000 MHz
    DMT 0x20:  1280x960    60.000000 Hz   4:3     60.000 kHz    108.000000 MHz
[... 31 lines cut ...]
    VIC  19:  1280x720    50.000000 Hz  16:9     37.500 kHz     74.250000 MHz
    VIC  20:  1920x1080i  50.000000 Hz  16:9     28.125 kHz     74.250000 MHz
    VIC  30:  1440x576    50.000000 Hz  16:9     31.250 kHz     54.000000 MHz
    VIC  31:  1920x1080   50.000000 Hz  16:9     56.250 kHz    148.500000 MHz
  Audio Data Block:
    Linear PCM:
      Max channels: 2
      Supported sample rates (kHz): 48 44.1 32
      Supported sample sizes (bits): 16
  Speaker Allocation Data Block:
    FL/FR - Front Left/Right
  Vendor-Specific Data Block (HDMI), OUI 00-0C-03:
    Source physical address: 1.0.0.0
  Detailed Timing Descriptors:
    DTD 2:   720x480    59.940060 Hz   3:2     31.469 kHz     27.000000 MHz (477 mm x 268 mm)
                 Hfront   16 Hsync  62 Hback   60 Hpol N
                 Vfront    9 Vsync   6 Vback   30 Vpol N
    DTD 3:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz (477 mm x 268 mm)
                 Hfront  110 Hsync  40 Hback  220 Hpol P
                 Vfront    5 Vsync   5 Vback   20 Vpol P
    DTD 4:   720x576    50.000000 Hz   5:4     31.250 kHz     27.000000 MHz (477 mm x 268 mm)
                 Hfront   12 Hsync  64 Hback   68 Hpol N
                 Vfront    5 Vsync   5 Vback   39 Vpol N
Checksum: 0x4e
```