raspberrypi/bookworm-feedback #28: Outdated GitHub.cli version

paulober, 2023-09-10:
The GitHub CLI version in this distribution is from February since then there have been about 10 bigger updates.
```
gh version 2.23.0 (2023-02-27 Debian 2.23.0+dfsg1-1)
https://github.com/cli/cli/releases/tag/v2.23.0
```

alexellis, 2023-09-13:
I'm not sure how the images are built, but I work on an OSS package manager called [arkade](https://arkade.dev). It could be used in the CI job or debootstrap to get the latest version with `arkade get gh`

@paulober if you wanted to update this and 140 other similar tools, it's worth checking out.

lurch, 2023-09-13:
> I'm not sure how the images are built

https://github.com/RPi-Distro/pi-gen  (but I'm not sure if the public repo contains the `bookworm` changes yet)

popcornmix, 2023-09-13:
Isn't this a debian issue rather than RPiOS?
I imagine you are getting an unmodified version from debian:
```
$ apt policy gh
gh:
  Installed: (none)
  Candidate: 2.23.0+dfsg1-1
  Version table:
     2.23.0+dfsg1-1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
```
which matches https://packages.debian.org/stable/utils/gh

XECDesign, 2023-09-22:
Yup, it's not one of ours.

closed 2023-10-09 by lurch