raspberrypi/bookworm-feedback #30: raspi-config do_configure_keyboard error on Lite

gitbls, 2023-09-12:
raspi-config shouldn't try to copy the wayfire stuff if wayfire isn't installed (as on Lite):

```
bash$ sudo raspi-config do_configure_keyboard us nonit
Reloading keymap. This may take a short while
cp: cannot stat '/etc/wayfire/gtemplate.ini': No such file or directory
grep: /etc/wayfire/greeter.ini: No such file or directory
/bin/raspi-config: 563: cannot create /etc/wayfire/greeter.ini: Directory nonexistent
grep: /etc/wayfire/greeter.ini: No such file or directory
sed: can't read /etc/wayfire/greeter.ini: No such file or directory
grep: /etc/wayfire/greeter.ini: No such file or directory
sed: can't read /etc/wayfire/greeter.ini: No such file or directory
grep: /etc/wayfire/greeter.ini: No such file or directory
sed: can't read /etc/wayfire/greeter.ini: No such file or directory
grep: /etc/wayfire/greeter.ini: No such file or directory
sed: can't read /etc/wayfire/greeter.ini: No such file or directory
```

XECDesign, 2023-09-12:
@spl237

gitbls, 2023-09-12:
It's possible to have a system with no users defined (pi user removed, no other users added), so this code may want to check whether the user exists before updating anything that's SUDO_USER specific.

lurch, 2023-09-12:
Huh? If you don't have any users defined, how would you log in to run `raspi-config` ?

gitbls, 2023-09-12:
> Huh? If you don't have any users defined, how would you log in to run `raspi-config` ?

No need to login. It runs as part of a script that runs during the first system boot.

spl237, 2023-09-12:
> > Huh? If you don't have any users defined, how would you log in to run `raspi-config` ?
> 
> No need to login. It runs as part of a script that runs during the first system boot.

The pi user exists during first system boot. It is the template which is changed to the user's requested username as part of the initial config process, but the system always has at least one user.

spl237, 2023-09-12:
What is more interesting is that you are only showing warnings from the second part of the operation, and not from the first. If you do not have any of the wayfire-specific packages installed, I would expect both the files /etc/wayfire/template.ini and /etc/wayfire/gtemplate.ini to be missing, and hence for this code, which does the same thing to both, to complain about both. So I am a little puzzled that it is only complaining about one of them. Did you not post the entire log, or is this all that was reported?

gitbls, 2023-09-12:
While it's true that the pi user exists in the distributed img, my tool (https://github.com/gitbls/sdm) does customization of the img before it boots. I agree that a system with no users is not super-interesting, but such a system can be created, either with sdm or simply editing the img with chroot or systemd-nspawn, before booting.

If it's not a stated requirement, then at least do_configure_keyboard should allow operation without a user.

I'm concerned about this particular setting because doing a do_configure_keyboard inside a systemd-nspawn doesn't work on Bookworm. It used to work fine on Bullseye, but something changed in core Debian causing it to fail, so I defer setting it until the first boot.

If you decide that you're going to assume a user account must be defined, I can live with that. But the wayfire errors are annoying, so I hope these can be fixed.

Lastly, in the log that I provided, there was a user defined, so no errors from the first part of the operation. Sorry for the confusion on that. My comment about 'no user' was from a thought exercise, not a real, demonstrated problem (although I could, if that is useful). Here's the whole log section:
```
Sep 12 05:49:08 p83 root[922]: FirstBoot: System time synchronization achieved
Sep 12 05:49:08 p83 root[928]: FirstBoot: Run Generated FirstBoot Script '/etc/sdm/0piboot/010-configure-keyboard.sh'
Sep 12 05:49:08 p83 sdm-firstboot[932]: Reloading keymap. This may take a short while
Sep 12 05:49:09 p83 sdm-firstboot[1171]: cp: cannot stat '/etc/wayfire/template.ini': No such file or directory
Sep 12 05:49:09 p83 sdm-firstboot[1172]: chown: cannot access '/home/bls/.config/wayfire.ini': No such file or directory
Sep 12 05:49:09 p83 sdm-firstboot[1173]: grep: /home/bls/.config/wayfire.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1182]: cp: cannot stat '/etc/wayfire/gtemplate.ini': No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1183]: grep: /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[932]: /usr/bin/raspi-config: 563: cannot create /etc/wayfire/greeter.ini: Directory nonexistent
Sep 12 05:49:10 p83 sdm-firstboot[1184]: grep: /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1185]: sed: can't read /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1186]: grep: /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1187]: sed: can't read /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1188]: grep: /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1189]: sed: can't read /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1190]: grep: /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 sdm-firstboot[1191]: sed: can't read /etc/wayfire/greeter.ini: No such file or directory
Sep 12 05:49:10 p83 systemd[1]: keyboard-setup.service: Deactivated successfully.
Sep 12 05:49:10 p83 systemd[1]: Stopped keyboard-setup.service - Set the console keyboard layout.
Sep 12 05:49:10 p83 systemd[1]: Stopping keyboard-setup.service - Set the console keyboard layout...
Sep 12 05:49:10 p83 systemd[1]: Starting keyboard-setup.service - Set the console keyboard layout...
Sep 12 05:49:11 p83 systemd[1]: Finished keyboard-setup.service - Set the console keyboard layout.
Sep 12 05:49:12 p83 root[1387]: FirstBoot: Run Generated FirstBoot Script '/etc/sdm/0piboot/020-trim-enable.sh'
Sep 12 05:49:12 p83 root[1391]: FirstBoot: Enable trim
```

Thx

spl237, 2023-09-12:
OK, that log makes more sense.

As to operation without a user, this will affect far more of raspi-config than just this function, and it has never been a design consideration. Making the whole script work in that scenario would be a lot of work and would benefit very few people, so it's not likely to be a high priority.

Fixing this particular issue is trivial; I'll do so tomorrow.

gitbls, 2023-09-12:
> Fixing this particular issue is trivial; I'll do so tomorrow.

Thanks! The 'no user` scenario is super low priority from my perspective. I mentioned it for completeness, not as a significant concern. Apologies for the red herring.

XECDesign, 2023-09-22:
I think this has been resolved.

gitbls, 2023-09-22:
I think so too.

closed 2023-09-22 by gitbls