raspberrypi/bookworm-feedback #31: `chromium-browser --kiosk` display anomaly

JinShil, 2023-09-18:
Step to reproduce:
1.  Use 2023-09-11 Bookworm Beta OS image
2. Boot to desktop
3. Open lxterminal
4. Run the command `chromium-browser --kiosk https://raspberrypi.com`
5. Observe that the browser appears to get reduced to a single-pixel horizontal line at the top of the screen.  The desktop panel is all black, and the desktop and lxterminal window are still displayed

lurch, 2023-09-19:
@tobybroberts I think you also ran into this at some point - do you remember what the fix was?

JinShil, 2023-10-15:
I can confirm that `--start-maximized` alone, without `--start-fullscreen` works around the issue.

But why is this necessary, and why does it work fine without those arguments when launching from and SSH terminal session?

ferdiamg, 2023-10-16:
Wayland/Wayfire are super restrictive when it comes to SSH commands changing apps state. No idea what mechanism intercepts the correct maximized window behaviour, however I do these from an autostart.sh now anyways which always runs them from the Pi itself. 

I am puzzled on why it even works for you through SSH as most commands result in "$DISPLAY not found" or similar error messages because the command is being run from your main machine and not the PI itself. Thinking about it now.. maybe it works because your main machine is having a correct setup that runs the --start-maximized under the hood already and when run on your Pi directly its missing and you need to state it explicitly.. Just a thought though.

ferdiamg, 2023-10-16:
> I can confirm that `--start-maximized` alone, without `--start-fullscreen` works around the issue.
> 
> But why is this necessary, and why does it work fine without those arguments when launching from and SSH terminal session?

Wayland/Wayfire are super restrictive when it comes to SSH commands changing apps state. No idea what mechanism intercepts the correct maximized window behaviour, however I do these from an autostart.sh now anyways which always runs them from the Pi itself.

I am puzzled on why it even works for you through SSH as most commands result in "$DISPLAY not found" or similar error messages because the command is being run from your main machine and not the PI itself. Thinking about it now.. maybe it works because your main machine is having a correct setup that runs the --start-maximized under the hood already and when run on your Pi directly its missing and you need to state it explicitly.. Just a thought though.

popcornmix, 2023-10-16:
I've heard this issue is resolved with 0.8.0 release of wayfire.
(Although the bump introduces new issues, so may not appear imminently).

JinShil, 2023-11-28:
As of today, with the latest updates to Bookworm, this is still an issue.

Guruabyss, 2024-02-22:
As of today Bookworm V12 still has this issue!

popcornmix, 2024-02-22:
I've tested:
```chromium-browser --kiosk https://raspberrypi.com```
On pi5 bookwork with labwc enabled (this is planned as a default update "soon").

The browser opened fullscreen just fine.
So I suspect this issue will be resolved in a coming update.

Zebra353, 2024-03-22:
On Pi4 64bit OD release 2024-03-15 still an issue.

PHCPartsAlexis, 2024-06-27:
Can confirm the same behaviour still on the latest updates today. Adding --start-maximised was the workaround.

niccoreyes, 2024-07-02:
I can also confirm as of July 2, 2024 , "chromium-browser --start-maximized --start-fullscreen" fixes the issue on wayland as a workaround

--start-fullscreen alone fails to render chromium properly in wayland/wayfire mode, but works alrite on X11 mode

ashah7, 2024-11-24:
to get around this in X11 I use matchbox-window-manager, but no fix for wayland

XECDesign, 2025-04-14:
Can't reproduce the problem with the provided steps. Can re-open if it's still an issue and steps to reproduce the issue can be provided.

closed 2025-04-14 by XECDesign