raspberrypi/bookworm-feedback #33: crypto journal error logged on a Pi0W

gitbls, 2023-09-23:
I booted an unmodified image_2023-09-22-raspios-bookworm-armhf-lite.img on a Pi0W:
```
Sep 21 17:27:12 p0w kernel: Booting Linux on physical CPU 0x0
Sep 21 17:27:12 p0w kernel: Linux version 6.1.0-rpi2-rpi-v6 (debian-kernel@lists.debian.org) (gcc-12 (Raspbian 12.2.0-14+rpi1) 12.2.0, GNU ld (GNU Binutils for Raspbian)>
Sep 21 17:27:12 p0w kernel: CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Sep 21 17:27:12 p0w kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Sep 21 17:27:12 p0w kernel: OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
```

Part way through the system boot (after creating user, setting keymap, and rebooting a few times) this is in the system journal during every boot:
```
Sep 21 17:27:13 p0w kernel: Loading compiled-in X.509 certificates
Sep 21 17:27:13 p0w kernel: Freeing initrd memory: 10120K
Sep 21 17:27:13 p0w kernel: ------------[ cut here ]------------
Sep 21 17:27:13 p0w kernel: WARNING: CPU: 0 PID: 1 at crypto/rsa-pkcs1pad.c:540 pkcs1pad_verify+0x134/0x154
Sep 21 17:27:13 p0w kernel: Modules linked in:
Sep 21 17:27:13 p0w kernel: CPU: 0 PID: 1 Comm: swapper Not tainted 6.1.0-rpi2-rpi-v6 #1  Raspbian 1:6.1.47-1+rpt2
Sep 21 17:27:13 p0w kernel: Hardware name: BCM2835
Sep 21 17:27:13 p0w kernel:  unwind_backtrace from show_stack+0x18/0x1c
Sep 21 17:27:13 p0w kernel:  show_stack from dump_stack_lvl+0x34/0x68
Sep 21 17:27:13 p0w kernel:  dump_stack_lvl from __warn+0x78/0xf4
Sep 21 17:27:13 p0w kernel:  __warn from warn_slowpath_fmt+0x74/0xd8
Sep 21 17:27:13 p0w kernel:  warn_slowpath_fmt from pkcs1pad_verify+0x134/0x154
Sep 21 17:27:13 p0w kernel:  pkcs1pad_verify from public_key_verify_signature+0x26c/0x38c
Sep 21 17:27:13 p0w kernel:  public_key_verify_signature from x509_check_for_self_signed+0xc4/0x104
Sep 21 17:27:13 p0w kernel:  x509_check_for_self_signed from x509_cert_parse+0x160/0x1cc
Sep 21 17:27:13 p0w kernel:  x509_cert_parse from x509_key_preparse+0x18/0x1a4
Sep 21 17:27:13 p0w kernel:  x509_key_preparse from asymmetric_key_preparse+0x54/0x88
Sep 21 17:27:13 p0w kernel:  asymmetric_key_preparse from key_create_or_update+0x194/0x448
Sep 21 17:27:13 p0w kernel:  key_create_or_update from x509_load_certificate_list+0x84/0xf4
Sep 21 17:27:13 p0w kernel:  x509_load_certificate_list from do_one_initcall+0x50/0x1d0
Sep 21 17:27:13 p0w kernel:  do_one_initcall from kernel_init_freeable+0x174/0x1f8
Sep 21 17:27:13 p0w kernel:  kernel_init_freeable from kernel_init+0x1c/0x11c
Sep 21 17:27:13 p0w kernel:  kernel_init from ret_from_fork+0x14/0x34
Sep 21 17:27:13 p0w kernel: Exception stack(0xdc821fb0 to 0xdc821ff8)
Sep 21 17:27:13 p0w kernel: 1fa0:                                     00000000 00000000 00000000 00000000
Sep 21 17:27:13 p0w kernel: 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sep 21 17:27:13 p0w kernel: 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Sep 21 17:27:13 p0w kernel: ---[ end trace 0000000000000000 ]---
Sep 21 17:27:13 p0w kernel: Problem loading in-kernel X.509 certificate (-22)
Sep 21 17:27:13 p0w kernel: Loaded X.509 cert 'Debian Secure Boot Signer 2022 - linux: 14011249c2675ea8e5148542202005810584b25f'
```
I have not had repeatable success connecting the WiFi to my network, although if it does connect it's eventually stable.

The rest of the time it seems to get `auth_failure` errors, and sometimes eventually connects, and sometimes not. This is with `iwconfig wlan0 power off` (seemed much flakier with power on), and `nmcli ipv6.method disabled` on the connection. It seemed flakier with ipv6 enabled, but my testing was not rigorously scientific, I only have so much patience for a ZeroW 🙄

pelwell, 2023-09-23:
Have you tried any Bullseye images, and if so, is Bookworm clearly worse?

Try with `over_voltage=2`?

gitbls, 2023-09-23:
To be clear, I'm not stuck trying to make this work. i was doing some testing of something else and decided I should try it on a Pi0W. When I saw the crypto error I backed off to using a fully vanilla IMG and the crypto error was still there.

After I posted the above, I booted one of my customized IMGs on the 0W and interestingly, at least for this boot of the system, the network seems stable.

So, it may be that the network connection issues are a red herring, but the crypto error seems highly repeatable, at least on my only Pi0W.

Thx

gitbls, 2023-09-23:
Ugh, I spoke too soon about the network. I let it run for a couple of hours, and it appears to have disconnected and reconnected every 6-15 minutes.

DHCP/DNS are isc-dhcp-server and bind9 running on a Bullseye-powered Pi4. WiFi is eero Pro, and the closest eero is about 5 feet from the Pi0W

I did another test AFTER generating the log below. First, I moved the Pi0W so it's only about 3 feet from the eero instead of 5 feet. After that I rebooted to get a clean test. Still seeing similar behavior with a DISCONNECT followed by a reconnect every 5-10 minutes.

I also tried over_voltage=2, same behavior.

Here's the log from `journalctl | grep -E "NetWorkManager|wpa_supplicant"`. The system finished booting at about 13:10, and the first disconnection occurred at 13:37:43
```
Sep 23 13:09:43 p0w systemd[1]: Starting NetworkManager.service - Network Manager...
Sep 23 13:09:47 p0w NetworkManager[400]: <info>  [1695499787.7517] NetworkManager (version 1.42.4) is starting... (boot:b3775482-b762-47a7-ad99-f56459355d79)
Sep 23 13:09:47 p0w NetworkManager[400]: <info>  [1695499787.7986] Read config: /etc/NetworkManager/NetworkManager.conf (lib: no-mac-addr-change.conf)
Sep 23 13:09:48 p0w NetworkManager[400]: <info>  [1695499788.1473] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Sep 23 13:09:48 p0w systemd[1]: Started NetworkManager.service - Network Manager.
Sep 23 13:09:48 p0w NetworkManager[400]: <info>  [1695499788.5128] manager[0x7de0f8]: monitoring kernel firmware directory '/lib/firmware'.
Sep 23 13:09:48 p0w NetworkManager[400]: <info>  [1695499788.5388] monitoring ifupdown state file '/run/network/ifstate'.
Sep 23 13:09:48 p0w systemd[1]: Starting NetworkManager-wait-online.service - Network Manager Wait Online...
Sep 23 13:09:48 p0w dbus-daemon[343]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.2' (uid=0 pid=400 comm="/usr/sbin/NetworkManager --no-daemon")
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.2975] hostname: hostname: using hostnamed
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.2979] hostname: static hostname changed from (none) to "p0w"
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.3177] dns-mgr: init: dns=default,systemd-resolved rc-manager=symlink (auto)
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.3535] rfkill0: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/20300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill0) (driver brcmfmac)
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.3580] manager[0x7de0f8]: rfkill: Wi-Fi hardware radio set enabled
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.3597] manager[0x7de0f8]: rfkill: WWAN hardware radio set enabled
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.5399] Loaded device plugin: NMTeamFactory (/usr/lib/arm-linux-gnueabihf/NetworkManager/1.42.4/libnm-device-plugin-team.so)
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.5602] Loaded device plugin: NMWifiFactory (/usr/lib/arm-linux-gnueabihf/NetworkManager/1.42.4/libnm-device-plugin-wifi.so)
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.5669] Loaded device plugin: NMAtmManager (/usr/lib/arm-linux-gnueabihf/NetworkManager/1.42.4/libnm-device-plugin-adsl.so)
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.6476] Loaded device plugin: NMWwanFactory (/usr/lib/arm-linux-gnueabihf/NetworkManager/1.42.4/libnm-device-plugin-wwan.so)
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.6727] Loaded device plugin: NMBluezManager (/usr/lib/arm-linux-gnueabihf/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.6892] manager: rfkill: Wi-Fi enabled by radio killswitch; enabled by state file
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.6960] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.7024] manager: Networking is enabled by state file
Sep 23 13:09:53 p0w dbus-daemon[343]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.2' (uid=0 pid=400 comm="/usr/sbin/NetworkManager --no-daemon")
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.7243] settings: Loaded settings plugin: ifupdown ("/usr/lib/arm-linux-gnueabihf/NetworkManager/1.42.4/libnm-settings-plugin-ifupdown.so")
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.7264] settings: Loaded settings plugin: keyfile (internal)
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.7280] ifupdown: management mode: unmanaged
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.7284] ifupdown:       interface-parser: parsing file /etc/network/interfaces
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.7287] ifupdown:       interface-parser: source line includes interfaces file(s) /etc/network/interfaces.d/*
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.7293] ifupdown: interfaces file /etc/network/interfaces.d/* doesn't exist
Sep 23 13:09:53 p0w NetworkManager[400]: <info>  [1695499793.7295] ifupdown:       interface-parser: finished parsing file /etc/network/interfaces
[... cut ...]
Sep 23 14:48:44 p0w dbus-daemon[343]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.2' (uid=0 pid=400 comm="/usr/sbin/NetworkManager --no-daemon")
Sep 23 14:48:44 p0w systemd[1]: Starting NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service...
Sep 23 14:48:45 p0w systemd[1]: Started NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service.
Sep 23 14:48:55 p0w systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
```