raspberrypi/bookworm-feedback #35: rc_gui: tooltip incorrectly mentions RealVNC

andrum993, 2023-09-24:
Just spotted a bug in rc_gui (the GUI-based Raspberry Pi Configuration tool). On the `Interfaces` tab, if you hover over the `VNC` switch it shows a tooltip which says `Turn on to enable remote access using RealVNC`. This is on a Pi 4B running Wayland, so it won't be using RealVNC. Probably `using RealVNC` should be removed to make it generic.

andrum993, 2023-09-28:
This has now been corrected in the English language version of the rc_gui UI: https://github.com/raspberrypi-ui/rc_gui/commit/ba4354cf99c8e190c5641630c4909311ff95cc5a

Translations have not been updated yet, and the `msgid` should probably be changed when they are, to avoid any confusion.

lurch, 2023-10-10:
Fixed, but not yet pushed.

closed 2023-10-10 by lurch