raspberrypi/bookworm-feedback #37: Today's update pulled in 77 new packages !

gitbls, 2023-09-28:
Presumably this was intentional, but definitely a surprise!
``` 
p84/usr/local/bin$ sudo apt list --upgradable
Listing... Done
dphys-swapfile/testing,testing 20100506-7.1+rpt2 all [upgradable from: 20100506-7.1+rpt1]
libcamera-apps-lite/testing 1.2.1-1+bookworm+1 arm64 [upgradable from: 1.2.1-1+bookworm]
libcamera-ipa/testing 0.1.0+pi5~0-1 arm64 [upgradable from: 0.0.5~83+bde9b04f-3]
libegl-mesa0/testing 23.2.0~rc3-1+rpt3 arm64 [upgradable from: 22.3.6-1+rpt2+deb12u1]
libgbm1/testing 23.2.0~rc3-1+rpt3 arm64 [upgradable from: 22.3.6-1+rpt2+deb12u1]
libgl1-mesa-dri/testing 23.2.0~rc3-1+rpt3 arm64 [upgradable from: 22.3.6-1+rpt2+deb12u1]
libglapi-mesa/testing 23.2.0~rc3-1+rpt3 arm64 [upgradable from: 22.3.6-1+rpt2+deb12u1]
libglx-mesa0/testing 23.2.0~rc3-1+rpt3 arm64 [upgradable from: 22.3.6-1+rpt2+deb12u1]
libraspberrypi-bin/testing 1:2+git20230913~095424+44a3953-1 arm64 [upgradable from: 1:2+git20230420~133613+cc1ca18-1]
libraspberrypi-dev/testing 1:2+git20230913~095424+44a3953-1 arm64 [upgradable from: 1:2+git20230420~133613+cc1ca18-1]
libraspberrypi-doc/testing,testing 1:2+git20230913~095424+44a3953-1 all [upgradable from: 1:2+git20230420~133613+cc1ca18-1]
libraspberrypi0/testing 1:2+git20230913~095424+44a3953-1 arm64 [upgradable from: 1:2+git20230420~133613+cc1ca18-1]
linux-image-rpi-v8/testing 1:6.1.54-1+rpt1 arm64 [upgradable from: 1:6.1.47-1+rpt2]
linux-libc-dev/testing 1:6.1.54-1+rpt1 arm64 [upgradable from: 1:6.1.47-1+rpt2]
python3-gpiozero/testing,testing 2.0-1 all [upgradable from: 2.0~0-1]
python3-libcamera/testing 0.1.0+pi5~0-1 arm64 [upgradable from: 0.0.5~83+bde9b04f-3]
python3-picamera2/testing,testing 0.3.12+pi5-1 all [upgradable from: 0.3.12-2]
raspberrypi-sys-mods/testing 20230922 arm64 [upgradable from: 20230921~rc1]
raspi-config/testing,testing 20230927 all [upgradable from: 20230919]
raspi-firmware/testing,testing 1:1.20230829+ds-1~rpt3 all [upgradable from: 1:1.20230829+ds-1~rpt2]
rpi-eeprom/testing,testing 18.4-1 arm64 [upgradable from: 17.0-1+bookworm]
p84~$ sudo apt upgrade --yes
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libcamera0.0.5
Use 'apt autoremove' to remove it.
The following NEW packages will be installed:
  libass9 libavc1394-0 libavcodec59 libavdevice59 libavfilter8 libavformat59 libavutil57 libbluray2 libboost-log1.74.0 libboost-thread1.74.0 libbs2b0 libcaca0
  libcamera0.1 libcdio-cdda2 libcdio-paranoia2 libcdio19 libchromaprint1 libcjson1 libcodec2-1.0 libdc1394-25 libdecor-0-0 libflite1 libgme0 libgsm1 libiec61883-0
  libjack-jackd2-0 liblilv-0-0 libmbedcrypto7 libmysofa1 libnorm1 libopenal-data libopenal1 libopenmpt0 libpgm-5.3-0 libpisp-common libpisp0.0.1 libplacebo208
  libpocketsphinx3 libpostproc56 librabbitmq4 libraw1394-11 librist4 librubberband2 libsdl2-2.0-0 libserd-0-0 libshine3 libsnappy1v5 libsndio7.0 libsord-0-0 libsoxr0
  libspeex1 libsphinxbase3 libsratom-0-0 libsrt1.5-gnutls libssh-gcrypt-4 libswresample4 libswscale6 libtheora0 libtwolame0 libudfread0 libva-drm2 libva-x11-2 libva2
  libvdpau1 libvidstab1.1 libvorbisfile3 libvulkan1 libx264-164 libxss1 libxvidcore4 libzimg2 libzmq5 libzvbi-common libzvbi0 linux-image-6.1.0-rpi4-rpi-v8
  ocl-icd-libopencl1 python3-av
The following packages will be upgraded:
  dphys-swapfile libcamera-apps-lite libcamera-ipa libegl-mesa0 libgbm1 libgl1-mesa-dri libglapi-mesa libglx-mesa0 libraspberrypi-bin libraspberrypi-dev
  libraspberrypi-doc libraspberrypi0 linux-image-rpi-v8 linux-libc-dev python3-gpiozero python3-libcamera python3-picamera2 raspberrypi-sys-mods raspi-config
  raspi-firmware rpi-eeprom
21 upgraded, 77 newly installed, 0 to remove and 0 not upgraded.
```

XECDesign, 2023-09-28:
That's mostly pi5 support and general fixes.

It looks like you were using an older 'lite' image with python3-picamera2 installed. It now needs ffmpeg, which ends up dragging in a lot of new dependencies. Since these dependencies are quite heavy, newer images are built without libcamera-apps or picamera2 pre-installed in the 'lite' images.

gitbls, 2023-09-28:
Thanks. Guess I wasn't aware of the older IMG having python3-picamera2 installed.

Yes, this system is my Bookworm "production" system and is running the 09/11 IMG. Obviously needs a rebuild with the newest IMG!

closed 2023-09-28 by gitbls