raspberrypi/bookworm-feedback #39: WiFi created using GUI fails to connect if no monitor attached / not logged into GUI

gitbls, 2023-10-05:
I built a bookworm desktop IMG and booted it up with a monitor attached. I then used the GUI to add my network connection. All good.

I then disconnected the monitor, unplugged the ethernet and rebooted.

After the reboot, the WiFi never came online.

While it was still offline I plugged in the monitor and logged in.

As soon as I logged in, bam! The WiFi came online.

I tried starting the system, monitor connected, but didn't login. Again, WiFi did not connect to the network.

I then removed the `permissions` line from the created .nmconnection file, unplugged the monitor and rebooted.

The WiFi connected correctly even though I hadn't logged in.

**Conclusion:** having the `permissions` statement in there prevents WiFi from starting until the GUI is logged in. This means that anyone using the GUI to define a WiFi network connection on Bookworm will find that they subsequently cannot connect to WiFi if they subsequently boot the Pi  and don't login on the desktop.

This precludes the system from being accessible on the network, even (e.g., SSH and other inbound services will all fail).

This is effectively what was reported in https://forums.raspberrypi.com/viewtopic.php?t=357230, but with a reproducible scenario.

lurch, 2023-10-05:
@nathan-contino Did you run into anything like this whilst writing all the networking documentation that you were working on recently?

spl237, 2023-10-06:
This should be fixed in the latest update to wf-panel-pi and lxplug-netman.

gitbls, 2023-10-06:
Confirmed fixed.

closed 2023-10-06 by gitbls