raspberrypi/bookworm-feedback #40: Wrong keyboard/char mapping via VNC

DeeJay, 2023-10-08:
Applications started from the Raspberry Pi Desktop GUI Applications menu have a mismatched keyboard/display character mapping. 

Prior background discussion in forum: [https://forums.raspberrypi.com/viewtopic.php?t=357390](https://forums.raspberrypi.com/viewtopic.php?t=357390)

User **any1** (wayvnc developer?) writes there -

> In short, the desired keyboard layout needs to be communicated to wayvnc using any of the available options: --keyboard command line option, xkb_layout config option, or the XKB_DEFAULT_LAYOUT environment variable.

spl237, 2023-10-09:
Done - added to new version of raspi-config so that this is added to the wayvnc command line.

ghollingworth, 2023-10-10:
Confirmed in the forums

closed 2023-10-10 by ghollingworth