raspberrypi/bookworm-feedback #41: RealVNC on 32-bit RasPiOS issue

gitbls, 2023-10-08:
No, it's not the Wayland VS x11 issue. On ARMHF-Desktop, even after switching to X11, starting service vncserver-x11-serviced gets an error in the system log:
```
/usr/bin/vncserver-x11-core: error while loading shared libraries: libmmal.so: cannot open shared object file: No such file or directory
```
Curiously this doesn't occur on the 64-bit desktop version.

CleoQc, 2023-10-08:
This error means that I cannot connect through VNC, either with RealVNC Viewer or TigerVNC viewer.

andrum993, 2023-10-08:
Link to forum discussion: https://forums.raspberrypi.com/viewtopic.php?t=357404

ghollingworth, 2023-10-10:
We need RealVNC to rebuild the 32bit package in the same way as the way they've rebuilt the 64bit package.  We've messaged them and are awaiting a response.

quitmeyer, 2023-10-11:
Dangg, i got this problem too. Picked the wrong day to start up a RPI headless project. hahhah

quitmeyer, 2023-10-11:
Dang, i can't read the above forum post because it says i'm not authorized. 

I'm not sure this is the same issue, but since i upgraded to bookworm today, I can no longer log in via RealVNC viewer for headless use. It gives me the error:

"Unable to connect to VNC Server using your chosen security setting.
Either upgrade VNC Server to a more recent version from RealVNC, or
select a weaker level of encryption."

everything worked fine two days ago when I was using bullseye

popcornmix, 2023-10-11:
@quitmeyer I've not tested this, but this post may be relevant.
https://forums.raspberrypi.com/viewtopic.php?p=2143201#p2143201

gitbls, 2023-10-11:
> Dang, i can't read the above forum post because it says i'm not authorized.
>
> I'm not sure this is the same issue, but since i upgraded to bookworm today, I can no longer log in via RealVNC viewer [...]

The post that you can't access is the discussion leading up to this issue. Did you read the Blog? (https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/), which says:

"
There is also currently a regression with the RealVNC server itself, which is still used in Bookworm on older Raspberry Pi models which don’t yet run Wayland. The 64-bit version works fine, but the 32-bit version is not currently compatible with Bookworm — we are waiting for an update from RealVNC. In the meantime, if you do need remote desktop access on a Raspberry Pi Zero, 1, 2, or 3, we recommend sticking with Bullseye for the time being.
"

quitmeyer, 2023-10-12:
This is very helpful you all, thanks for this information. I tried the things from the link of both commenting out the enable_auth and generating a key, but neither worked for me on my 64x bookworm. Im gonna have to rollback to bullseye.  :)

Arhum2, 2023-10-12:
Hello, im also having this issue when launching vnc-server on my pi using 
`vncserver-x11` or `vncserver-virtual`, I had it running just fine but reflashed raspbian on my pi and now it doesn't seem to work or open on startup even though its installed and enabled within the pi;s settings

mehulboricha, 2023-10-15:
Experiencing the same issue, I also tried tigervnc and switching to Wayfire from raspi-config if that matters.

loaderladdy, 2023-10-16:
Is this a Raspberry Pi Firmware issue?. Because the only place I can see reference to the file "libmmal.so" is in Raspberry Pi's GitHub repository for the Pi's firmware. [https://github.com/raspberrypi/firmware/tree/master/opt/vc/lib](url)

Has the latest firmware code not been compiled correctly?

Dennis1993, 2023-10-16:
I don‘t know if I‘m right here: I‘ve the problem, that if I activate VNC in raspi-config the icon in the taskbar is not shown. I can‘t set a password or something else. I downloaded the latest version by raspiberry pi imager in 64Bit (lite without recommended tools).

I switched back to the version before and vnc is working fine.

user514, 2023-10-16:
> I switched back to the version before and vnc is working fine.

...sorry, I don't understand right your last phrase. Not sure if it works for you with 64b, etc.

I will try to summarize what I understand by comments on this thread and my own experience. Correct me if I'm wrong:

   -RASPI-4: vnc works through wayland protocol (autodetected raspi4 on installation).
       -In that case, you should use tigervnc to connect which supports that protocol (wayland). (realvnc client does not work).

   -RASPI-3 (and below): vnc protocol is switches to old x11 by raspbian, but...
       -64b: works fine, connecting with realvnc from a windows machine for example. Same as the old times.
       -32b: does not work. Error with 'libmmal.so' as in the first post. (I have this same problem). Not realvnc or tigervnc connecting.

loaderladdy, 2023-10-16:
@user514 I have raised a case with RealVNC and sent screen prints of the libmmal.so error message.  They are aware of the issues and are investigating them and hope to have a fix for 32bit PIs soon.

On their web site they say they are working on a Wayland version of their VNC server next year, which is great news.

user514, 2023-10-16:
> to have a fix for 32bit PIs soon

  Thanks loaderladdy, great to know. We will wait for the fix. Anyone, any news, post them here!, thanks!

doneforaiur, 2023-10-25:
I successfully installed version 7.7.0 of RealVNC Server and got it to work properly.

vaknin34, 2023-10-25:
How do you install it? and do you use Bookworm 32bit?

doneforaiur, 2023-10-26:
@vaknin34 oh, my bad. It was version 6.11.0. This is the script I used;

```
#!/bin/bash

mkdir realvncserversetup && cd realvncserversetup
echo Add armhf arch
sudo dpkg --add-architecture armhf && sudo apt update

echo Download and install RealVNC server
wget -nv https://downloads.realvnc.com/download/file/vnc.files/VNC-Server-6.11.0-Linux-ARM.deb && sudo apt install ./VNC-Server-6.11.0-Linux-ARM.deb

echo Download and install dependencies
files=( libbcm_host.so libvcos.so libmmal.so libmmal_core.so libmmal_components.so \
        libmmal_util.so libmmal_vc_client.so libvchiq_arm.so libvcsm.so libcontainers.so )
for i in "${files[@]}"
do
        wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/$i
done
sudo mv *.so /usr/lib/
cd ..

echo Enable VNC service on boot
sudo systemctl enable vncserver-x11-serviced.service
echo Start VNC service
sudo systemctl start vncserver-x11-serviced.service
``` 

I'm on 32-bit Bookworm;

`Linux raspberrypi 6.1.0-rpi4-rpi-v7 #1 SMP Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv7l GNU/Linux`

loaderladdy, 2023-10-26:
hi @doneforaiur i updated your script to install VNC-Server-&.0.0-Linux-ARM.deb because it also works now that someone clever i.e., you 😁 👍 has found and provided a method to install those pesky VC scripts.

OS: Raspbian GNU/Linux 12 (bookworm) armv7l 
Host: Raspberry Pi 2 Model B Rev 1.1 
Kernel: 6.1.0-rpi4-rpi-v7 
Uptime: 9 days, 19 hours, 10 mins 
Packages: 1884 (dpkg) 
Shell: bash 5.2.15 
Terminal: /dev/pts/0 
CPU: BCM2835 (4) @ 1.000GHz 
Memory: 213MiB / 921MiB

loaderladdy, 2023-10-26:
@user514 @doneforaiur /i updated my case on RealVNC Help to reference doneforaiur’s solution.

It’ll be interesting to see how RealVNC’s devs fix this considering the VC libraries where apparently deprecated in Bullseye.

JamesH65, 2023-10-26:
> @user514 @doneforaiur /i updated my case on RealVNC Help to reference doneforaiur’s solution.
> 
> It’ll be interesting to see how RealVNC’s devs fix this considering the VC libraries where apparently deprecated in Bullseye.

They will need to redo their backend to support Wayland/DRM. Quite a bit of work to do, but this change has been coming for some time not just on PiOS. The VC libraries need dispmanx support in the firmware, and this is not present on the Pi5, so there is not option going forward except supporting Wayland.

loaderladdy, 2023-10-26:
RealVNC's help desk replied to my case as follows

- [Thanks for sharing this. While that script restores the missing libraries to allow VNC Server to start, as you have noted Raspberry Pi OS have actually deprecated (and in the latest Pi hardware, removed) the libraries and hardware they interfaced with. We will be updating RealVNC Server to remove the dependency on the libraries, as the function they enabled is not supported by Raspberry Pi OS anymore. ] 

They have published that they are working on support for Wayland in H1 next year [Raspberry Pi 5, Bookworm and RealVNC Connect](https://help.realvnc.com/hc/en-us/articles/14110635000221-Raspberry-Pi-5-Bookworm-and-RealVNC-Connect)

pelwell, 2023-10-26:
Your quote is not of something written by Raspberry Pi - you're whingeing up the wrong tree. As stated above:

> We need RealVNC to rebuild the 32bit package in the same way as the way they've rebuilt the 64bit package. We've messaged them and are awaiting a response.

ryanrolds, 2023-10-27:
A better question maybe how did 32bit Bookworm pass QA with a broken VNC server? If it's not a critical Raspbian component, maybe it should be removed from `raspi-config` and other Raspberry PI configuration UIs.

CHN-STUDENT, 2023-11-03:
@doneforaiur  Thanks! It works well for me on my pi4! By the way, the vnc newest version is 7.7.0.

lucas-bernardino, 2023-11-04:
I was having problems when accessing the vnc viewer to control my raspberry pi and i found this article: https://picockpit.com/raspberry-pi/tigervnc-and-realvnc-on-raspberry-pi-bookworm-os/ where i was able to control my raspberry py using a vnc viewer alternative called tigervnc. I just follow the tutorial (setting up the tigervnc in the raspberry py and in the pc) and it worked flawlessly.

user514, 2023-11-04:
> control my raspberry py using a vnc viewer alternative called tigervnc.

  Yes. This is an option, but if you read my conclusions post up in this thread...: https://github.com/raspberrypi/bookworm-feedback/issues/41#issuecomment-1764868667

> -RASPI-4: vnc works through wayland protocol (autodetected raspi4 on installation).
>-In that case, you should use tigervnc to connect which supports
> that protocol (wayland). (realvnc client does not work).

  Not good for other options (raspi3, 32b version, etc...)

  ...Which OS and raspberry model have you successfully connect?...

lucas-bernardino, 2023-11-04:
> ...Which OS and raspberry model have you successfully connect?...

I am using the Raspberry Pi OS (64 Bit) in a Raspberry Pi 4 B

user514, 2023-11-04:
> I am using the Raspberry Pi OS (64 Bit) in a Raspberry Pi 4 B

  Thanks for sharing. That match my conclusions...

MohanRaj1110, 2023-11-10:
I'm using the same version of OS on raspberry pi
link:https://github.com/raspberrypi/bookworm-feedback/issues/41
In loop command this link is not working for me.

MohanRaj1110, 2023-11-10:
> @vaknin34 oh, my bad. It was version 6.11.0. This is the script I used;
>
> ```
> #!/bin/bash
>
> mkdir realvncserversetup && cd realvncserversetup
> echo Add armhf arch
> sudo dpkg --add-architecture armhf && [...]

I'm using the same version of OS on raspberry pi
link:https://github.com/raspberrypi/bookworm-feedback/issues/41
In loop command this link is not working for me.

dabble53, 2023-11-20:
FWIW - Starting with a clean bookworm install on Pi 3B+ (32-bit), all I did was download the list of files given by vaknin34 and copy them to /usr/lib
(I did download the VNC-Server-6.11.0-Linux-ARM.deb file, but did not install it.)
Restarted the server "sudo systemctl restart vncserver-x11-serviced.service" and I can now connect with my regular Windows-11 vnc viewer (RealVNC 6.21.406 64-bit)

user514, 2023-11-21:
> bookworm install on Pi 3B+ (32-bit), all I did was download the list of files given by vaknin34 and copy them to /usr/lib

  Wow!. Thanks to dabble53 and vaknin34.

  To help others to clarify, this are my steps (also a bookworm install on Pi 3B (32-bit), and not installing anything, just copying those 10 files:

```
  # mkdir ~/delete_after_move
  # cd ~/delete_after_move

  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libbcm_host.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libvcos.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libmmal.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libmmal_core.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libmmal_components.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libmmal_util.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libmmal_vc_client.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libvchiq_arm.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libvcsm.so
  # wget -nv https://github.com/raspberrypi/firmware/raw/master/opt/vc/lib/libcontainers.so

  # sudo mv ~/delete_after_move/*.so /usr/lib/
  # cd ..
  # rmdir ~/delete_after_move

  # sudo systemctl restart vncserver-x11-serviced.service
```

  ...try to connect from windows 8.1 using Realvnd v7.7.0... and... WORKING!!!.
Thank you guys...

martin762, 2023-12-01:
Just want to say thank you.
Just added the *.so list to a bookworm fresh install on a Raspberrypi zero(W) (arm6 32 bit) and the fix works well. Real VNC server works!

Shindenfudoryu, 2023-12-02:
To make it work with my raspberry pi 4 I just update it using these two commands
sudo apt-get update
sudo apt-get upgrade

It was quite a headache to find it

lurch, 2023-12-02:
> sudo apt-get update
> sudo apt-get upgrade
> 
> It was quite a headache to find it

I don't know why you found that hard to find, we provide those instructions at https://www.raspberrypi.com/documentation/computers/os.html#updating-and-upgrading-raspberry-pi-os

Shindenfudoryu, 2023-12-02:
> > sudo apt-get update
> > sudo apt-get upgrade
> > It was quite a headache to find it
> 
> I don't know why you found that hard to find, we provide those instructions at https://www.raspberrypi.com/documentation/computers/os.html#updating-and-upgrading-raspberry-pi-os

I am not talking about the commands themselves but to make my raspberry pi recognized by VNC server I had to run these commands which I did by chance and then my VNC was working

lurch, 2023-12-03:
> I am not talking about the commands themselves but to make my raspberry pi recognized by VNC server I had to run these commands which I did by chance and then my VNC was working

As it says in the page that I linked to: "It’s important to keep your Raspberry Pi up to date. ... The second reason, related to the first, is that the software you are running on your device most certainly contains bugs. ... By keeping your software up to date, you are lowering the chances of hitting these bugs." :slightly_smiling_face:

reukiodo, 2024-02-09:
These files seem no longer in the master branch and need to be downloaded from oldstable branch:
```
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libbcm_host.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libvcos.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libmmal.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libmmal_core.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libmmal_components.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libmmal_util.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libmmal_vc_client.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libvchiq_arm.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libvcsm.so
wget -nv https://github.com/raspberrypi/firmware/raw/oldstable/opt/vc/lib/libcontainers.so
```

MarkHodson, 2024-02-09:
Thanks to vaknin34 and reukiodo.

I am now able to use realvnc to view my pi.

Hardware: Pi4 4Gb 32bit Raspberry Pi OS 
Image version: 2023-12-05-raspios-bookworm-armhf.img
Client: Windows 11.  RealVNC Viewer 7.9.0

### Summary of procedure...
1. Run with fresh image and let the installation run the operating system updates.
2. Enable X11.  In a terminal run `sudo raspi-config` go to `Advanced Options` -> `Wayland` -> `X11`.  Allow the pi to reboot.
3. Download the missing shared libraries..
`cd /usr/lib`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libbcm_host.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libvcos.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libmmal.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libmmal_core.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libmmal_components.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libmmal_util.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libmmal_vc_client.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libvchiq_arm.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libvcsm.so`
`sudo wget -nv https://github.com/raspberrypi/firmware/raw/stable/opt/vc/lib/libcontainers.so`
4. Enable the interface VNC.  Use either `raspi-config` or the `Preferences → Raspberry Pi Configuration` from the start button.  Enable the interface VNC
5. Connect with RealVNC viewer from client computer

This works for me.

reukiodo, 2024-02-10:
I didn't need to use raspi-config to switch to X11, at least on my RPi2 (with DSI display) - I haven't yet tried on my RPi4. To be fair, VNC just worked on my CM3+ (with HDMI display) without needing to download separate libs or change to X11.

reukiodo, 2024-02-28:
Just paste the commands from https://github.com/raspberrypi/bookworm-feedback/issues/41#issuecomment-1935492585 into your RPi 3b+ 's terminal, then enable VNC and reboot?

albfan, 2024-02-28:
This issue shows a error detecting target. raspberry pi offers vnc with wayland protocol and realvnc needs to deal with that, as tigervnc does.

So this issue is really for RealVNC and not for rpi

loaderladdy, 2024-02-28:
@jr3us 

The good news is that with the release of RealVNC Server 7.10.0 RealVNC now works on 32-bit RaspberryPiOS Bookworm without the legacy raspios Bullseye modules.

[RealVNC Release Notes](https://help.realvnc.com/hc/en-us/articles/360002253138-Release-Notes)

loaderladdy, 2024-02-28:
> This issue shows a error detecting target. raspberry pi offers vnc with wayland protocol and realvnc needs to deal with that, as tigervnc does.
> 
> 
> 
> So this issue is really for RealVNC and not for rpi

True @albfan, and RealVNC recognises the need to embrace Wayland support so they are the default VNC provider for RaspiOS in the future

[RealVNC Statement on RPi5 Bookworm and RealVNC Connect](https://help.realvnc.com/hc/en-us/articles/14110635000221-Raspberry-Pi-5-Bookworm-and-RealVNC-Connect)

lurch, 2024-07-11:
> The good news is that with the release of RealVNC Server 7.10.0 RealVNC now works on 32-bit RaspberryPiOS Bookworm

Closing this issue as resolved. Raspberry Pi OS Bookworm has RealVNC Server 7.11.0.18 available.

closed 2024-07-11 by lurch