raspberrypi/bookworm-feedback #42: VNC server (wayvnc) can be used only once, then exits

DeeJay, 2023-10-10:
The wayvnc server running on the RasPiOS host can be accessed only once. On quitting the workstation client Viewer app the wayvnc session on the host also terminates

To reproduce.
Fresh install of current RasPiOS 64 Desktop beta. Update/full-upgrade. Enable VNC via raspi-config. Reboot.

After reboot, check wayvnc server listening -

```
david@rpbw0928full64ssk:~ $ sudo netstat -tnlp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      830/cupsd
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      843/sshd: /usr/sbin
tcp        0      0 0.0.0.0:5900            0.0.0.0:*               LISTEN      1091/wayvnc
tcp        0      0 127.0.0.1:6010          0.0.0.0:*               LISTEN      1524/sshd: david@pt
tcp6       0      0 ::1:6010                :::*                    LISTEN      1524/sshd: david@pt
tcp6       0      0 :::22                   :::*                    LISTEN      843/sshd: /usr/sbin
tcp6       0      0 ::1:631                 :::*                    LISTEN      830/cupsd
david@rpbw0928full64ssk:~ $
```
Establish VNC Desktop login via TigerVNCViewer.

From the Applications ('Start) Menu - Shutdown -> Logout

Recheck network listener status on RasPiOS:

```
david@rpbw0928full64ssk:~ $ sudo netstat -tnlp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      830/cupsd
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      843/sshd: /usr/sbin
tcp        0      0 127.0.0.1:6010          0.0.0.0:*               LISTEN      1524/sshd: david@pt
tcp6       0      0 ::1:6010                :::*                    LISTEN      1524/sshd: david@pt
tcp6       0      0 :::22                   :::*                    LISTEN      843/sshd: /usr/sbin
tcp6       0      0 ::1:631                 :::*                    LISTEN      830/cupsd
david@rpbw0928full64ssk:~ $
```
wayvnc server is no longer running.

Posted for forum confirmation: [https://forums.raspberrypi.com/viewtopic.php?t=357489](https://forums.raspberrypi.com/viewtopic.php?t=357489)

spl237, 2023-10-10:
"From the Applications ('Start) Menu - Shutdown -> Logout"

That's going to kill the server, because you lose the user credentials when the user logs out. It will restart when the desktop restarts. I don't think you can log out of a running session and have it continue to run.

DeeJay, 2023-10-10:
Thanks for the thought Simon.

I'll need to go back to a non-Wayland OS platform to see if this is a change of behaviour.

My initial reaction is that this is unexpected.

DeeJay, 2023-10-10:
So maybe PEBKAC, in the form of inappropriate expectations, that might be managed by Documentation?

I have confirmed, as you have, that vncserver-x11 on a Buster system works as I expected: that the Server is started by the OS and is 'persistent.'

I had seen no indication that wayvnc would be run as the logged-in user, far less that it would exit when a particular remote session is logged out.

(as an aside, I've just created a second user on the host RasPiOS Bookworm OS, and TigerVNC won't let me create a remote session as that user. The error is 'invalid pixel format')

lurch, 2023-10-10:
Simon, does this mean that wayvnc can't be used if you've setup your Pi to boot to Desktop but *not* auto-login?

DeeJay, 2023-10-10:
> does this mean that wayvnc can't be used if you've setup your Pi to boot to Desktop but not auto-login?

And what are the implications for access by a 'third party' who has valid /etc/passwd credentials on the host system but is not the logged-in Desktop user?

spl237, 2023-10-10:
Pass! Given the developer of Wayvnc is responding on the thread linked above, he'd probably be better placed to answer these questions than I am...

ghollingworth, 2023-10-10:
But yes, if you're not logged in then the computer cannot start a VNC session as you....

I don't know if there is any solution to this in a Wayland world...  In this world the window manager is run as the user to manage the windows of the currently running desktop and it is this window manager that WayVNC gives you a view into.  Since the window manager can't exist before the user is logged in then it seems WayVNC can't also give you access to it...

It's possible (The WayVNC develop any1 should have more information here...), that the WayVNC server could be a systemd service which provides login credentials to the lightdm system a little like the greeter does locally.  The WayVNC server could then be spawned off as that logged in user.

But that does sound like quite a lot of 'might' and 'possible'

lurch, 2023-10-10:
If a user has a Pi running "headless" with no logged-in-user, can they login over SSH and then start up whatever-it-is that's needed for them to then connect to a graphical desktop session over VNC?

spl237, 2023-10-10:
I think it's possible to start the desktop over SSH, but there needs to be some fiddling to make sure it knows what display to use, as the SSH session doesn't have the requisite environment variables set, IIRC.

lurch, 2023-10-10:
I think this is an "operating mode" (i.e. Pi is always headless, but connect to it over SSH and VNC) that a lot of people use?

DeeJay, 2023-10-10:
Certainly what AS has just described is my everyday use-case.

I have just checked the 'no user' scenario.

With wayvnc already enabled and proven to be accepting connections.

raspi-config Option B3 Desktop, require login.

TigerVNC Viewer on Win10 workstation can connect and complete user authentication and brings up a useable Desktop GUI rendered on the workstation.

Making my 'innocent mistake' of using Raspberry/Start -> Shutdown -> Logout still has the effect of killing the wayvnc instance and subsequent Viewer connections are rejected.

Using 'Red X/Close' to crash out of the Viewer leaves the wayvnc server running on the host.

any1, 2023-10-11:
You can run wayvnc however you like. There only needs to be a running wayland compositor which can be selected by setting the `WAYLAND_DISPLAY` environment variable appropriately. You can have it attach to a running session, or you can create a headless output and have it attach to that.

When you're logged into a graphical session, the `XDG_*` environment variables will be set, and the compositor will set the `WAYLAND_DISPLAY` variable for you, but there's nothing stopping you from putting the config file wherever you want and attaching to whatever wayland compositor your want, from outside.

spl237, 2023-10-11:
> There only needs to be a running wayland compositor which can be selected by setting the `WAYLAND_DISPLAY` environment variable appropriately.

Does this mean that you cannot start wayvnc before you have started the compositor? And does it mean that if the compositor stops running, wayvnc will also stop?

any1, 2023-10-11:
> Does this mean that you cannot start wayvnc before you have started the compositor? And does it mean that if the compositor stops running, wayvnc will also stop?

Yes.

spl237, 2023-10-11:
So in that case, it sounds as if the current behaviour will not be fixable - if we can only start wayvnc once wayfire is running, and logging out of wayfire will terminate wayvnc, then wayvnc will not remain active once a user session is logged out, as this terminates the wayfire session for that user and starts a different wayfire session for the greeter.

ghollingworth, 2023-10-11:
Would it be possible to use the WayVNC interface to actually log into the system?  i.e. do the LDAP (I'm saying this as someone who doesn't understand this stuff) authentication though a WayVNC systemd service? 

That would then (assuming that user doesn't already have a logged-in session), start a new display session.  In other words its kind of like a secondary method of doing the greeter?

Gordon

any1, 2023-10-11:
> Would it be possible to use the WayVNC interface to actually log into the system? i.e. do the LDAP (I'm saying this as someone who doesn't understand this stuff) authentication though a WayVNC systemd service?

Yeah, wayvnc has a control interface for that sort of thing, but we would have to make it possible for it to run detached from a compositor, which is currently not implemented.

DeeJay, 2023-10-11:
As the OP who kicked this off, can I offer a 'naive' non-technical viewpoint?

As things stand I think the implementation breaches the principle of 'Least Surprise.'

If I know there is a host on my network offering services via ssh I can connect to it, am prompted to authenticate, do some work, log out of my session, and the ssh server carries on running ready for the potential next user.

If I know that there is a host on my network offering services via VNC I can connect to it, am required to authenticate, do some some work, log out of my session, and the VNC server [and its compositor] stop working so that potential next users are locked out.

The difference in behaviour is what I regard as the end-user issue. It can be 'fixed' by Documentation that recommends NOT logging out. But it is still an accident waiting to happen.

any1, 2023-10-12:
I think that it would be good to have a detailed description of the desired behaviour.

notGMman, 2023-10-12:
The behaviour I've seen, using the latest Bookworm on a Pi 4b

1. Boot the pi with an attached screen. All fine. Wayvnc running. Remotely connect with TigerVNC, all good. 
2. HDMI switch the screen away to a new device. Pi is now headless, picks up new resolution, kills wayvnc.
3. Reconfigure pi headless setting to use the same resolution as the screen. Boot headless - can connect/disconnect remotely multiple times from either windows EXE client or linux java.
4. Switch HDMI-attached screen back to the pi - all remains fine.
5. Switch screen away again - wayvnc dies.

So, alas, not much of a solution for my use cases at the moment. I haven't checked if it copes with screen blanking - one assumes not. Luckily for me I've been using VNC for going on 30 years so I'm used to frustration. But it is very disappointing to go from an out-of-the-box solution that works on Buster/Bullseye to a not-even-documented solution on Bookworm. How does wayvnc actually get started on the desktop? Autostart config seems to have changed again.

DeeJay, 2023-10-12:
> > does this mean that wayvnc can't be used if you've setup your Pi to boot to Desktop but not auto-login?
> 
> And what are the implications for access by a 'third party' who has valid /etc/passwd credentials on the host system but is not the logged-in Desktop user?

I have now tested this and can answer my own question.

RasPiOS being run physically headless. raspi-config run to set Boot to Desktop + REQUIRE login. Enable VNC server. (RPi[5], so defaults to wayland/wayvnc.

TigerVNC client on Win10. Connect to RPi5, start VNC session using credentials for a 'secondary' user with an account in /etc/passwd. VNC session starts, displays RP 'fisherman' GUI.

Start LXTerminal. Run **whoami** Shell reports the userid of the 'primary' user defined by Raspberry Pi Imager at installation, NOT the secondary account used to login via VNC client.

This is NOT what I expected! It seems to give ALL users of the system access to the primary user account.

ghollingworth, 2023-10-12:
It is started from /etc/xdg/autostart/wayvnc.desktop

SteffenDuenner, 2024-01-25:
Yesterday I had an update to wayvnc: v0.8-rc0 and VNC is working reliably now.

felker, 2024-01-25:
I just tested the update from 
```
wayvnc/stable,now 0.8.0-rc0-1 arm64 [installed,automatic]
```
to
```
wayvnc/stable,now 0.8.0-rc0-2 arm64 [installed,automatic]
```
and I am now able to connect and disconnect to the VNC server without an HDMI output on my Raspberry Pi 5