raspberrypi/bookworm-feedback #43: Touchscreen not working in latest Bookworm image

lurch, 2023-10-10:
There's a kernel change needed ( https://github.com/raspberrypi/linux/pull/5622 ) in order to get the touchscreen-part of the Raspberry Pi Touch Display working, which isn't yet in the APT-packaged kernel. The display-part of the Raspberry Pi Touch Display still works fine.

As a temporary workaround, until the stable APT-packaged kernel gets updated, you can run `rpi-update` to get the latest bleeding-edge kernel.

steveiliop56, 2023-10-11:
It works now. I tested it.

XECDesign, 2023-10-12:
We haven't done anything on our end to fix it, so I don't think it would've magically started working.

willyvaessen, 2023-10-12:
I performed an rpi-update and indeed, the screen now responds to a touch. However, the touch functionality seems to be inverted after the first reboot following the update.
I had to set the orientation back to "normal" and then inverted it again to get it to perform as desired.

lurch, 2023-10-12:
On a Raspberry Pi 4, I've just tested all combinations of 32-bit / 64-bit Bookworm and DSI-only / DSI & HDMI, and in no situations was the "touch" of the touchscreen working.

pelwell, 2023-10-12:
Did you rpi-update? We know that the fix isn't in the current images.

steveiliop56, 2023-10-12:
Umm I have a dfrobot 7 inch dsi display and it works just fine...

lurch, 2023-10-12:
Please re-read the initial comment... "Raspberry Pi Touch Display" i.e. https://www.raspberrypi.com/products/raspberry-pi-touch-display/

steveiliop56, 2023-10-12:
Yeah I am just saying that something is wrong in the gpio communication part of the official screen I guess because in the screen I have that's only with a dsi cable it works...

SerdarAbali, 2023-10-13:
Official 7" display, the touch function does not work. I am here to acknowledge and confirm the problem.

6by9, 2023-10-13:
@SerdarAbali 
> Official 7" display, the touch function does not work. I am here to acknowledge and confirm the problem.

and as per https://github.com/raspberrypi/bookworm-feedback/issues/43#issuecomment-1759556114
> Did you rpi-update? We know that the fix isn't in the current images.

SerdarAbali, 2023-10-13:
@6by9 my bad, thank you. rpi-update fixes it for good.
fresh bookworm 64-bit
Pi 4

WarrenJD, 2023-10-19:
> I performed an rpi-update and indeed, the screen now responds to a touch. However, the touch functionality seems to be inverted after the first reboot following the update. I had to set the orientation back to "normal" and then inverted it again to get it to perform as desired.

What did you do exactly? I am having the same issue, I reverted back to normal then inverted, and the touch functionality is still inverted.

WarrenJD, 2023-10-21:
> I performed an rpi-update and indeed, the screen now responds to a touch. However, the touch functionality seems to be inverted after the first reboot following the update.

I have done this several times, and the touch functionality is inverted still.

WarrenJD, 2023-10-21:
> I performed an rpi-update and indeed, the screen now responds to a touch. However, the touch functionality seems to be inverted after the first reboot following the update. I had to set the orientation back to "normal" and then inverted it again to get it to perform as desired.

I have done this several times, and the touch functionality is inverted still.

jvoss11, 2023-10-21:
very new to raspberry pi. just moved from bullseye to a fresh bookwarm install.
i have only done ssh, and rpi-update as stated above however to the fresh OS.  
However i noticed that the mouse pointer is not showing when dragging or when selecting any item.  it will return if i use a connected usb mouse.  not sure if this has been addressed.

willyvaessen, 2023-10-22:
> I have done this several times, and the touch functionality is inverted still.

Sorry to hear that. I performed the update that was suggested and the steps I described. Nothing more.

raccoon-pi, 2023-10-28:
https://www.okdo.com/getting-started/get-started-with-7-touchscreen-for-raspberry-pi/

I wired on the above site.

And on debian bookworm Release 2023-10-10(bake with pi imager v1.7.5),

it worked after `sudo rpi-update`

I'm not sure if that wiring actually works.

GeekNJ, 2023-10-28:
Thanks - the rpi-update worked for me after reflashing a SD card with the latest 32-bit image that Pi Imager offered.

6by9, 2023-10-28:
@raccoon-pi 

> https://www.okdo.com/getting-started/get-started-with-7-touchscreen-for-raspberry-pi/
> 
> I wired on the above site.

That wiring is wrong. You only need two wires for power - please see the Raspberry Pi documentation for the display at https://www.raspberrypi.com/documentation/accessories/display.html#powering-the-touch-display

The extra two wires are covered in https://www.raspberrypi.com/documentation/accessories/display.html#legacy-support and are only needed for the very original Pi1 Model A and Model B.

tigernero79, 2023-10-29:
double tapping doesn't work in Debian 12. this doesn't allow you to open folders or launch apps that are on the desktop

ysiegel29, 2023-11-03:
Hello, does anyone know if this screen will work with OS Lite Bookworm and Pygame? The above discussion relates to the desktop version but does it matter?
I have a hdmi 7" display and can't get touchscreen to work... Thank you!

pelicanmedia, 2023-11-06:
> double tapping doesn't work in Debian 12. this doesn't allow you to open folders or launch apps that are on the desktop

Same here. After a rpi-update. Changing orientation and rebooting, twice. Single touch works, but double tap does not.

MartinNK, 2023-11-06:
I installed bookworm 64bit on a fresh microSD with image on a Pi 3b ... now having some problems. My original 7" touch display is used 180 degrees rotated because of the frame I used.

Out of the box 2 problems ... display is now upside down, but I want to see console mode (at boot with logo and gui - wayland insteed of X - rotated right   ... touch does not work (seen above solution ...  rpi-update) ...

problem with vnc ... heard about realvnc server not compatible with wayland at the moment :-(

Some data I put on install with newest rpi-imager (v1.7.1 I think) is not copied right ...

tigernero79, 2023-11-06:
Rip imager online version 1.8.1

MartinNK, 2023-11-07:
Yes ... sorry seen it was 1.8.1. Thanks.Von meinem/meiner Galaxy gesendet

  Bookworm image (Issue #43)

DNSGeek, 2023-12-16:
It's still not working for me on an RPi 3B running bookworm-64 bit. Yes, I've run rpi-update.

aneitzel, 2023-12-16:
same problem here, i also applied latest rpi-updates + reboot

BUT ORIGINAL 7" SCREEN TOUCH NOT WORKING :-(

alpha@pi2:~ $ cat /proc/version 
Linux version 6.1.68-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1708 SMP PREEMPT Thu Dec 14 16:09:41 GMT 2023

alpha@pi2:~ $ sudo rpi-update 
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 6.1.21-v8+
#############################################################
WARNING: This update bumps to rpi-6.1.y linux tree
See: https://forums.raspberrypi.com/viewtopic.php?t=344246

'rpi-update' should only be used if there is a specific
reason to do so - for example, a request by a Raspberry Pi
engineer or if you want to help the testing effort
and are comfortable with restoring if there are regressions.

DO NOT use 'rpi-update' as part of a regular update process.
##############################################################
Would you like to proceed? (y/N)

 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  144M  100  144M    0     0   189k      0  0:13:03  0:13:03 --:--:--  177k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 6.1.68-v7+
 *** depmod 6.1.68-v7l+
 *** depmod 6.1.68-v8+
 *** depmod 6.1.68+
 *** depmod 6.1.68-v8-16k+
 *** Updating VideoCore libraries
 *** Using SoftFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to ceecdc222d448aa708b1de59f7053aa55c43b283
 *** A reboot is needed to activate the new firmware

tigernero79, 2023-12-16:
the only way to have touch working is:

to type:

1)

`sudo nano /boot/config.txt`

put a # inside
(comment out) then disable the entry:

comment on the entry like this:

`dtoverlay=vc4-kms-v3d`

like this:

`#dtoverlay=vc4-kms-v3d`

Then

2) 

enable xorg mode from "sudo raspi-config ---> advance options ---> Wayland ---> select --- > W1 X11

reboot and the touch will work

if you have to reverse it, remember to put the directive in confix.txt:

`sudo nano /boot/config.txt`

insert

`lcd_rotate=2`

steveiliop56, 2023-12-16:
No why? The whole point is to work on wailand. Not switch back to x11.

tigernero79, 2023-12-16:
> No why? The whole point is to work on wailand. Not switch back to x11.

the touch with wayland works for me, I'm talking on an official pi5 4gb ram and 7" touch, the touch problem happens when I rotate the screen, it doesn't rotate the touch so clicking on the left results in a touch on the right but above all in wayland the double touch doesn't work Going back to 100% certified with their official display.

steveiliop56, 2023-12-16:
Maybe try to go disable the full kms driver?

tigernero79, 2023-12-16:
Unfortunately it doesn't solve the problem, the double touch would actually work but according to the Raspberry technicians the finger doesn't put pressure on the same point as the touch and therefore it doesn't perform. with x11 instead the system senses a nearby area and performs a double tap. it is used to open folders and run a program. going back to x11 is currently the ideal solution for those who want to use official displays with bookworm

steveiliop56, 2023-12-16:
Yeah indeed it's unstable for now. It was even hard to set up a keyboard until recently. It's a big change but needs a lot of work to be stable.

lurch, 2023-12-17:
There seems to be a couple of different issues being discussed here:
1. Double-touch not working: known problem, see https://github.com/raspberrypi/bookworm-feedback/issues/15
2. Display / touchscreen rotation: see the documentation at https://www.raspberrypi.com/documentation/accessories/display.html#changing-the-screen-orientation

DNSGeek, 2023-12-17:
> There seems to be a couple of different issues being discussed here:
> 
> 1. Double-touch not working: known problem, see [Double-click doesn't work with the Touch Display #15](https://github.com/raspberrypi/bookworm-feedback/issues/15)
> 2. Display / touchscreen rotation: see the documentation at https://www.raspberrypi.com/documentation/accessories/display.html#changing-the-screen-orientation

No, we're also talking about the complete lack of touch functionality at all on an RPi 3B running bookworm. 64.

lurch, 2023-12-17:
@DNSGeek Sure, I was going to check that when I'm in the office tomorrow, and update this issue as appropriate :slightly_smiling_face:

lurch, 2023-12-18:
The original problem reported in this issue (where an `rpi-update` was needed in order to get the touch functionality of the Raspberry Pi Touch Display working) has now been incorporated into the December 5th 2023 release of Raspberry Pi OS, so closing this issue.

@DNSGeek I've been doing extensive testing today, and I've not found any situations where the touch functionality isn't working. Note that if you have both an HDMI and DSI screen attached, you need to use the Screen Configuration program to assign the touch-inputs to the DSI screen, as described in https://www.raspberrypi.com/documentation/accessories/display.html#changing-the-screen-orientation
(If you're still having problems, please open a separate issue)

I **did** find a separate problem with the 3B+ though (see https://github.com/raspberrypi/linux/issues/5801 ), but that also affects the display-output of the screen, not just the touch-input.

closed 2023-12-18 by lurch

tigernero79, 2023-12-19:
With wayland after reading the link above:

[htt](https://www.raspberrypi.com/documentation/accessories/display.html#changing-the-screen-orientation)

placing the directive in boot/confix.txt

`dtoverlay=vc4-kms-dsi-7inch,invx,invy`

and disabling the entry:

`display_auto_detect=1`

putting the hash sign # at the beginning

`#display_auto_detect=1`

the touch works with the rotation done via "GUI" but the double touch still doesn't work

I hope they will resolve it soon

for the moment I return to x11 with the config.txt configuration I reported above

colejohnson66, 2024-01-18:
Experiencing this issue on an image created yesterday (with official imaging tool). We have a CM4 with the official 7-inch display on DSI0. We ran `rpi-update` and `apt update+upgrade`, but the touchscreen still does not work.

6by9, 2024-01-18:
> Experiencing this issue on an image created yesterday (with official imaging tool). We have a CM4 with the official 7-inch display on DSI0. We ran `rpi-update` and `apt update+upgrade`, but the touchscreen still does not work.

DSI0 is known to have issues on CM4 though - https://github.com/raspberrypi/linux/issues/4946.

The `dsi0` override to `vc4-kms-dsi-7inch` is currently intended for Pi5 only.
And neither display_auto_detect nor camera_auto_detect are intended to work on CMs as there are too many wiring permutations to make it worth trying to handle.

colejohnson66, 2024-01-18:
We don't have any issues with the display part of the module - only the touchscreen. And this was working fine when we were running Buster.

6by9, 2024-01-18:
> We don't have any issues with the display part of the module - only the touchscreen. And this was working fine when we were running Buster.

In which case you're using the touch screen with the legacy display stack, and not vc4-kms-v3d as is the default with Bookworm.
If you've changed that, then there is no saying what else you've changed. At that point it's hardly the fault of the Bookworm image.

DNSGeek, 2024-01-18:
I’ve changed nothing, and the touchscreen which worked fine in buster does not work in bookworm. The display works fantastic. I can’t touch a damn thing. I’ve made no changes to bookworm other than doing updates.

6by9, 2024-01-19:
> I’ve changed nothing, and the touchscreen which worked fine in buster does not work in bookworm. The display works fantastic. I can’t touch a damn thing. I’ve made no changes to bookworm other than doing updates.

I was responding directly to colejohnson66 as he's using a CM4 and on DSI0.
The CM4 has no autodetection for the display, or configuration in the legacy stack, and DSI0 has other issues.

In https://github.com/raspberrypi/bookworm-feedback/issues/43#issuecomment-1861176023 lurch said
> (If you're still having problems, please open a separate issue)

Did you do that?

mangelozzi, 2025-01-02:
> > I’ve changed nothing, and the touchscreen which worked fine in buster does not work in bookworm. The display works fantastic. I can’t touch a damn thing. I’ve made no changes to bookworm other than doing updates.
> 
> I was responding directly to colejohnson66 as he's using a CM4 and on DSI0. The CM4 has no autodetection for the display, or configuration in the legacy stack, and DSI0 has other issues.
> 
> In [#43 (comment)](https://github.com/raspberrypi/bookworm-feedback/issues/43#issuecomment-1861176023) lurch said
> 
> > (If you're still having problems, please open a separate issue)
> 
> Did you do that?

So what is the fix? "DSI0 has other issues", how do we fix these "other issue" ? I am also using a CM4.

6by9, 2025-01-03:
> So what is the fix? "DSI0 has other issues", how do we fix these "other issue" ? I am also using a CM4.

DSI0 issues were resolved back in April 2024, and isn't the subject of this issue.
Use a kernel from after that date and it should al be working.

If you have a specific problem, then open a new issue.
Commenting on closed issues which are only tangentially related won't help.