raspberrypi/bookworm-feedback #44: OpenGL: skipping each second vblank event issue in fullscreen mode

ghollingworth, 2023-10-10:
a quick update on this: we identified the source of the issue, it is a
downstream patch we have to limit CMA memory usage through DRI, which
limits the number of back buffers to 2 when vsync is enabled (instead
of the default of 3). It seems that this is not enough to ensure
maximum performance is delivered in some scenarios, and fullscreen
through XWayland is one such case, which was the case being tested by
the reporter.

The original patch was made to address issues with running out of CMA
with dual 4K screens. We are currently working on a fix that would keep
the limit at 2 only in the case where we actually have 2 outputs with
larger than fullHD resolution. In that case, we would be accepting a
performance trade-off in fullscreen vsync mode in exchange of reducing
CMA pressure, but in any other scenario we would be allowing maximum
performance.

Expecting a patch soon...

lurch, 2023-10-10:
Related forum post, for context: https://forums.raspberrypi.com/viewtopic.php?t=357353

qrp73, 2023-10-14:
I can add that there is possible workaround for this issue, if app do the following calls after switch to fullscreen mode:
```
glXSwapInterval(0); 
glXSwapInterval(1);
```
it will fix the issue

qrp73, 2024-02-03:
any update? The issue is still present...