raspberrypi/bookworm-feedback #47: Sense Hat Emu Problem - Duplicate

steveiliop56, 2023-10-11:
Yeah about sense hat emulator... The problem doesn't seem to be the np.float thingy which I fixed a long time ago because version 1.2.1 exists with a fix. The problem is that the app just doesn't launch at all with no logs...

steveiliop56, 2023-10-11:
Update! Got it to work.. An other instance was running in the background 🤣. The fix for everyone interested is ```sudo pip3 install --break-system-packages sense-emu==1.2.1``` and then it should work!

lurch, 2023-10-11:
This was already specifically mentioned in https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/ so I'm going to close this issue.

Using `--break-system-packages` isn't the approach we're recommending - see http://rptl.io/venv

closed 2023-10-11 by lurch

steveiliop56, 2023-10-12:
Yes but that's the only way to fix the sense-emu-tools package error

aallan, 2023-10-12:
You can install a version of the Sense Hat Emulator, along with a version of Thonny that can talk to the emulator, with these commands:

```
$ sudo apt install libgirepository1.0-dev gcc libcairo2-dev pkg-config python3-dev gir1.2-gtk-4.0
$ python -m venv ~/local
$ source ~/local/bin/activate
(local) $ pip install pycairo PyGObject
(local) $ pip install sense-emu thonny
```

Whenever you want to run the Sense Hat Emulator, you'll need to open a Terminal, run `source ~/local/bin/activate`, then run the Sense Hat Emulator with `sense_emu_gui`. If you want to use Thonny with the Sense Hat Emulator, you need to run `thonny` from the terminal after you run the `activate` command. See https://rptl.io/venv for more information about working with Python virtual environments.

A future update to Raspberry Pi OS will eliminate the need for this workaround.

steveiliop56, 2023-10-21:
Yeah but I am talking about a global fix of you install the sense-emu-tools package...