raspberrypi/bookworm-feedback #50: Cant find newly installed Raspberry Pi OS (2023-10-10) in headless mode

mats-nk, 2023-10-12:
### Describe the bug

I found that a freshly installed image created with imager has lost the support for mDNS (avahi), workstation.

The impact is that it was previously easy to find a headless installation without hassling with scanning or checking routers DHCP leases.

Please change this back to the previous behavior.

File affected: `/etc/avahi/avahi-daemon.conf`

Change needed, from `publish-workstation=no`  to `publish-workstation=yes`

### Steps to reproduce the behaviour

Install `sudo apt avahi-utils`

Browse mDNS, `avahi-browse -a` and no "workstation" can be found.

Enable it and restart avahi service and it can be discovered, see bug report for details.

### Device (s)

Raspberry Pi Zero W / WH, Raspberry Pi Zero 2 W, Raspberry Pi 3 Mod. B+, Raspberry Pi 4 Mod. B, Raspberry Pi 400

### System

System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

lurch, 2023-10-12:
`ping yourhostname.local` still seems to work fine (which I believe is what most people will be using), so I guess this is similar to https://github.com/raspberrypi/rpi-imager/issues/655 where it's looking for a specific "advertised service", rather than just the hostname.

ghollingworth, 2023-10-12:
I guess this has come from upstream:  Titled: "Don't leak host info by default"

https://github.com/lathiat/avahi/commit/530fbb59abafb970ef1dd8f61571b13fb0918b23

XECDesign, 2023-10-12:
Do we want to do anything about that? It seems like a sensible upstream change, especially if we couple it with the proposed change Andrew linked.

mats-nk, 2023-10-13:
Before disregarding this I would like to ask "What is the goal what Raspberry Pi OS"?

As I understand it, it is broadly used as a first step into Linux and the Linux foundation has made a wonderful job in that aspect.

As for myself I am actively assisting new users to get started with Linux, and as I have been working with Linux servers since 1999 I have seen that to get a system up and running the first time is hard.

So to skip this is a blow back for the "Lite OS" and headless installation usability and I propose to investigate a bit further what the implication would be for the project.

And as for security concerns, "security by obscurity" is bad security policy in my world.

And lastly if Andrew is lurch, then it is my feature request he linked to ;-)

ghollingworth, 2023-10-13:
I understand, but if the upstream developers have made a change to a package configuration we shouldn't just go reversing it without any thought to the consequences.  I know there are usability ones that you point out, but I don't know if there are security concerns as well that we don't know about.  For example, what if this is a commonly used path for some known hacking software and that is why they changed it?

That's why I want to understand it, if you want to get in touch with those developers to make sure they don't know of any 'known attack vectors' that would really help

mats-nk, 2023-10-13:
I agree, to understand what you do and what the implications is, is important. I am all in for not taking quick decisions.

mats-nk, 2023-10-13:
I would like to propose a workaround/solution.

If a "First boot script" feature could be introduced, ex `/boot/1boot.sh`. The idea would be that this script is run when the Raspberry Pi OS has completed its first boot completely.

The benefits would be huge for the Raspberry Pi community.

lurch, 2023-10-13:
> If a "First boot script" feature could be introduced

See the discussions in https://github.com/raspberrypi/rpi-imager/issues/174 and https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/40 and https://github.com/raspberrypi/rpi-imager/issues/554 etc.

mats-nk, 2023-10-13:
And should I understand it that the door is closed for this feature?

ghollingworth, 2023-10-14:
The concept of running some extra script may be added to Imager in a later revision.  But it is not going to be available in the next one

mats-nk, 2023-10-17:
Ok, thx for the answer. But kind of the same answer when I proposed it in 2016.

lurch, 2023-10-17:
It's probably something that we don't officially support, but if you want to hack this in yourself, see https://github.com/raspberrypi/bookworm-feedback/issues/72#issuecomment-1764080678