raspberrypi/bookworm-feedback #51: pinctrl missing BCM2712 support

trejan, 2023-10-12:
pinctrl gives a "No GPIO chips found" error when run on a Pi 5. gpiochip_bcm2835.c is missing support for BCM2712 so gpio_find_chip() never finds anything.

Is raspi-gpio deprecated now that pinctrl has released? That also gives an error due to unexpected ranges data.

pelwell, 2023-10-12:
I don't know why the image includes an old version of pinctrl - you can see the full version with 2712 and RP1 support (and even build it yourself) in our utils repo: https://github.com/raspberrypi/utils/tree/master/pinctrl

pelwell, 2023-10-12:
And yes, raspi-gpio is deprecated - it won't be getting Pi 5 support.

aallan, 2023-10-12:
@pelwell So `raspi-gpio` turns up a couple of times in the documentation, we'll switch that out for `pinctrl`. Can you let me know when you get the package updated in the repo, and I'll get that done?

pelwell, 2023-10-12:
@XECDesign Should I have nudged you? Does the utils repo need a `stable` branch?

XECDesign, 2023-10-13:
Looks like the pi5 version of the package got lost somewhere while the repos were being shuffled. The version in apt should be okay now.

closed 2023-10-13 by lurch