raspberrypi/bookworm-feedback #52: How to change the I2C interface rate of CSI/DSI on Bookworm 6.1.0?

Tiger0215, 2023-10-13:
The traditional i2c_vc_baudrate=200000 does not work.
Are there any new directive changes, or am I doing something wrong?

pelwell, 2023-10-13:
You mean `dtparam=i2c_vc_baudrate=200000`?

Tiger0215, 2023-10-13:
> You mean `dtparam=i2c_vc_baudrate=200000`?
Yes
`dtparam=i2c_vc_baudrate=200000 `
It works with Bullseye but not with Bookworm, maybe Bookworm has new instructions to control the I2C rate of the DSI interface?

pelwell, 2023-10-13:
This feature comes from the kernel and Device Tree - it's independent of  the OS version, only the kernel version should matter (and the firmware, but that hasn't changed in this regard).

What output do you get from this command?
```
$ od -i --endian=big /proc/device-tree/soc/i2c@7e205000/clock-frequency
0000000      200000
0000004
```
I've included the output I get on a Pi 4 running Bookworm on a 6.1 kernel with your dtparam line.

One thing that has changed in Bookworm is the mount point of the FAT partition - it's now /boot/firmware/. There is a symbolic link from /boot/config.txt to /boot/firmware/config.txt (the same goes for cmdline.txt), but it would be easy for something to replace the link with an actual file, leaving you with two config.txt files - one that works and one that doesn't.

Tiger0215, 2023-10-13:
Thanks, based on your steps, now I got the correct answer.

closed 2023-10-13 by Tiger0215