raspberrypi/bookworm-feedback #53: console/terminal raspi-config selection of no autologin not working

PitHerm, 2023-10-13:


spl237, 2023-10-13:
This definitely works on a standard image - it has been tested repeatedly.

Can you please provide more information? Are you referrring to desktop autologin or console autologin? What exact steps are you taking to enable this? Does raspi-config give an error, or does autologin just not work once selected? What changes have you made to the image since installing it? Are you running locally or over SSH or VNC?

Just saying something "isn't working" doesn't give us enough information to diagnose the problem.

XECDesign, 2023-10-13:
Just tried it out myself. Works fine here.

PitHerm, 2023-10-13:
used raspberry imager to download bookworm full 32 bit applied user info and wlan info and wrote it to micro sd

attached monitor and keyboard directly to Pi 4b

started , ok, standard autologin on desktop

opened terminal > sudo raspi-config > selected desktop without autologin > restart > still autologin.
using desktop app raspi-config  > disabled autologin > restart > everything fine.

XECDesign, 2023-10-13:
Okay, just tried desktop with no autologin and it logged in. So yeah, there's an issue somewhere.

spl237, 2023-10-13:
OK, thanks - I can now see the same happening here. Looks like raspi-config isn't removing the autologin user from the lightdm config file for some reason, but it is not obvious why - I'll have a look.

XECDesign, 2023-10-13:
I see why. I'll get it, Simon.

spl237, 2023-10-13:
Yes, just saw what was causing it! Missing wildcard... ;) Thanks, @XECDesign

XECDesign, 2023-10-13:
Updated and pushed to apt. This version works for me now.

https://github.com/RPi-Distro/raspi-config/commit/77710e8ec885c601791eda81701b996895ab4ac6

PitHerm, 2023-10-13:
Thank's works here too now (just tested on 64 bit , 32 bit will follow later)

closed 2023-10-13 by lurch