raspberrypi/bookworm-feedback #56: setting window position doesn't works

qrp73, 2023-10-13:
Here is simple test code in python, but exactly the same issue happens with C++ code which uses X11 window.

Steps to reproduce:
1) Open Thonny and copy-paste this code:
```
import tkinter as tk

def resize_window(event):
    if event.keysym == '1':
        root.geometry("320x240+100+100")
    elif event.keysym == '2':
        root.geometry("320x240+50+50")
    elif event.keysym == '3':
        display_width = root.winfo_screenwidth()
        display_height = root.winfo_screenheight()
        root.geometry(f"{display_width}x{display_height}+0+0")

root = tk.Tk()
root.title("Resizable Window")
root.bind("<Key>", resize_window)
root.mainloop()
```
2) run the code and click on opened window to give focus on it
3) press key 1

Expected result: window position should be changed to 100,100; window size should be changed to 320x240

Actual result: window size is changed to 320x240, but position is not changed - BUG

4) press key 2

Expected result: window position should be changed to 50,50

Actual result: window position is not changed - BUG

5) press key 3 (it will resize window on full display surface)
6) press key 1

Expected result: window position should be changed to 100,100 and window size is restored to 320x240

Actual result: window size is restored to 320x240, but window position is not changed and remains at 0,0 - BUG

spl237, 2023-10-13:
This sort of thing has changed significantly in Wayland compared to X. In many cases, commands which worked in X now need to be changed, or issued in a different order relative to the presentation of the window. It's likely to be the application doing something that Wayland doesn't expect which is the problem, so the application itself will probably need to change to fix it.

spl237, 2023-10-13:
Also, tkinter does not have a native Wayland backend, so is using Xwayland; that adds another layer of complexity. This probably isn't going to get fixed as a result, as only native Wayland toolkits will be supported in future.

qrp73, 2023-10-13:
> In many cases, commands which worked in X now need to be changed, or issued in a different order relative to the presentation of the window.

I want to change window position, how can I change test code to do it?

spl237, 2023-10-13:
I don't know how to do it in tkinter - as I said, it doesn't have a Wayland backend, which means it is going to be obsolete in the near future anyway. Try asking the tkinter developers.

qrp73, 2023-10-13:
In c++ there is XMoveWindow function which just don't works as expected on bookworm. How to fix it?

aallan, 2023-10-13:
> In c++ there is XMoveWindow function which just don't works as expected on bookworm. 

The function starts with "X", which suggests it attempting to access X11 libraries under the hood. Bookworm no longer uses X11, instead it uses Wayland.

lurch, 2023-10-13:
An [old StackExchange post](https://stackoverflow.com/questions/67318357/how-to-set-the-position-of-a-wayland-window-on-the-screen) links to https://wayland-book.com/xdg-shell-in-depth/interactive.html which says "However, a deliberate design trait of Wayland makes application windows ignorant of their exact placement on screen or relative to other windows. This decision affords Wayland compositors a greater deal of flexibility".

qrp73, 2023-10-13:
As I know there is Xwayland to support X11 API, is it possible to fix it in such way get working calls that requesting window position change through X11 API?

lurch, 2023-10-13:
I hope you don't mind me asking, but _why_ do you need to set a specific window position?

qrp73, 2023-10-13:
> _why_ do you need to set a specific window position?

There may be different reason to do it. For example if some app using many window to display different information it needs to be able to display it at proper location to not mix them on the display. It needs to store user preferences for different window, etc. There is no way to give all window management about window position and size to Wayland, because app may have very complicated window system with a lot of windows. If system doesn't allow it, app will needs to implement it's own window manager and manager it's window within single Wayland window. I think this is not a good idea to have separate window manager for each app.

Another example is to switch into fullscreen mode. I know, current raspi os bookworm support window manager extension for that and it works. But some system may not have it and then app needs alternative way to do it. Some old apps may don't know about extension to enter fullscreen mode and using old approach, they will fail on bookworm wayfire.

pidloop, 2023-10-14:
Sorry guys, I can't help myself but here goes: Wayland is, what, fifteen years old now? and it still can't do half the important things X11 does. Screen locking/sharing/recording/blanking? nope. Cross-app communication? nope. Screen color calibration? nope. Run an app over a network? nope. Swap to a different window manager? nope.

Don't get up, I'll let myself out.

KenT2, 2023-10-14:
I know, current raspi os bookworm support window manager extension for that and it works. 

@qrp73 could you tell me more. I have an application, Pi Presents, that needs fullscreen undecorated windows for each display, which I could do in Tkinter under X11, but it does not work with Wayfire.

qrp73, 2023-10-14:
@KenT2 I don't know if it's possible with tkinter, I done it in c++ code, you can find example in freeglut source: https://github.com/FreeGLUTProject/freeglut/blob/87ab8867190a5b0100d065a217090fa242b92b79/src/x11/fg_window_x11.c#L94

ghollingworth, 2023-10-14:
I don't believe this is possible through Wayland at all.  The reason is that if you can force your window to be in a particular position it means you might be able to develop an application that appears in front of, for example, a password entry box.

qrp73, 2023-10-14:
> if you can force your window to be in a particular position it means you might be able to develop an application that appears in front of, for example, a password entry box.

There is no need to set window position do develop window which appears in front of password window.
Here is example of such code  which shows overlap window in front of password window and it don't use window position at all:
```
import tkinter as tk

root1 = tk.Tk()
root1.title("Overlap Window")
root2 = tk.Tk()
root2.title("Password Window")
root1.mainloop()
```

window position is not related to such kind of security, this is a window focus issue.

qrp73, 2023-10-14:
For example, if you develop application which needs to show 50 windows with different tools. They should be shown at specific position stored by user. Without ability to set window position you will get random mix of windows at app startup and needs to move all 50 windows manualy to their proper position. And it needs to be repeated each time when application is started... Just nightmare for user...

qrp73, 2023-10-14:
it will be naive to think that X11 API will be deprecated soon. So, remove X11 support means that many apps will be broken and cannot be used. No one want to use such system which cannot run most of the applications. X11 API should be supported

qrp73, 2023-10-14:
It appears that Wayland/Plasma have client positioning support in custom protocol: https://github.com/KDE/plasma-wayland-protocols/blob/master/src/protocols/plasma-shell.xml#L159.

Is it possible to do the same for Wayfire in raspi os bookworm?

qrp73, 2023-10-14:
Do I understand correctly that [popup menu in VLC appears at random location](https://github.com/raspberrypi/bookworm-feedback/issues/59) away from mouse cursor, because set window position is missing from XWayland? Do you planning to fix it? Or it will remain in non usable state just because you don't want to add support for XMoveWindow?

ghollingworth, 2023-10-15:
You should go to the upstream wayfire repo to ask what they’re planning to support

closed 2023-10-15 by ghollingworth

spl237, 2023-10-18:
Just to follow up on this.

Wayland does not allow an application to specify the location of a window, full stop. The location of a window is entirely up to the compositor, and any commands, X, GTK or otherwise, which set the location of a window will have no effect.

The only way you can position a window is to use the layer-shell protocol extension to Wayland, which requires a backend which supports it - GTK has one, gtk-layer-shell - which allows a limited degree of control over window position; you can pin a window to one edge of the screen and set a margin away from it. But as far as I am aware, that is the only way to do it - X window positioning commands will be simply ignored by the compositor. This is a deliberate design decision in Wayland and cannot be changed.