raspberrypi/bookworm-feedback #57: pcmanfm: huge memory leak and system freeze

qrp73, 2023-10-13:
pcmanfm process has memory leak, each time when some app enters into fullscreen mode, pcmanfm process eats additional 15 MB memory. 

When pcmanfm process eats all available memory it leads to system freeze due to out of memory. 

Steps to reproduce:
1) Start firefox (you can use any other app which allows to enter fullscreen)
2) Press F11 to enter fullscreen mode
3) Press F11 to exit fullscreen mode
4) repeat steps 2-3 for 60 times.
5) open gnome system monitor or btop and check memory consumption for pcmanfm

Expected result: pcmanfm keeps stable memory usage

Actual result: pcmanfm memory usage increased for 1000 MB

6) continue pressing F11 until system will be freezed due to out of memory

Expected result: no out of memory issue

Actual result: pcmanfm process eats all available memory and system is freezed

popcornmix, 2023-10-13:
I can reproduce.

I sat there pressing F11 watching the RSS number in task manager for pcmanfm increase.
It got to about 2.6G (on a 4G Pi4) before system froze (can move mouse pointer, ssh is unresponsive).

spl237, 2023-10-13:
Yes, also here. No idea what is happening, though - fullscreening apps should make no difference at all to pcmanfm, which is just drawing the desktop - it shouldn't need to redraw when an app has fullscreened. I'll have a look with valgrind next week and see if I can see what is going on.

spl237, 2023-10-13:
A quick look at the pcman code suggests the issue may be a cached copy of the desktop background image which is not being properly freed under Wayland - the function which I thought would free the cached image may not be releasing the memory as I had hoped. If the desktop is redrawing whenever windows are shown and hidden on top of it, this could well be the problem.

spl237, 2023-10-14:
I've found a fix for this - it was indeed that the cached background images were not being released from memory, and for some reason under Wayland the cache gets reloaded a lot more often than it did under X. My fix is currently a bit of a kludge, and I want to properly understand what is going wrong before I push it out, so it'll be a day or two.

spl237, 2023-10-14:
Fixed - the line of code which freed an image buffer had mistakenly been included in a conditional block which did not execute on Wayland. 

We'll push a new version to apt next week.

closed 2023-10-14 by spl237

timg236, 2023-10-14:
Sounds good. Another way to reproduce this is to power cycle the monitor (HDMI hotplug). I power off a shared monitor at night and noticed that this morning the desktop was just grey + mouser cursor with OOM messages in the console.

spl237, 2023-10-14:
Yes, I'd expect monitor hotplugs to have exactly the same effect - anything that causes the desktop to redraw will invoke the leak. So the same fix should also fix the hotplug issue.

qrp73, 2023-10-16:
today at 18:30 new pcmanfm appears in apt update, just applied it. 
At a glance it looks that the bug is fixed and can be closed. Thanks

winlog, 2023-12-01:
Hi guys! I faced the same issue. RPI4B, 4 Gb RAM.
New fresh install
pcmanfm --desktop --profile LXDE-pi process eat 4 Gb RAM in 24 hours. No desktop activity (because use ssh and services via local network).

JamesH65, 2023-12-01:
And it is definitely Bookworm, 32 or 64bit?

winlog, 2023-12-01:
lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
6.1.0-rpi6-rpi-v8

64 bit, donwloaded from rpi site

PedroMartinSteenstrup, 2023-12-20:
@winlog I think I had same problem on my Raspberry Pi 5, where the process pcmanfm was eating up all memory as you can see on my screenshot.

At 03:02 I stopped my docker container for Plex, since that's the only app I was running that could use so much pcmanfm.
I then updated all packages on the raspberry (upgrade to pcmanfm arm64 1.3.2-1+rpt18 ) and restarted Plex container, before viewing some video to test.

As you can see, memory consumption is now under control.
So if you have already updated your OS, I can encourage you to figure out any app you're running in case it's containerised.

[image]

ghollingworth, 2023-12-20:
It would be interesting to unplug the HDMI monitor and then wait for 24 hours (or however long it takes for it to be obvious the problem is either still there or gone away).

That should give us evidence to suggest that it is something like the hotplug (which might be turning on occasionally?) causing some problem with Wayfire/pcmanfm

You can also do (from your ssh terminal)

pkill wfrespawn 
pkill pcmanfm
WAYLAND_DISPLAY="wayland-1" pcmanfm --desktop --profile LXDE-pi

Then see if anything changes (i.e. does pcmanfm give you any errors if you power on / off the TV or plug / unplug the cable?)

Gordon

Swift42, 2024-07-23:
I have a similar (or the same) problem with pcmanfm version 1.3.2-1-rpt20 on my Raspberry 5.
My Raspberry runs 24/7 and I switch off the monitor overnight.
Nearly every morning, when I switch the monitor back on, I have the same problem: The desktop background is not showing the bg image, but is suddenly black. Also pcmanfm is at 100% cpu (but has a normal memory consumption of 1.2%).
The only way to get all running again is a `sudo killall -9 pcmanfm`
After issuing the command, the bg image is instantly shown again and pcmanfm is usable again.
Is there something I can do to fix this problem?