raspberrypi/bookworm-feedback #58: window title disappearing issue

qrp73, 2023-10-13:
Steps to reproduce:

1) install mesa-utils:
```
sudo apt install mesa-utils
```
2) run glxgears from console
3) click on window with gears to get focus on it and press Ctrl+Alt+CursorLeft and then Ctrl+Alt+CursorRight

Expected result: window dock on left display side and then on right display side, window title is present

Actual result window dock works ok, but window title disappears after dock to the right display side - BUG

Note: the same issue with window title disappearing happens with any app window and in different scenarios. I just wrote steps to reproduce for most easy and stable reproducible way to catch it. Most of the time it happens with maximized windows.

popcornmix, 2023-10-13:
I can reproduce.

spl237, 2023-10-16:
Fixed - when switching between two tiled configurations, the titlebar was mistakenly being cleared. Will be in next update.

closed 2023-10-16 by spl237

qrp73, 2023-10-17:
yes, it is fixed after last update