raspberrypi/bookworm-feedback #59: VLC picture layout issues

qrp73, 2023-10-13:
Steps to reproduce:

1) Use display resolution 1280x1024
2) Try to open video with resolution 1280x720 or 1920x1080 or 3840x2160

Expected result: VLC window size is limited to available display size and video picture is scaled and placed at center

Actual result: VLC window size is limited to available display size, video picture is scaled but placed at top part of window with a large black gap at the bottom - BUG

Note video with resolution 1280x720 don't have this bug in window mode

3) enter fullsreen mode

Expected result: video picture is scaled and placed at center of display

Actual result: video picture is scaled but placed at top half of the display, part of the picture is outside display, bottom half of display is just black empty space - BUG

Note: this issue happens even with video resolution 1280x720 which can fit into display size with no scale.

4) try to right click with mouse to get context menu

Expected result: context menu appears under mouse cursor and properly react on mouse

Actual result: context menu appears far away from mouse cursor and there is offset between real mouse cursor and menu hot tracking, no way to click on menu item - BUG

Note: sometimes menu appears normally, but most of the time it appears with some offset and with incorrect hot tracking position.

popcornmix, 2023-10-13:
Ping @jc-kynesim

John did comment recently:
> VLC/Wayland: Video positioning when letter/pillarboxing can be dodgy - I’m pretty sure this is a Wayland issue (don’t know which layer), I have created a simple test prog and sent it to Igalia.

so your issue may be an instance of this.

jc-kynesim, 2023-10-14:
Yeah there is a wayfire/wlroots issue with fullscreen where positioning is bugged or indeed any window where the video surface is smaller than than window. Igalia know about it (now) and are (I hope) working on it but from other mailing comments I've seen I think nearly all of them are off at a conference or something next week.
I'm a bit surprised by the "part of the video is offscreen" variation which I hadn't seen myself.
I do have an alternate wayland video output driver which works around this issue, but it is still a bit alpha.

lurch, 2023-10-14:
> I'm a bit surprised by the "part of the video is offscreen" variation which I hadn't seen myself.

There's some screenshots in the forums https://forums.raspberrypi.com/viewtopic.php?t=357299

JohnFarl, 2023-10-22:
I have problems with VLC too. It opens always too large and goes out of screen as if this isn't annoying enough, In full screen mode it is in center of the screen but it won't scale to fit the screen, so I'm forced to watch black bands on the sides.

jc-kynesim, 2023-10-22:
When you say "won't scale to fit" - is that because you are viewing a (say) 4:3 video on a 16:9 monitor or is this with videos that are the correct aspect ratio for your monitor?

JohnFarl, 2023-10-22:
> When you say "won't scale to fit" - is that because you are viewing a (say) 4:3 video on a 16:9 monitor or is this with videos that are the correct aspect ratio for your monitor?

Yes, previously in the case of 4:3 video on a 16:9 display it was possible to fit the video to the entire screen simply by enlarging or cropping part of the image. Now none of these modes work properly, 

At the start it opens always too large and goes out of screen and all adjustment to window  reset every time.

Also if I use right click on VLC video area, the contextual menu rather than opening near the cursor it opens in a weird position.

qrp73, 2023-11-13:
> > When you say "won't scale to fit" - is that because you are viewing a (say) 4:3 video on a 16:9 monitor or is this with videos that are the correct aspect ratio for your monitor?
> 
> Yes, previously in the case of 4:3 video on a 16:9 display it was possible to fit the video to the entire screen simply by enlarging or cropping part of the image. Now none of these modes work properly,
> 
> At the start it opens always too large and goes out of screen and all adjustment to window reset every time.
> 
> Also if I use right click on VLC video area, the contextual menu rather than opening near the cursor it opens in a weird position.
> 
> I thought we migrated to Wayland because it was easier to work with modern applications that use the GPU, considering all the problems, now I wonder if all this was just for the sake of having fancy windows effects.

window auto size works, but since XMoveWindow is not implemented in raspios Xwayland it can't move window so it appears outside screen.

But the more critical issue is that picture scale don't works properly. 
The picture appears at top part of window and part of picture appears outside window. In window mode you can reduce window height in order to avoid large gap at the bottom half of the window, but for fullscreen it doesn't works and picture always shown at top half of the screen instead of center.
It happens when I'm trying to play HD or UHD video on display with standard SVGA resolution 1280x1024 (5:4).

qrp73, 2023-11-13:
> I'm a bit surprised by the "part of the video is offscreen" variation which I hadn't seen myself. I do have an alternate wayland video output driver which works around this issue, but it is still a bit alpha.

It appears outside window because picture position within window is shifted from center. You can reproduce it by playing with setting Video->Aspect Ratio, just try to change it to any value and part of picture appears outside of window.

qrp73, 2023-11-13:
There is also third issue with picture layout. When window size is small (height about 256 and smaller) the picture starts to crop and right half of picture is cut (black color). The same issue happens when you try to set Video->Aspect Ratio->5:4 for video 1280x720

qrp73, 2023-11-13:
mpv player don't have such picture layout issues. But it can't properly set window position, so it leads to issue with switch windowed/fullscreen mode (impossible to exit from fullscreen mode with double click, but there is workaround you can use Ctrl+Alt+Left to switch into window mode). Also it leads to invalid picture scale in full screen mode, but you can workaround it by double click - mpv trying to exit from fullscreen, but can't (probably because XMoveWindow is not implemented), but picture scale is fixed and you can watch video.

jc-kynesim, 2023-11-13:
I have a VLC with a new Wayland video output module. It should fix most of the video positioning issues (though the random main window sizing & positioning issues remain) . With a bit of luck it'll turn up later this week.

andrum993, 2023-11-13:
Seems to be fixed as of today's update 👍

qrp73, 2023-11-13:
yes, with "Wayland dmabuf video output" video output module picture layout within window is fixed. This is good, now it is usable in fullscreen mode. 

But the same issue present for "Wayland shared memory video output" and other modules like "OpenGL for Embedded Systems 2 video output".

Window position issue also still present.

jc-kynesim, 2023-11-13:
> yes, with "Wayland dmabuf video output" video output module picture layout within window is fixed. This is good, now it is usable in fullscreen mode.
> 
> But the same issue present for "Wayland shared memory video output" and other modules like "OpenGL for Embedded Systems 2 video output".
Do you have a particular reason you want to use these modules? I wrote wl-dmabuf to avoid the current issues with them (it should also be a bit more efficient).

> Window position issue also still present.
I know. The problem is that Wayfire/Wayland doesn't really want to let you position windows and Qt doesn't manage it.

> Also context menu issue in fullscreen mode (right mouse click, it don't works most of the time, but sometimes works) still not fixed even for "Wayland dmabuf video output" module.
It should work a lot more of the time now - you need to not have the fullscreen controller onscreen - so right-click without moving the mouse should work. Again this is a Qt/Wayland positioning/focus issue.

qrp73, 2023-11-13:
Also I notice that now, with new wayland module, VLC is able to play video with Dolby Vision HVEC (H.265 dvhe) codec with resolution 3840x2160@60. It lagging like hell, but now picture is visible at least with low fps. Previously it was just black screen.

jc-kynesim, 2023-11-13:
I didn't know we had such a codec - I'm pleased it works for you at least a bit. The extra processing required to render DolbyVision is enough that I doubt you are ever going to get properly useful output on any current Pi. Other HDR methods where we can just pass the parameters to the display are a lot more plausible and work if you can use the DRM output module (start VLC outside Wayland).

qrp73, 2023-11-13:
There is some issue with resize filter in VLC, the quality is very bad and it's annoying to see this in full screen mode. Here is example:

Picture with origital video resolution:
[image]
[image]

Picture in fullscreen mode:
[image]
[image]

Usually such issue can be solved with selecting scale filter, Nearest Neighbor and Bicubic covers almost all cases. But I don't see such setting in VLC. 

I'm not sure, is it related to video output module or it happens in another module?

Also there is some annoying issue with window size. 

Steps to reproduce:
1) Open some video and resize window with mouse
2) Return original window size with menu Video->Zoom->1:1 Original
3) Click outside window

Exepcted result: window size is set to 1:1 and don't change

Actual result: window size is set to 1:1, but unexpectedly resize back when you click outside window

closed 2023-11-15 by qrp73