raspberrypi/bookworm-feedback #60: Keyboard configuration is incorrect on login screen - no 105-key PC (intl.) option

rmitchellgit, 2023-10-13:
At the log in screen for the desktop environment, the keyboard configuration is incorrect.

Once I log into the desktop environment, I am able to type correctly.

I have an English US keyboard
Shift+2 should be @
Shift+3 should be #

When I type Shift+2, I get "
When I type Shift+3, I get £
So it seems to be set to a European / UK Keyboard

This is on a new, fully updated version of bookwork 32-bit and 64-bit on a Raspberry Pi 4.

This is not present on a fully updated 32-bit version of bullseye on a Raspberry Pi 3+.

The difference between the two:
In raspi-config I discovered a difference in the  Localisation options for the keyboard.

bookworm option:
Keyboard = Generic 105-key PC

bullseye option:
Keyboard = Generic 105-key PC (intl.)

There is no Generic 105-key PC (intl.) in bookworm, and I am unsure if this would really fix my issue, since I can type fine once I am logged in.

qrp73, 2023-10-13:
As I know there are two files for keyboard setup:

1) sudo nano /etc/default/keyboard
```
XKBMODEL="pc105"
XKBLAYOUT="us,ru,ua"
XKBVARIANT=""
XKBOPTIONS="grp:alt_shift_toggle"
```

2) nano ~/.config/wayfire.ini
```
[input]
xkb_model = pc105
xkb_layout = us,ru,ua
xkb_options = grp:alt_shift_toggle
```

1 works for console. 2 works for wayland. 
Try to set pc105 for both, may be it can help

rmitchellgit, 2023-10-13:
Thanks, I didn't know there was a different file for console vs. Wayland.  But since both would be using Wayland I don't think that would make a difference (I assume the log in screen is through Wayland since it is GUI?)

They are both already set to pc105.

1. sudo nano /etc/default/keyboard

```
XKBMODEL="pc105"
XKBLAYOUT="us"
XKBVARIANT=""
XKBOPTIONS=""
```

2. nano ~/.config/wayfire.ini

```
[input]
xkb_options=
xkb_model=pc105
xkb_layout=us
xkb_variant=
```

The layout for both is set to English (US).

I tried Alt+Shift on bookworm in text edit and it doesn't seem to be making a difference to my layout, at least when I type @ and # it doesn't alternate between different options when I go through a series of Alt+Shift and testing

I didn't test your last suggest about `dpkb-reconfigure` since I don't think this has to do with the console.  Would the console settings control the log in screen?

qrp73, 2023-10-13:
Just tried login screen, it really don't support keyboard layout config even it it's configured in /etc/default/keyboard and ~/.config/wayfire.ini. 

I think this is a bug, or maybe there is a third config file for login screen :)

lurch, 2023-10-14:
> I assume the log in screen is through Wayland since it is GUI?

Yeah, but the Wayland settings are in `~/.config/wayfire.ini` which is a user-level setting (i.e. you might have multiple users on your system, each with theoretically-different keyboard settings), and so until you've logged in, the log-in screen won't know which settings to use. I found something similar when I was writing the documentation around [screen rotation](https://www.raspberrypi.com/documentation/computers/configuration.html#setting-the-desktop-environment-resolution-and-rotation).
I'm not able to test right now, but to see if my theory is correct, could you try copying your `[input]` section from `~/.config/wayfire.ini` to  `/usr/share/greeter.ini` and seeing if that (after a reboot) then modifies the log-in screen to behave as you expect?

villeneuve, 2023-10-14:
Yes lurch you are right. It works.
From yesterday I was struggling with that. Fresh new Bookworm 64bit install. Keyboard set to french, everything ok in console and in GUI but not ok on graphical login (lightdm greeter). Tried to set keyboard from rpi-imager option or from raspi-config (another sd) : same problem. Tried to modify several files but no success.

Your solution is working. In `/usr/share/greeter.ini`   `xkb_layout` was set to `gb`.
Copying the [input] section from `~/.config/wayfire.ini `to `/usr/share/greeter.ini`  with this setting set to `fr` did it!
It's ok now.
However the bug remains:
raspi-config and rpi-imager should modify this file when the keyboard is changed by the user.

lurch, 2023-10-14:
At the very least, I suspect the first-boot wizard should modify `/usr/share/greeter.ini` with the value selected during the initial configuration - ping @spl237
How likely is it that people will be selecting a different keyboard layout _after_ they've used the first-boot wizard? :thinking: 

And if you've selected a keyboard layout in RPi Imager, perhaps that ought to get set in `/usr/share/greeter.ini` too? ping @XECDesign

qrp73, 2023-10-14:
so, I was right, there is third config for login screen: sudo nano /usr/share/greeter.ini
:)

appleimperio, 2023-10-15:
Thank You this worked for me too. The greeter.ini was set to gb not the same as the wayfire.ini

spl237, 2023-10-16:
> At the very least, I suspect the first-boot wizard should modify `/usr/share/greeter.ini` with the value selected during the initial configuration - ping @spl237 How likely is it that people will be selecting a different keyboard layout _after_ they've used the first-boot wizard? 🤔
> 
> And if you've selected a keyboard layout in RPi Imager, perhaps that ought to get set in `/usr/share/greeter.ini` too? ping @XECDesign

spl237, 2023-10-16:
> At the very least, I suspect the first-boot wizard should modify `/usr/share/greeter.ini` with the value selected during the initial configuration - ping @spl237 

It should be doing this already. The wizard calls do_change_keyboard_rc_gui in raspi-config; this in turn calls update_wayfire_keyboard, which copies the new keyboard layout to both the user config file and the greeter config file.

> How likely is it that people will be selecting a different keyboard layout _after_ they've used the first-boot wizard? 🤔

rc_gui does the same thing by calling the same function.

But now I look in my greeter.ini, I can't see any sign of the relevant settings, so either something is overwriting them later, or they aren't being set correctly. This did work, cos I tested it - something must have broken somewhere since - I'll have a look.

KenT2, 2023-10-16:
How likely is it that people will be selecting a different keyboard layout after they've used the first-boot wizard? 

Quite often in Buster?/Bullseye? as even though I selected English everything at first boot the keyboard was set to US.

You cannot specificallly select a keyboard at this time except to tick the Use US keyboard box.

spl237, 2023-10-16:
OK, I can see what is going on - the settings are being written to one greeter.ini, and they are then being read from a completely different greeter.ini elsewhere on the file system...

Should be easy enough to fix!

spl237, 2023-10-16:
Fixed - will be in next release.

closed 2023-10-16 by spl237