raspberrypi/bookworm-feedback #61: Removing HDMI disables VNC server until reboot

rmitchellgit, 2023-10-13:
I don't think this is a duplicate of the issue described here, but please forgive me if it is.

https://github.com/raspberrypi/bookworm-feedback/issues/42#issuecomment-1759951844

I have bookworm 32-bit on a Raspberry Pi 4.

If I remove the HDMI cable from my Raspberry Pi 4, VNC is disabled until I turn it back on or reboot the Raspberry Pi.

Here are the steps to reproduce my issue:

Raspberry Pi is connected to a monitor with HDMI
sudo raspi-config
Interface Options
VNC
Enabled
Yes

Connect via TigerVNC to verify

Remove HDMI cable

TigerVNC session terminates

Reconnect HDMI cable
sudo raspi-config
Interface Options
VNC
VNC is disabled

This is reproducible through Raspberry Pi Configuration GUI as well

VNC will remain disabled until I turn it back on in raspi-config or Raspberry Pi Configuration, or I reboot the Raspberry Pi.

(This is different than the post I linked above as the user states that on their step 5 "Switch HDMI-attached screen back to the pi - all remains fine." That is not the case for me.  VNC is disabled completely until I turn it back on or reboot.)

notGMman, 2023-10-16:
I think the above description is accurate. My observation should have made it clear that switching back to HDMI only works in the specific situation that the pi booted headless and is then switched to a screen with the same forced headless dimensions as the switched in screen.
Any subsequent "HDMI switch on" stops wayvnc. Including a "transient" switch caused when scrolling past the pi to another device on the HDMI switch.
My fix is a script to check is wayvnc is running and to start it if not. This also logs what happens and serves to demonstrate this is a systematic issue - wayvnc consistently fails under these circumstances.

chepo92, 2023-12-06:
Same issue, Rpi 4, Rpi OS bookworm, having a VNC connection and hdmi display, after removing the hdmi cable, the VNC session is shut down, and cannot be restablished, pluggin in hdmi again to check I find the VNC option under interface in raspi-config is disabled, after enabling i can connect VNC again

BajaYeti, 2024-01-20:
I have a similar problem, just bought a RasPi 5 with NOOBS xd card, turn it all on, tried to use it headless like I have my previous 5 Pis and it doesn't work very well. 

VNC can be enabled via the Raspberry Pi Configuration and connected to, but as soon as the monitor is turned off the VNC connection is terminated and cannot be restored. I'm not using any HDMI swtichers and am connected directly to the TV. I need to use my Pi headless so this is a show-stopper problem for me at the moment.