raspberrypi/bookworm-feedback #63: vcgencmd display_power 0 does nothing with HDMI

pidloop, 2023-10-13:
Installed bookworm on 1MB Pi4 with HDMI display. Works ok but I need a CLI means to control display on/off. In bullseye I used

`vcgencmd display_power 0`

which now returns `display_power=1` and the display stays on. Also tried

`vcgencmd display_power 0 7`

which replies `display_power=-1` and the display stays on.

So... is there a different CLI command now to control HDMI on/off now? Thanks.

pidloop, 2023-10-14:
Update: I found this:

`wlr-randr --output HDMI-A-1 --off`

which does blank the screen but it comes right back on by itself a few seconds later. This is repeatable, I took care not to touch kb or mouse to cause any new input activity.

pelwell, 2023-10-14:
Now a duplicate of #67.

closed 2023-10-14 by pelwell

lurch, 2023-10-14:
> `vcgencmd display_power 0`

That vcgencmd sub-command doesn't work any more, and has been removed from the [vcgencmd documentation](https://www.raspberrypi.com/documentation/computers/os.html#vcgencmd).