raspberrypi/bookworm-feedback #64: Dektop app windows update/refresh/redraw bug

PitHerm, 2023-10-14:
Pi4 B , Bookworm 32bit or 64Bit (both contain the bug) fresh installation and all avaible updates done.

Working directly at the pi (keyboard,display,mouse connected to the Pi or using TigerVnc = same bug

How to simple reproduce:

open smartsim > app window opens
click on "new project" > nothing happens , that's OK
click on "new component" > nothing happens, that's wrong, the working area should show a dotted grid !

now click on "new component" again and close the new appearing window OR open a terminal window or something else and close it > NOW the working area of the first smartsim window appears with the dotted grid.

select a component from the right side of the selection menu and click within the dotted area tp place it > the component doesn't appear !

now click on "new component" again and close the new appearing window OR open a terminal window or something else and close it > NOW the working area of the first smartsim window appears with the component.

lurch, 2023-10-14:
Does it redraw correctly if you switch from Wayland to X11? See the [blog post](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/) for details.

PitHerm, 2023-10-14:
Sorry , forgot to test at that time.
Just selected X11 at raspi-config and tested it
with X11 redraw works fine.

lurch, 2023-10-14:
Thanks.
Have you _only_ seen this issue with [SmartSim](http://www.smartsim.org.uk/), or does it happen with other "Dektop app windows" too?
Looking at SmartSim's [GitHub repo](https://github.com/ashleynewson/SmartSim), the last [stable release](https://github.com/ashleynewson/SmartSim/releases) was 10 years ago (!), and unfortunately the source-code seems to have not been touched for the last [6 years](https://github.com/ashleynewson/SmartSim/branches).

ping @ashleynewson in case they have any feedback.

PitHerm, 2023-10-14:
actually i need to test others.
but smartsim works fine on bullseye and on x11 so it should work on wayland

ashleynewson, 2023-10-14:
I have neglected SmartSim for a while now and it is somewhat in a state of disrepair.

There are a few different rendering bugs in SmartSim which I had investigated in the past (and fixed), but I'm not sure they made it into any official releases. They certainly didn't make it onto any official Raspberry Pi builds. The most recent code I'd uploaded to SmartSim's repo was in a reorganisation branch (https://github.com/ashleynewson/SmartSim/commits/reorganise), which contains fixes for various bugs. I also have some local commits I never uploaded.

Unfortunately, Vala isn't the most stable of languages, so even that code has since stopped compiling.

https://github.com/ashleynewson/SmartSim/commit/87149acf090f6e0541abc73a8e7a2deac5acd27a included some rendering bug fixes. I have somewhat lost context on the exact nature of the problem, but I'd made mistakes when migrating from GTK+2 to GTK+3, which somewhat changed how rendering updates were handled.

ashleynewson, 2023-10-14:
If it's at all of use for debugging, I've updated my https://github.com/ashleynewson/SmartSim/commits/reorganise branch such that is can compile on Vala 0.56.13. However, you might need to tweak the vala/Makefile in order for it to compile on your system. (The autotools build mechanism is meant for the C code, which isn't updated.)

Doing something like `make -C vala debug && vala/smartsim` might work.

In general though, I don't currently have plans to maintain SmartSim thoroughly.

PitHerm, 2023-10-14:
@ashleynewson thank you very much , i will give it a try the next days.

lurch, 2023-10-14:
Thank you for the information Ashley! :+1: