raspberrypi/bookworm-feedback #67: Screen Blanking not working

kmpm, 2023-10-14:
I am running 64-bit bookworm (fresh install) on a 400 and I cant get screen blanking to work.
The screen never shuts of and it used to in bullseye.

I do get the same behavior as described in issue #63 and `wlr-randr --output HDMI-A-1 --off`shuts of the screen, for a short while.
Don't know if its related or not.

kmpm, 2023-10-14:
Let me add something I found after investigating.
- I changed idle.dpms_timeout in `~/.config/wayfire.ini`to 60 (from 600).
- Reboot
- Waited for about a minute

The screen went blank but came back after about 10 seconds, just as when using `wlr-randr --output HDMI-A-1 --off`

qrp73, 2023-10-14:
it seems that some app requests to wake up...
May be browser?

kmpm, 2023-10-14:
Thanks @qrp73 for the tips, but...
With all extra panel plugins removed (CPU and CPU-temp), a reboot, with no windows open or mouse connected.

Goes blank for ~10 seconds __once__ after ~idle.dpms_timeout seconds after desktop shows after reboot. 
If I start using the keyborad to take it out of idle state I get a new blank after ~dpms_timeout seconds again.

So no windows open and no mouse connected... same buggy behavior.

kmpm, 2023-10-14:
More investigating.
Just as my monitor, from AOC, shows "NO SIGNAL" the screen comes back to life. 
That message usually shows for a second and a half before the monitor goes into standby but now I barely register it before the monitor turns on again and shows the desktop. 
Are there any logs that I could monitor from a remote ssh session that might show whats going on?

And blanking works if I switch to X11.

qrp73, 2023-10-14:
In my case blanking works as expected on bookworm 64 with Wayland/wayfire. 

There is configuration in ~/.config/wayfire.ini
```
[idle]
screensaver_timeout = -1
dpms_timeout = 300
```
After 300 seconds the signal disappears and display is going in sleep mode. It woke up when I move windows.
I tried different values for dpms_timeout, it works as expected.

kmpm, 2023-10-14:
Here is my wayfire.ini
```
[idle]
dpms_timeout=30

```

I even tried adding your `screensaver_timeout=-1` but it had no effect what so ever.
It goes blank after 30 seconds (or whatever I put there) but it doesn't stay blank, wakes up just as "NO SIGNAL" shows on the monitor. 
And still, no input on keyboard, mouse disconnected.
I am booting from an USB connected SSD though, but that shouldn't really matter. Nothing else is connected.

kmpm, 2023-10-14:
- Installed Raspberry PI OS Bookworm 64-bit on an SD-card.
- Disconnected USB-drive. Only mouse connected on USB.
- Ran apt update and apt upgrade to get all the latest packages.
- Rebooted
- Enabled screen blanking using raspi-config
- Rebooted
- Got into desktop
- Waited for 600 seconds
- Screen went blank for about 10 seconds, once.

So USB-boot or from SD-card does not make a difference.
A fresh install+upgrade before enabling screen blanking does not make a difference.
Official RPI powersupply, official HDMI-cable, official mouse.

kmpm, 2023-10-15:
Ok. Got hold of a Dell monitor that unfortunately only has DVI so I had to use and HDMI to DVI adapter but with that setup, blanking works.
The AOC monitor that is messing with me has speakers and audio via HDMI, can the new audio framwork have anything to do with it, I see a lot of pipewire and wireplumber entries in journalctl around blanking?

kmpm, 2023-10-15:
This log is just me switching the monitor of and on again. Similar pipewire messages as when blanking.
[log-20231015-2.txt](https://github.com/raspberrypi/bookworm-feedback/files/12908965/log-20231015-2.txt)

qrp73, 2023-10-16:
Are you testing blanking on virtual console or graphical desktop?

I found that display blanking doesn't works on virtual console.
~/.config/wayfire.ini settings is for wayland graphical desktop. 

So there is probably another config file for virtual console blanking or it's just not implemented yet...

kmpm, 2023-10-16:
Graphical desktop, nothing else.

lurch, 2023-10-16:
> So there is probably another config file for virtual console blanking

https://www.raspberrypi.com/documentation/computers/configuration.html#console

qrp73, 2023-10-16:
for blanking in console, there is needs to add ```consoleblank=300``` to /boot/cmdline.txt string

kmpm, 2023-10-16:
Sorry @qrp75. Same issue in console mode. Goes blank for about 10 seconds then back but with cursor not blinking and no repeated blanks.

popcornmix, 2023-10-16:
@kmpm can you try adding `vc4.force_hotplug=1` to start of line in `/boot/firmware/cmdline.txt`?

I have a suspicion your display is waggling hotplug after it sees the display go blank (I don't know why).
But that behaviour does wake up the display system (it looks like a display has been unplugged and replugged).

kmpm, 2023-10-16:
Could someone, where blanking works in graphical mode, please check with `sudo journalctl -xe` or something if you have entries similar to this on or around when the screen went black.

```
Oct 16 20:50:29 kmpm-pi-4c0625 xdg-desktop-portal-wlr[2232]: 2023/10/16 20:50:29 [ERROR] - pipewire: fatal error event from core
...
Oct 16 20:50:29 kmpm-pi-4c0625 xdg-desktop-por[1132]: Caught PipeWire error: connection error
...
Oct 16 20:50:29 kmpm-pi-4c0625 systemd[915]: xdg-desktop-portal-wlr.service: Main process exited, code=exited, status=1/FAILURE
Oct 16 20:50:29 kmpm-pi-4c0625 systemd[915]: xdg-desktop-portal-wlr.service: Failed with result 'exit-code'.
...
Oct 16 20:50:29 kmpm-pi-4c0625 systemd[915]: xdg-desktop-portal-wlr.service: Scheduled restart job, restart counter is at 6.
Oct 16 20:50:29 kmpm-pi-4c0625 systemd[915]: Stopped xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
Oct 16 20:50:29 kmpm-pi-4c0625 systemd[915]: Starting xdg-desktop-portal-wlr.service - Portal service (wlroots implementation)...
Oct 16 20:50:29 kmpm-pi-4c0625 xdg-desktop-portal-wlr[2696]: 2023/10/16 20:50:29 [ERROR] - xdpw: Could not find render node
Oct 16 20:50:29 kmpm-pi-4c0625 xdg-desktop-portal-wlr[2696]: 2023/10/16 20:50:29 [ERROR] - xdpw: Could not find render node
```
I'm curious about why xdg-desktop-portal-wlr fails and restarts.
When I use the Dell monitor, that works, I don't get any of that crap.
That is why I'm suspecting that audio is involved somehow. The AOC monitor has it through HDMI, the Dell one doesn't

kmpm, 2023-10-16:
@popcornmix , with `vc4.force_hotplug=1`it could not boot to desktop. Just a blank screen after bootsplash when it normally switches to desktop.

kmpm, 2023-10-16:
Is there a way of not starting the pipewire stuff at all. I'm really OK without sound while testing :)

there must be a config somewhere that controls those kind of things but I haven't found it.

qrp73, 2023-10-17:
> Could someone, where blanking works in graphical mode, please check with `sudo journalctl -xe` or something if you have entries similar to this on or around when the screen went black.

no there is no log records when screen blanking happens.

note: I'm using kernel 6.6.0-rc5-v8+, because there are some unfixed bugs on 6.1

lurch, 2023-10-17:
> Is there a way of not starting the pipewire stuff at all.

There's an option in `raspi-config` to toggle between using PipeWire and PulseAudio, as mentioned in https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

kmpm, 2023-10-17:
@lurch, thanks for what on paper would have been fine.
But even if i switch to PulseAudio I still get those log entries about pipewire...
So I'll rephrase then question. Does anyone know how to __disable__ all audio frameworks? Not uninstall, not switching just disable.

popcornmix, 2023-10-17:
Possibly:
```
systemctl --user stop pulseaudio.socket
systemctl --user stop pulseaudio.service
systemctl --user stop pipewire.socket
systemctl --user stop pipewire.service
```
to stop now, and if you don't want it to restart after boot:
```
systemctl --user disable pulseaudio.socket
systemctl --user disable pulseaudio.service
systemctl --user disable pipewire.socket
systemctl --user disable pipewire.service
```

kmpm, 2023-10-18:
`sudo apt purge pipewire pulseaudo && sudo reboot` does not help with blanking on the AOC monitor. 
And it seems as if wf-panel-pi needs it somehow because it didn't show at all when neither pulseaudio or pipewire were present.

lurch, 2023-10-18:
I wonder if you might need to remove the Bluetooth / Volume / etc. taskbar plugins first, and that's why wf-panel-pi isn't launching?

kmpm, 2023-10-21:
@lurch, removing the taskbar plugins fixes wf-panel-pi. The issue with screen blanking still exists though.

BGRideout, 2023-10-21:
From response on Pi forum, adding vc4.force_hotplug=3 to end of command in /boot/cmdline.txt (value =1 for hda1, 2 hda2, 3 both) corrected the issue in pi400 with 2 monitors.

pidloop, 2023-10-21:
Tried this on 1 MB Pi4 running bookworm. Now my GUI doesn't run at all, boots to console login. Removing it does not return to normal either, still boots to console. So this broke the GUI completely.

pidloop, 2023-10-21:
But this lead to some really good news!! I did a fresh download/install of bookworm, then the usual sudo apt update/upgrade and now `wlr-randr --output HDMI-A-1 --off` works **and it stays off**! And --on brings it back. Yah!

pidloop, 2023-10-21:
BTW, bookworm on my 1GB Pi4 is pretty much worthless, starting most any app crashes the whole OS from lack of memory. I added a 10 GB swap file and now everything runs fine. I should add that I use an external 1TB USB-3 SSD as my main drive, I don't think swapping on a microSD is a good idea for a long term solution.

For those who are not familiar with adding a swap file here are the basic steps:

```
SWAPFN=/var/10GB.swap
sudo dd if=/dev/zero of=$SWAPFN bs=1M count=10K
sudo chmod 600 $SWAPFN
sudo chown root $SWAPFN
sudo mkswap $SWAPFN
sudo swapon $SWAPFN
```

Only thing is I don't know how to make it do the swapon automatically when booted. Anyone know?

pidloop, 2023-10-21:
Well well. Even though wlr-randr --off now works, keyboard and mouse activity do **not** turn the screen back on. You really must run wlr-randr --on. So evidently this is not cooperating with (or even using) the usual screensaver gizmo.

So I'm still not where I need to be: a command line means to turn off the display that turns it on again with user input.

Maybe I should just go back to x11 for now....

kmpm, 2023-10-22:
@pidloop, good that wlr-randr works for you but no change for me. Just did a fresh install and update+upgrade without any improvment. The screen goes black for about 10 seconds but as soon as "NO SIGNAL" shows on the screen it turns back on, without user input.

kmpm, 2023-10-25:
@popcornmix, `vc4.force_hotplug=1` works now. Probably some of the updates pulled something in.

But if it isn't needed in X11 then I shouldn't have to do it for wayland either, in my opinion at least.

pelwell, 2023-12-08:
> when i ssh from terminal it says fail to connect to display.

Please show a complete, unedited log of what you typed and all output.

1a2a3a1q2s3c, 2023-12-08:
on the pc terminal i enter-
$ wlr-randr
result = failed to connect to display

on the pi i enter-
$ wlr-randr
result = 
HDMI-A-2 "Philips monitor brand
physical size: 530x300mm
enabled: yes
modes: bunch of resolution and hz
position: 0, 0
transform: 90
scale: 1.0

1a2a3a1q2s3c, 2023-12-08:
Sorry so where do I put this code at and what’s the expected outcome? Is it put it at the cron job?

my aim is to put the auto schedule on off as how vcgencm used to do for me in pi3b+

popcornmix, 2023-12-08:
> on the pc terminal i enter-
$ wlr-randr
result = failed to connect to display

So do the same but with:
```$ WAYLAND_DISPLAY="wayland-1" wlr-randr```

pelwell, 2023-12-08:
Or you could `export WAYLAND_DISPLAY="wayland-1"` somewhere.

1a2a3a1q2s3c, 2023-12-08:
yes $ WAYLAND_DISPLAY="wayland-1" wlr-randr shows the information in the terminal.

i tried on terminal  - WAYLAND_DISPLAY="wayland-1" wlr-randr --output HDMI-A-2 --off and screen turns off but it auto came back on without me inputting WAYLAND_DISPLAY="wayland-1" wlr-randr --output HDMI-A-2 --on which is weird?

1a2a3a1q2s3c, 2023-12-08:
added 

[idle]
dpms_timeout=30

and it doesnt work.

popcornmix, 2023-12-08:
> i tried on terminal - WAYLAND_DISPLAY="wayland-1" wlr-randr --output HDMI-A-2 --off and screen turns off but it auto came back on 

Can you try this suggestion: https://github.com/raspberrypi/bookworm-feedback/issues/67#issuecomment-1765094899

Note: use
`vc4.force_hotplug=1` for HDMI-A-1
`vc4.force_hotplug=2` for HDMI-A-2
`vc4.force_hotplug=3` for HDMI-A-1 and HDMI-A-2

1a2a3a1q2s3c, 2023-12-08:
getting the same issue after doing the hotplug = it could not boot to desktop. Just a blank screen after bootsplash.

1a2a3a1q2s3c, 2023-12-08:
to update, my display is in vertical. when i turn off and turn on, the display revert back to horizontal.

1a2a3a1q2s3c, 2023-12-08:
anyone has any idea how to automate or cronjob the on off? nothing seems to work on my end

bembudo, 2023-12-08:
Very interesting....I  too have this problem. I have two monitors a 22" ViewSonic (with Speakers built in) AND a 22" Dell (no Speakers)
when the ViewSonic is connected to my Raspberry Pi 5 Screen blanking does not work. If i connect my Dell monitor to my Raspberry Pi 5 Screen blanking works without problems.

toggling between pulseAudio and PipeWire does not make a difference either.

1a2a3a1q2s3c, 2023-12-08:
Are you able to do a cron job to the dell?

1a2a3a1q2s3c, 2023-12-08:
It supposed to turn off on a specific set time and on at a specific time. This is inserted into the cron job. Those code that I gave does not function in cron as expected and I don’t know why…

pidloop, 2023-12-08:
@1a2a3a1q2s3c your cron jobs don't pick up your env. so try setting the
wayland display name the same as suggested by @popcornmix

52 17 * * * WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output
HDMI-A-2 --on

pidloop, 2023-12-08:
# introduces a comment, anything after that on the same line is ignored.

1a2a3a1q2s3c, 2023-12-08:
Sorry what do you mean? I don’t quite understand.

1a2a3a1q2s3c, 2023-12-08:
Yea so I did the wordings before the # and i still doesn’t works.

lurch, 2023-12-19:
I've had a go at experimenting with this myself...
I'm testing with an up-to-date Bookworm 64-bit image running on a Pi 4B, and I've SSHed into the Pi from my laptop.

With just a single HDMI monitor plugged in to HDMI0, if I run
```
WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-1 --off
```
over SSH, then the display turns off, and stays off even if I jiggle the USB mouse or type on the USB keyboard directly attached to the Pi. If I run
```
WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-1 --on
```
over SSH then the display turns back on again, and everything is fully functional.

If I swap the monitor over to HDMI1, then running the command above to turn off HDMA-A-1 says "unknown output HDMI-A-1", which is to be expected. If I run
```
WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-2 --off
```
then the display turns off, but then automatically turns back on again after about 4 seconds!

With a monitor plugged into each of HDMI0 and HDMI1, then running
```
WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-1 --off
```
turns off the display connected to HDMI0, and it stays off. HDMI1 stays on, and any application-windows that were on HDMI0 then jump over to HDMI1.
Running
```
WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-1 --on
```
turns the display back on again.

With both displays turned on, running
```
WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-2 --off
```
turns off the second display (with the first display remaining enabled), and again the HDMI1 display automatically turns back on again after about 4 seconds.

If I run the commands to turn off HDMI-A-1 and then turn off HDMI-A-2 then both displays are off briefly, but they _both_ wake up again after a few seconds.

I also ran the same tests on a Pi5, and that exhibited the same behaviour.

TL;DR - it looks like the `wlr-randr` command can (currently) only "properly" turn off a single display connected to HDMI0 (the micro-HDMI port nearest to the USB-C power-input port).

popcornmix, 2023-12-19:
Can you add `vc4.force_hotplug=3` to cmdline.txt (on existing line) and check if behaviour changes?

1a2a3a1q2s3c, 2023-12-19:
> I've had a go at experimenting with this myself... I'm testing with an up-to-date Bookworm 64-bit image running on a Pi 4B, and I've SSHed into the Pi from my laptop.
>
> With just a single HDMI [...]

No dice for me even if I use the hdmi nearest to the usb c. The only way it work properly for me is to insert the vc hotplug command

lurch, 2023-12-19:
Ahhh, the culprit (and cause of my confusion) was that one of my micro-HDMI cables must have a broken (or unconnected) hotplug pin! So this "faulty" cable was the one that I was always connecting to HDMI1, and this is why the monitor connected to that display was "staying off" :facepalm: 

So if I swap over my two HDMI cables, I then see the reversed behaviour - HDMI0 is the display that keeps "waking up" and HDMI1 is the display that "stays asleep". As @popcornmix said in an earlier comment, the act of sending the display to sleep causes the display to wiggle its hotplug pin, and that makes Wayland think that a new display has been plugged in, and so it switches the display back on again.

So (as both @6by9 and @popcornmix suggested), if I add `vc4.force_hotplug=3` to the end of the line in `/boot/firmware/cmdline.txt` this then causes the Pi to totally ignore the state of the hotplug pin for both HDM

popcornmix, 2023-12-19:
Are you sure it is the cable and not the display?
I'd expect a cable with faulty hotplug to not be detected (unless you'd added something like a `video=HDMI-A-1:3840x2160@60D` line to cmdline.txt.

lurch, 2023-12-19:
> Are you sure it is the cable and not the display?

Ahh yes, it seems you're right - with so many different connections I must have been getting confused :laughing: 
If I swap over _both_ ends of each HDMI cable, then it remains HDMI-A-2 that keeps waking up again. And that means that I can "get away with" just `vc4.force_hotplug=2` in `cmdline.txt`, and I can switch both displays off with `wlr-randr`, and they both stay off, regardless of which cable I use to connect which monitor.

Based on all of that, does everyone agree that this issue can be closed now?

pidloop, 2023-12-19:
> Based on all of that, does everyone agree that this issue can be closed now?

Yes, many thanks to all involved.

But I could use a recap. Please summarize:

- what the proper commands are for on/off
- what one-time changes are required on a fresh install
- whether these will eventually become the default (and thus unnecessary).

lurch, 2023-12-19:
> * what the proper commands are for on/off

Turn off HDMI0: `/usr/bin/wlr-randr --output HDMI-A-1 --off`
Turn on HDMI0: `/usr/bin/wlr-randr --output HDMI-A-1 --on`
Turn off HDMI1: `/usr/bin/wlr-randr --output HDMI-A-2 --off`
Turn on HDMI1: `/usr/bin/wlr-randr --output HDMI-A-2 --on`

If you're running over SSH (or from within a script) you might need to prefix those commands with
 `WAYLAND_DISPLAY="wayland-1" `

> * what one-time changes are required on a fresh install

None.
But if the monitor you have connected to HDMI0 "wiggles its hotplug" and automatically turns back on again, add `vc4.force_hotplug=1` to the end of the existing line in `/boot/firmware/cmdline.txt` (don't add any extra lines) and reboot. If the monitor you have connected to HDMI1 automatically turns back on again, add `vc4.force_hotplug=1` instead. And if _both_ monitors automatically turn back on again after switching them off with `wlr-randr`, use `vc4.force_hotplug=3`.

> * whether these will eventually become the default (and thus unnecessary).

We won't be adding those force_hotplug entries to `cmdline.txt` by default, but we _are_ starting to investigate how much work would be involved in implementing my earlier suggestion of ignoring any hotplug events that occur soon after `wlr-randr` has switched off a display.

closed 2023-12-19 by lurch

p1r473, 2024-02-13:
Ive tried all of these commands but no matter what I get:

```
❯ WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-1 --off
failed to connect to display                                                                                                                              
❯ WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-1 --on
failed to connect to display                                                                                                                              
❯ WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-2 --off
failed to connect to display                                                                                                                              
❯ WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-1 --off
failed to connect to display                                                                                                                              
❯ WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-1 --on
failed to connect to display                                                                                                                              
❯ WAYLAND_DISPLAY="wayland-1" /usr/bin/wlr-randr --output HDMI-A-2 --off
failed to connect to display   
```
Im using a Pi 5 x64 with Bookworm and Im trying to turn off my HDMI monitor.
Ive tried both HDMI ports.

popcornmix, 2024-02-14:
> Im using a Pi 5 x64 in CLI mode with Bookworm and Im trying to turn off my HDMI monitor.
Ive tried both HDMI ports.

wayland isn't running in CLI mode, so wlr-randr isn't expected to do anything.

[11 more replies]