raspberrypi/bookworm-feedback #68: xrdp problem.

PitHerm, 2023-10-14:
Pi 4 B booworm 32 and 64  both X11 and wayland both
all updates
installed xrdp

desktop appears to be fine but
simply launch an internet browser you will see what happens

lurch, 2023-10-14:
A more detailed description than "you will see what happens" would be useful. Are you seeing the problem (whatever it is) on the locally-attached HDMI screen, or are you seeing this problem when remotely connected to the Pi over xrdp?

PitHerm, 2023-10-14:
Argh ... i beg your pardon
i was sure that i wrote it :-(

remotely connected to the Pi over xrdp
simply launch an internet browser
the browser window shows absolutely unuseable things
[image]

lurch, 2023-10-14:
Thank you for the more detailed feedback :+1:

CharlesGodwin, 2023-10-16:
I see the same thing.

Developdog, 2023-10-17:
I have same problem, not only chromium but also game

[image]

I'm using Pi 4 B 2G 
Raspberry Pi OS Full(64-bit) BookWorm(all updates)
Installed xrdp
How can I solve it?

cct-eng, 2023-10-22:
I am getting the same, with both X11 or Wayland, when accessing over rdp, with xrdp installed.
If I access using attached KB and monitor there is no problem.
It only seems tpo affect grahic based programs - a lot of standard programs work fine. In particular either browser fails, as does the Rpi Imager
A post on Reddit suggested that affected programs launched witout GPU support can work fine, for example. Can this be done globally in the X11 and/or wayland settings?

CharlesGodwin, 2023-10-22:
Someone suggested creating a new user and login with that user. That works for me

cct-eng, 2023-10-22:
Thanks, but that makes no difference at all

cct-eng, 2023-10-24:
Neither  firefox --disable-gp & nor chromium-browser --disable-gp & make any difference, so I presume that is a red herring

Chris

juliankay, 2023-11-17:
This still seems to be a problem, was there a workaround?

cct-eng, 2023-11-17:
No real workaround. I fixed Chromium browser as above, and then fixed Firefox by switching off any hardware acceleration in its settings, when directly connected. So I am still unable to use many graphical GUI tools. This despite quite a few updates coming down the line.

juliankay, 2023-11-18:
I resolved the issue by using option 2 here: https://forums.raspberrypi.com/viewtopic.php?p=2151746

cct-eng, 2023-11-18:
Thanks - that works.

CharlesGodwin, 2023-11-19:
Option 1 or Option 2 worked for me too.

PitHerm, 2023-11-28:
Yes, i tested option 2 and it works here too
but these are still workarounds, i hope this bug will be solved

purplealien51, 2025-01-28:
Hi,
I am having issue with xrdp on Pi ZeroW bookworm 32 bit 
Can not even rdp after installing both xrdp and xorgrdp, anybody tried it?

only see blue screen from microsoft RDP application and then error message.