raspberrypi/bookworm-feedback #70: In X11 mode different desktops on two monitors does not work

KenT2, 2023-10-14:
Steps to show:
Select X11 mode in raspi-config
In the screen editor ensure the two monitors are side by side. hdmi0 is primary
In appearance settings untick 'show identical desktop on second monitor' and make the two desktops look different.

Expect to see:
Two different desktops

What I actually see with two HD monitors:
hdmi0 desktop is duplicated on hdmi1.
hdmi1 desktop is hidden
Pointer moves correctly between the two desktops
Cannot click or drag on the hdm1 desktop
Changed the hdmi1 monitor to a  VGA one ,then the righthand  part of the hdmi0 desktop is shown, no scaling to monitor resolution.

Other info:
xrandr gives the correct result
Seem to have identical boot information on both screens which is different to Bullseye

spl237, 2023-10-16:
I can repeat. Looks like a display driver problem - forwarded to relevant people.

6by9, 2023-10-16:
https://github.com/raspberrypi/linux/pull/5654 should fix this.
Once CI has completed (takes about an hour), running ```sudo rpi-update pulls/5654``` should get the kernel for testing.

KenT2, 2023-10-17:
Fixed, thanks

closed 2023-10-17 by KenT2