raspberrypi/bookworm-feedback #71: KDE xdg-desktop-portal ScreenCast does not work (Bookworm ARM64)

theofficialgman, 2023-10-14:
Attempting to use kde's ScreenCast xdg-desktop-portal does not work in any software tested. Attempted to use OBS-Studio (deb), flameshot  (deb), chromium (deb), and Kooha (flatpak) which all use the portal in a KDE Wayland session under SDDM and they all failed.

OBS-Studio, pipewire reports there are no captures available.
flameshot refuses to open.
chromium does not allow selecting window/fullscreen (fails to get it).
Kooha fails also saying ScreenCast is not available.

theofficialgman, 2023-10-15:
Upstream debian bug was filed last month -> https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1051276

lurch, 2024-07-18:
Closing, as we only officially support the default desktop supplied with Raspberry Pi OS; we don't support KDE ourselves.

closed 2024-07-18 by lurch