raspberrypi/bookworm-feedback #73: HDMI/CEC: not able to power up CEC device anymore

chincheta0815, 2023-10-15:
After updating to bookworm I am not able to switch on a CEC device anymore by e.g. using libcec's cec-client. Once powered up everything else via CEC works well.

There was not any change made to the whole assembly. I just updated to bookworm and got that strange issue.

chincheta0815, 2023-10-15:
I forgot to mention: I use a Raspberry Pi 4.

Any logs I could provide?

chincheta0815, 2023-10-16:
Well, I did respect all you mention already.

The issue https://github.com/Pulse-Eight/libcec/issues/538 was address by some commits made by popcornmix (https://github.com/popcornmix/libcec/commit/6239ae1b092d85e54530e95bbefbdc4fa520dba1 & https://github.com/popcornmix/libcec/commit/b30fc8333e3f80bf39ebddf4241507e779b51d77) both did work.

I raised an issue at Pulse-Eight too, as I did not know how to solve my issue (kernel issue after update or libcec).
The issue itself is seemlingly affecting just the powering up of a device via CEC. Afterwards everything works well.

So far, I get the same issue/result with cec-ctl - which imposes that there is something more general going wrong here.
To me it seems that there is a connection to the bookworm update rather than libcec, as it worked on bullseye.

Are there any measure I can take to narrow the issue down (logfiles, tests, etc...)?

popcornmix, 2023-10-16:
@XECDesign 
```
pi@pi5:~/projects/libcec $ cec-client
No device type given. Using 'recording device'
* failed to open vchiq instance
```

I think you are building with `-DHAVE_LINUX_API=1` `-DHAVE_RPI_API=1`
I've rebuilt using  `-DHAVE_LINUX_API=1 -DHAVE_RPI_API=0` and it looks more plausible. 
```
pi@pi5:~/projects/libcec $ LD_PRELOAD=build_linux/src/libcec/libcec.so ./build_linux/src/cec-client/cec-client
No device type given. Using 'recording device'
CEC Parser created - libCEC version 6.0.2
no serial port given. trying autodetect: 
 path:     /sys/devices/platform/soc/107c701400.hdmi/cec0
 com port: /dev/cec0

opening a connection to the CEC adapter...
DEBUG:   [              27]	Broadcast (F): osd name set to 'Broadcast'
DEBUG:   [              27]	CLinuxCECAdapterCommunication::Open - m_path=/dev/cec0 m_fd=3 bStartListening=1
DEBUG:   [              27]	CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_G_PHYS_ADDR - addr=1500
DEBUG:   [              27]	CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [              29]	CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=8000 num_log_addrs=1
NOTICE:  [              29]	connection opened
DEBUG:   [              29]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=8000 phys_addr=1500
DEBUG:   [              29]	processor thread started
DEBUG:   [              29]	<< Broadcast (F) -> TV (0): POLL
TRAFFIC: [              29]	<< f0
DEBUG:   [             381]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=f0 opcode=ffffffff
TRAFFIC: [             381]	<< f0
DEBUG:   [             453]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=f0 opcode=ffffffff
DEBUG:   [             453]	>> POLL not sent
DEBUG:   [             453]	TV (0): device status changed into 'not present'
DEBUG:   [             453]	registering new CEC client - v6.0.2
DEBUG:   [             453]	SetClientVersion - using client version '6.0.2'
NOTICE:  [             453]	setting HDMI port to 1 on device TV (0)
DEBUG:   [             453]	<< Broadcast (F) -> TV (0): POLL
TRAFFIC: [             453]	<< f0
DEBUG:   [             524]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=f0 opcode=ffffffff
TRAFFIC: [             524]	<< f0
DEBUG:   [             596]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=f0 opcode=ffffffff
DEBUG:   [             596]	>> POLL not sent
DEBUG:   [             596]	SetConfiguration: double tap timeout = 200ms, repeat rate = 0ms, release delay = 500ms
DEBUG:   [             596]	detecting logical address for type 'recording device'
DEBUG:   [             596]	trying logical address 'Recorder 1'
DEBUG:   [             596]	<< Recorder 1 (1) -> Recorder 1 (1): POLL
TRAFFIC: [             596]	<< 11
DEBUG:   [             667]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=11 opcode=ffffffff
TRAFFIC: [             667]	<< 11
DEBUG:   [             739]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=11 opcode=ffffffff
DEBUG:   [             739]	>> POLL not sent
DEBUG:   [             739]	using logical address 'Recorder 1'
DEBUG:   [             739]	Recorder 1 (1): device status changed into 'handled by libCEC'
DEBUG:   [             739]	Recorder 1 (1): power status changed from 'unknown' to 'on'
DEBUG:   [             739]	Recorder 1 (1): vendor = Pulse Eight (001582)
DEBUG:   [             739]	Recorder 1 (1): CEC version 1.4
DEBUG:   [             739]	AllocateLogicalAddresses - device '0', type 'recording device', LA '1'
DEBUG:   [             739]	CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0
DEBUG:   [             739]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=1500
DEBUG:   [             886]	CLinuxCECAdapterCommunication::SetLogicalAddresses - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0002 num_log_addrs=1
DEBUG:   [             886]	Recorder 1 (1): osd name set to 'CECTester'
DEBUG:   [             886]	Recorder 1 (1): menu language set to 'eng'
DEBUG:   [             886]	using auto-detected physical address 1500
DEBUG:   [             886]	Recorder 1 (1): physical address changed from ffff to 1500
DEBUG:   [             886]	<< Recorder 1 (1) -> broadcast (F): physical address 1500
TRAFFIC: [             886]	<< 1f:84:15:00:01
DEBUG:   [             886]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0002 phys_addr=1500
DEBUG:   [            1299]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=5 addr=1f opcode=84
NOTICE:  [            1299]	CEC client registered: libCEC version = 6.0.2, client version = 6.0.2, firmware version = 0, logical address(es) = Recorder 1 (1) , physical address: 1.5.0.0, git revision: libcec-6.0.2+3-b30fc83~dirty, compiled on 2023-10-16 16:36:11 by pi@pi5 on Linux 6.1.57-v8+ (aarch64), features: P8_USB, DRM, P8_detect, randr, RPi, Linux
DEBUG:   [            1299]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
DEBUG:   [            1299]	<< Recorder 1 (1) -> TV (0): POLL
TRAFFIC: [            1299]	<< 10
[... cut ...]
DEBUG:   [            1518]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=10 opcode=ffffffff
TRAFFIC: [            1518]	<< 10
DEBUG:   [            1589]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=24 len=1 addr=10 opcode=ffffffff
DEBUG:   [            1589]	>> POLL not sent
DEBUG:   [            1589]	not sending command 'give device power status': destination device 'TV' marked as not present
waiting for input
```

As we don't support firmware cec any more, it probably makes sense to drop support for that.

popcornmix, 2023-10-16:
I don't have a cec TV handy but I can detect the Onkyo receiver I have:
```
pi@pi5:~/projects/libcec $ echo "scan" | LD_PRELOAD=build_linux/src/libcec/libcec.so ./build_linux/src/cec-client/cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #1: Recorder 1
address:       1.5.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

device #5: Audio
address:       1.0.0.0
active source: no
vendor:        Onkyo
osd string:    TX-NR609
CEC version:   1.4
power status:  on
language:      ???

currently active source: unknown (-1)
```

chincheta0815, 2023-10-16:
@popcornmix Are you also able to power up the receiver and send it to standby? This is actually my issue. I cannot power it up. ("on 5" in the cec-client)

What do you mean by droppingvthe support for cec firmware?

popcornmix, 2023-10-16:
On bullseye and now bookworm, the kms (firmware) driver has been deprecated and we use the kms (arm side) driver.
We've basically dropped any fkms tools (like tvservice) from bookworm. fkms support for libcec should be dropped as well.

I can power off my Onkyo with `cec-ctl -d0 --standby --to 5`. I seem to be able to power it on with `cec-ctl -d0  --active-source phys-addr=1.5.0.0 --to 5`.

I'm not sure that's possible with libcec with my setup (no cec cable tv) due to [this](https://github.com/Pulse-Eight/libcec/issues/104)

chincheta0815, 2023-10-16:
okay. that really helped a lot.
I now (really) build without HAVE_RPI_API. Even with HACE_RPI_API=0 the respective files were compiled. I cannot if linked. Anyway,I will add that to my libcec repo then.

@popcornmix and @6by9 Thanks a lot!

closed 2023-10-16 by chincheta0815