raspberrypi/bookworm-feedback #74: Bookworm raspi-config  keyboard selection bug

PitHerm, 2023-10-15:
bookworm (tested only on x64)

raspi-config

5 Localisation Options 
L3 Keyboard > appears Reloading keymap > then raspi-config back at main menu
no setting possible

XECDesign, 2023-10-16:
I'm guessing that was meant to be arm64, now x64?

Desktop or lite image?

PitHerm, 2023-10-16:
Bookworm 64bit full image installed on Pi4 B using Pi-Imager

XECDesign, 2023-10-16:
Thanks. It worked for me when I tried, but I was running 32bit desktop image and sshing in without using putty. I'll try to reproduce the issue with your setup when I get a chance.

PitHerm, 2023-10-16:
I just tested Bookworm Desktop 32 and 64 bit ; Bookworm OS FULL 32 and 64 bit
using Pi Imager with applying settings
Enable SSH
user = pi
PW = raspberry
enable Wifi > setting ssid and password (even triid leaving the country setting and keyboard setting to GB vs setting them to DE)

wrote every image

tested putty connection > sudo raspi-config > set keyboard >FAILED
installed xrdp > tested remote Desktop from Windows > login > xterm >sudo raspi-config > set keyboard >FAILED

then: sudo apt update > sudo apt upgrade 
tested putty connection > sudo raspi-config > set keyboard >FAILED
 tested remote Desktop from Windows > login > xterm >sudo raspi-config > set keyboard >FAILED

[image]

[image]

qrp73, 2023-10-17:
Keboard layout config in raspi-config is broken by design. It doesn't allows you to configure layout list and define hot key to change layout. It only allows to select single layout, so you will be unable to use ASCII keys if you select non engilsh layout. 

The only way to configure keyboard properly is to edit config files directly. 
Here is example with 3 layouts: US, RU, UA and Alt+Shift key for layout switch:

1) for console: sudo nano /etc/default/keyboard
```
XKBMODEL="pc105"
XKBLAYOUT="us,ru,ua"
XKBVARIANT=""
XKBOPTIONS="grp:alt_shift_toggle"
```

2) for Wayland: nano ~/.config/wayfire.ini
```
[input]
xkb_options = grp:alt_shift_toggle
xkb_model = pc105
xkb_layout = us,ru,ua
xkb_variant =
```

3) For login screen: sudo nano /usr/share/greeter.ini
```
[input]
xkb_options = grp:alt_shift_toggle
xkb_model = pc105
xkb_layout = us,ru,ua
xkb_variant =
```

lurch, 2023-10-17:
> It doesn't allows you to configure layout list and define hot key to change layout.

Even if that is the case, it seems to be unrelated to the issue being reported here? :thinking:

qrp73, 2023-10-17:
Since it talking about keyboard layout in raspi-config, I wrote it here just to note that this functionality of raspi-config is invalid and needs to be rewritten, also added example on how raspi-config should hanlde keyboard settings, it will be helpful for developers who will working on that feature.

lurch, 2023-10-17:
> I wrote it here just to note that this functionality of raspi-config is invalid and needs to be rewritten, also added example on how raspi-config should hanlde keyboard settings,

That should probably be logged as a separate issue, rather than added to this one.

qrp73, 2023-10-18:
There is no sense to fix it separately, because it will be automatically fixed if keboard layout setting in raspi-config will be rewritten in proper way.

lurch, 2024-07-18:
@XECDesign I was digging in to this today, and finally managed to figure out how to reproduce it...
`raspi-config` does "Localisation Options" -> "Keyboard" by running `dpkg-reconfigure keyboard-configuration`.

If I'm connected to my Pi over SSH and I have a USB keyboard attached, then `sudo dpkg-reconfigure keyboard-configuration` displays the expected interactive keyboard-selection program. But if I **don't** have a USB keyboard attached to the Pi, then `sudo dpkg-reconfigure keyboard-configuration` just silently exits without doing anything. And this explains why it always worked for @XECDesign (because they always had a USB keyboard attached) but why it didn't work for @PitHerm (because they never had a USB keyboard attached) :joy: 

@XECDesign / @spl237 Is having no keyboard attached something that we could "detect" inside `raspi-config`, and then display a suitable warning message to the user if they try to run "Localisation Options" -> "Keyboard" ?

spl237, 2024-07-18:
> @XECDesign / @spl237 Is having no keyboard attached something that we could "detect" inside `raspi-config`, and then display a suitable warning message to the user if they try to run "Localisation Options" -> "Keyboard" ?

Detecting the lack of a keyboard is harder than you think! The problem is wireless dongles, which invariably seem to identify themselves as keyboards even when they aren't associated with one. Don't know if that would be a problem here though.

lurch, 2024-07-18:
Alternatively, could we force `dpkg-reconfigure keyboard-configuration` to _always_ run, even if it thinks that there isn't a keyboard connected?

XECDesign, 2024-07-22:
I had to add an [ugly workaround](https://github.com/RPi-Distro/pi-gen/commit/165e9602376b1ee25d8599389e5ae5643dd9754c) for a somewhat related problem in pi-gen. The scripts under the hood of keyboard-configuration are a bit of a mess and I wouldn't want to start patching them without understanding why the behave the way they do.

If a user tries to configure a keyboard that does not exist and debian's scripts don't take that situation into account, I don't think that's a problem with Raspberry Pi OS. I don't think we're doing anything wrong or unusual.

lurch, 2024-07-23:
> I don't think we're doing anything wrong or unusual.

Fair enough, I'll close this issue. @nathan-contino perhaps https://www.raspberrypi.com/documentation/computers/configuration.html#localisation-options needs to be updated to mention that trying to configure a keyboard won't do anything if there's not actually a USB keyboard attached?

jumperger, 2024-08-14:
I'm still having this raspi-config keyboard problem even with a wired usb-Keyboard connected at boot

XECDesign, 2024-08-14:
Are you running raspi-config over ssh or not?