raspberrypi/bookworm-feedback #76: Right click on Audio icon crashes taskbar  in X11/ Pulseaudio mode

KenT2, 2023-10-15:
How to reproduce:
Use sudo raspi-config to select X11 mode and Pulseaudio mode
Reboot
Right click on the audio icon on the taskbar to change audio device

Expect to see:
Menu to select audio device
 
What I see:
Task bar disappears then comes back

Query:
Do I actually need to select Pulseaudio in X11 mode? Leaving it on Pipewire seems to work OK

qrp73, 2023-10-15:
In my case it works as expected. I'm using CX31993 usb dongle as sound card.

spl237, 2023-10-16:
I can repeat. You shouldn't need to use PulseAudio with X11 though.

spl237, 2023-10-16:
Problem found - it was caused by a fix for a different problem we introduced last week! Fixed - will be in next update.

closed 2023-10-16 by spl237