raspberrypi/bookworm-feedback #79: In rc_gui, the Interfaces-->VNC switch is disabled

JinShil, 2023-10-16:
It appears rc_gui is checking if realvnc-vnc-server is installed instead of wayvnc.  Therefore, it's not possible to enable VNC through the GUI.  Enabling through raspi-config seems to work fine though.

spl237, 2023-10-16:
realvnc-vnc-server is always installed, so this should not be an issue.

JinShil, 2023-10-16:
RealVNC cannot be used for for commercial purposes.  Therefore, in our OS, which is generated using pi-gen, real-vnc-server is not installed.

wayvnc can be used for commercial purposes, so in our OS it can be installed.  rc_gui should probably check for real-vnc-server OR wayvnc.

spl237, 2023-10-16:
Fixed - will be in next release.

lurch, 2023-10-16:
> in our OS

This repo is for tracking issues with Raspberry Pi OS Bookworm, not third-party OSes. Bugs with `rc_gui` should probably be reported at https://github.com/raspberrypi-ui/rc_gui