raspberrypi/bookworm-feedback #80: What are the instructions for Bookworm to configure the RP1 I2C rate?

Tiger0215, 2023-10-16:
The default I2C rate of RP1 is 400K? How to modify to lower I2C rate？
The RP1 document says "if required for device configuration",but I did not find the relevant I2C configuration method.

[image]

pelwell, 2023-10-16:
It can be the same as on the older Pis:
```
pi@raspberrypi:~$ dtparam -h i2c_arm_baudrate
i2c_arm_baudrate        Set the baudrate of the ARM's i2c interface
                        (default "100000")
```
However, all the Pi 5 versions of the i2c overlays also have a baudrate parameter:
```
$ dtoverlay -h i2c1-pi5
Name:   i2c1-pi5

Info:   Enable i2c1 (Pi 5 only)

Usage:  dtoverlay=i2c1-pi5,<param>=<val>

Params: pins_2_3                Use GPIOs 2 and 3 (default)
        pins_10_11              Use GPIOs 10 and 11
        baudrate                Set the baudrate for the interface (default
                                "100000")
```
Note that when you request the `i2c1` overlay on a Pi 5 the firmware translates that into a request for `i2c1-pi5`. The same will be true of the `dtoverlay` Linux utility, but that change hasn't rolled out yet.

So, to answer your question, either of the following should work:
```
dtparam=i2c_arm_baudrate=200000
dtoverlay=i2c1,baudrate=200000
```
Your question has highlighted that, contrary to the overlay and parameter help, the default baudrate is not 100000 as it should be. That will be fixed in a future update.

Tiger0215, 2023-10-16:
Thank you very much for your quick reply
In addition, how to check whether the I2C rate corresponding to the DSI/CSI of RP1 has changed after modification?
As mentioned in #52 , the way to view the Pi4 is:
`$ od -i --endian=big /proc/device-tree/soc/i2c@7e205000/clock-frequency
0000000      200000
0000004`

The viewing position corresponding to Pi5 seems to be different, and I did not see the modified data.

Thanks for the default baudrate change, I will also try to install the updated kernel and dtb files.

pelwell, 2023-10-16:
It's essentially the same command, but with a different DT path. The following command allows us to find the correct paths - it will work on all Pis, just with different output:
```
pi@raspberrypi:~$ grep -a . /proc/device-tree/aliases/i2c*
/proc/device-tree/aliases/i2c:/axi/pcie@120000/rp1/i2c@74000
/proc/device-tree/aliases/i2c0:/axi/pcie@120000/rp1/i2c@70000
/proc/device-tree/aliases/i2c1:/axi/pcie@120000/rp1/i2c@74000
/proc/device-tree/aliases/i2c10:/soc/i2c@7d005600
/proc/device-tree/aliases/i2c2:/axi/pcie@120000/rp1/i2c@78000
/proc/device-tree/aliases/i2c3:/axi/pcie@120000/rp1/i2c@7c000
/proc/device-tree/aliases/i2c4:/axi/pcie@120000/rp1/i2c@80000
/proc/device-tree/aliases/i2c5:/axi/pcie@120000/rp1/i2c@84000
/proc/device-tree/aliases/i2c6:/axi/pcie@120000/rp1/i2c@88000
```
So for i2c_csi_dsi1 (i2c4) use:
```
$ od -i --endian=big /proc/device-tree/axi/pcie@120000/rp1/i2c@80000/clock-frequency
```
And for i2c_csi_dsi0 (i2c6) use:
```
$ od -i --endian=big /proc/device-tree/axi/pcie@120000/rp1/i2c@88000/clock-frequency
```
Note that for the current Pi 5 dtb the clock-frequency is not specified, so both commands will fail with `No such file or directory`.