raspberrypi/bookworm-feedback #83: Broken kernel USB drivers for RPI4 rev 1.5 with VIA chip

qrp73, 2023-10-17:
Just intalled latest bookworm image with apt update && apt upgrage. And there is still present USB drivers bug which was fixed for beta version rpi-6.1.y. See link: https://forums.raspberrypi.com/viewtopic.php?t=357473#p2143222

This fix is missing from current bookworm release (6.1.0-rpi4-rpi-v8).

qrp73, 2023-10-17:
Here is the patch for this issue from bullseye raspi os: https://github.com/P33M/linux/commit/fd915cc0176bb0c04281f6c01689445a6cac0dfd

And here is discussion with some details:  https://github.com/raspberrypi/linux/issues/3060

popcornmix, 2023-10-17:
Yes, there hasn't been a stable (apt) kernel update yet.
It will happen soon. For now, you need to use rpi-update to get the fix.

qrp73, 2023-10-18:
here is report that this issue also affect USB HDD devices:
https://forums.raspberrypi.com/viewtopic.php?t=357879#p2146182

The pull request sudo rpi-update pulls/5642 solves it for USB HDD, the same as for RTLSDR USB dongle

CRGer, 2023-10-25:
Hey @popcornmix, 
do you know when the next stable (apt) update will be? And if not where I can track if the PR #5462 is included in apt? The use of rpi-update is unfortunately not possible for me, because other packages had problems with it. Or is there another workaround available? Thanks!

qrp73, 2023-10-26:
@CRGer, it seems that they published it today. I tried to update, at a glance works ok and the issue looks like solved. But needs to be tested on clean bookworm install, because my system was updated to 6.6, so the test is not clean

CRGer, 2023-10-26:
Tried it on a clean install, the issue is definitely still present. But I have the latest release of Pi OS installed, where the following is used (extract from change notes), so might be different for you:
"* Switched from raspberrypi-kernel to Debian-based kernel packaging (linux-image-rpi-*)
  * Switched from raspberrypi-bootloader to Debian based firmware packaging (raspi-firmware)"

eric1tran, 2023-11-02:
Hi @popcornmix , do you have an estimated date when a new Raspberry Pi OS will be released, which presumably has this fix in?

ghollingworth, 2023-11-02:
We're intending on making a new release next week.

qrp73, 2023-11-15:
not relevant anymore

CRGer, 2023-11-15:
Hi @qrp73 
why is this not relevant anymore? Was it fixed and is included in apt-get?

qrp73, 2023-11-15:
I decided to quit from rpi community, I'm tired of censorship and bullying from rpi staff. If it still actual for you, you can open your issue. I don't want to deal with rpi anymore.

JamesH65, 2023-11-16:
Still appears to be relevant.

pelwell, 2023-11-16:
@qrp73 I was saddened to hear that you've experienced censorship and bullying. I don't know the incidents to which you are referring, but I have seen that your alter ego has been banned on the forums. After questioning, the moderator concerned said they had received a few complaints - again, I've not seen any examples of the complaints or what might have triggered them - but that in hindsight they may have reacted hastily.

I believe you have been unbanned, and I hope you reconsider your decision - you're clear, detailed bug reports have been very helpful.

JamesH65, 2023-11-16:
It also might be worth mentioning that many of the forum moderators are not actually employees of Raspberry Pi. If you have a problem with anything on the forum, please report the post and a staff member of Raspberry Pi will investigate; as has happened in this case.

qrp73, 2023-11-18:
@JamesH65, @pelwell I already reported it on raspberrypi.com and security@raspberrypi.com, but after 6 days there is no response, all my complaints were just ignored. 

As I remember, the last thing that I wrote was a post that 27W PSU is not enough information to decide if it can be used with rpi and user needs to make sure that PSU has 5V output (which is precisely correct answer, because 27W PSU can have any voltage on the outpout and 12V 27W PSU can burn out rpi), but I see that this message was deleted and I was banned with reason "wrong, misleading and/or confusing posts".  

From my point of view, banning user for correct and polite message looks very crazy. 
I believe there is a strong reason to leave places and communities where such behavior are practiced.

In any case thanks to all guys who helped me with issues.

pelwell, 2023-11-18:
It's a shame that the actions of one private individual, not a Raspberry Pi employee, led to this, but do what you feel you have to do.

pelwell, 2023-11-18:
P.S. You can email me at phil@raspberrypi.com, should you wish.

Joe-raspi, 2023-11-21:
Hi,
same problem here with Raspi 4 and 
SDR stick RTL2838UHIDIR

I upgraded fro  1042 to  kernel 5.15.0-1043-raspi today but still no improvement.
Same symptoms as shown above by YipYup.

With 1041 everything worked perfectly.

cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.3 LTS"
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

Joe

popcornmix, 2023-11-21:
> This issue was fixed on our 6.1 tree via https://github.com/raspberrypi/linux/pull/5642, but I'm not sure it's hit a stable kernel release yet.

It should be in stable. apt is:
```Linux version 6.1.0-rpi6-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27)```

which is newer than that commit.
The source linux tree tag `stable_20231030` also includes the commit.

Joe-raspi, 2023-11-21:
Thanks a lot for feedback.
So I will have to place the bug at another place or try a newer kernel ...

mattfox27, 2023-12-03:
following, same issue RPI4

mattfox27, 2023-12-03:
apt get rtl_4

> Hi, same problem here with Raspi 4 and SDR stick RTL2838UHIDIR
>
> I upgraded fro 1042 to kernel 5.15.0-1043-raspi today but still no improvement. Same symptoms as shown above by YipYup.
>
> With 1041 [...]

Did you ever figure this out?  I have the same issue

Joe-raspi, 2023-12-03:
Hi,

Am 03.12.23 um 17:45 schrieb Matt Klimmer:
> apt get rtl_4
>
>     Hi, same problem here with Raspi 4 and SDR stick RTL2838UHIDIR
>
>     I upgraded fro 1042 to kernel 5.15.0-1043-raspi today but still no improvement. Same symptoms
>     as shown [...]

Not yet, sorry.

My observations:

- I have a RTL2832 stick and one older Noxon for reference (but with less sensitivity)

Both sticks are running well with openSUSE Leap 15.4 and 15.5 (x64 platform)
On Tumbleweed RTL2832 is unusable.

Things were getting more worse:
As I started this thread the Noxon worked stable with Ubuntu on Raspi, kernel 5.15.0-1043-raspi,
the RTL2832 only worked the first time after plugging it in.

Some updates later (firmware?) both sticks don't work at all now :-(

So I will try to open a ticket at bugzilla next week.
(But this would mean starting my project from scratch in case the problem can be solved for TW)

Joe

Joe-raspi, 2023-12-08:
Hi Matt,

bug report is open:

https://bugzilla.opensuse.org/show_bug.cgi?id=1217860

Maybe you want to comment it.

Joe

Am 03.12.23 um 17:45 schrieb Matt Klimmer:
> apt get rtl_4
>
>     Hi, same problem here with Raspi 4 and SDR stick RTL2838UHIDIR
>
>     I upgraded fro 1042 to kernel 5.15.0-1043-raspi today but still no improvement. Same symptoms
>     as shown [...]